chipsalliance/Surelog #65: Switch from flatbuffers to Cap'n'Proto

mithro, 2019-11-17:
For all the [SymbiFlow project](https://symbiflow.github.io) we are standardizing on [Cap'n'Proto library](https://capnproto.org/). 

> Cap’n Proto is an insanely fast data interchange format and capability-based RPC system. Think JSON, except binary. Or think Protocol Buffers, except faster. In fact, in benchmarks, Cap’n Proto is INFINITY TIMES faster than Protocol Buffers.

https://capnproto.org/news/2014-06-17-capnproto-flatbuffers-sbe.html

[image]

hzeller, 2019-11-17:
Guiding factors in the context of Surelog should be ease of generating code, simplicity, ease of getting people involved.
Also probably speed :)
That it is used in a Symbiflow project might be a contributing factor but should not be the main one.

alaindargelas, 2023-07-17:
Sure, no objection.

alaindargelas, 2023-07-17:
Please make sure the preprocessing/parsing is still incremental per-file-unit as it is now.