chipsalliance/Surelog #68: Options: Make gnu conform: Make long options start with double-dash

hzeller, 2019-11-17:
`surelog -help` works, but the more common expected name `surelog --help` does not (this is a [gnu-convention])

Short options should be understood with or without space in the middle.
`surelog -l foo.log` works, but `surelog -lfoo.log` does not.
(this is implemented as an exception for `-I` for instance. But this should always work as expected).

Support long-options with only one dash (e.g. `-help` instead of `--help`) can be tricky of course when we make short-options work without spaces, as it can introduce ambiguities. So if not possible, we can probably even drop it.

[gnu-convention]: https://www.gnu.org/software/libc/manual/html_node/Argument-Syntax.html

hzeller, 2019-11-17:
Might be also worthwhile evaluating https://abseil.io/docs/cpp/guides/flags
(since we anyway might want to look at abseil to geti things such as c++11 compatible `absl::string_view`)

alainmarcel, 2019-11-18:
The tool specific options can be made to accept both with and without space, 
For the Verilog Standard compilation options, I think we need to be strict and only accept what the standard says: -v -f -y -L -map -cfg -cfgfile falls into that category.

hzeller, 2019-11-18:
Though nobody would complain if we also fix it to also be reasonable and accpet the single-char options followed without space and the long-options such as -cfg and -cfgfile also valid as --cfg and --cfgfile ...

alainmarcel, 2019-11-18:
Ok
-- Alain Marcel