chipsalliance/Surelog #97: Carve out a 'surelog api' to programmatically parse and iterate over results

hzeller, 2019-11-20:
This is essentially already there, but we need to have an install target that moves the relevant headers in $(PREFIX)/include/surelog and a library to link against, then have some example.

(Psuedocode...)
```C++
#include <surelog/compiler.h>
#include <surelog/design.h>  // making up these names here

int main() {
   SURELOG::CompileFlags flags;
   flags.compile = true;
   flags.elaborate = false;
   auto compiled = SURELOG::Compile("myfile.sv", flags);
   for (auto module : getModules(compiled)) {
      // do a thing
  }
```

It might be easier to first re-org some header files inside Surelog to have an `include/surelog/` directory already that contains the relevant headers that we then intend to install in `$(PREFIX)/include/surelog` - this keeps it simple to separate.

(all public header files should be lower-case and with dashes or underscores (pick one) as separators if such thing is needed. `#include <surelog/foo-bar.h>`)

hzeller, 2019-11-20:
Ah, looks like a duplicate of #93

alainmarcel, 2019-11-21:
Mostly fixed by #101 modulo the renaming of the files. There is a surelog.h that encapsulates the class headers.

closed 2019-11-21 by alainmarcel