chipsalliance/Surelog #108: Probably classetype not initialized

Thomasb81, 2019-11-23:
I experiment crash when running `make` to build the tool:

> $make
> [...]
> [100%] Linking CXX executable dist/Release/surelog
>        Creating staging for precompiled packages
>        Creating OVM precompiled package...
> src/tcmalloc.cc:283] Attempt to free invalid pointer 0x7ffcf13523e0 
> make[3]: *** [CMakeFiles/surelog-bin.dir/build.make:99: dist/Release/surelog] Aborted (core dumped)
> make[3]: *** Deleting file 'dist/Release/surelog'
> make[3]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make[2]: *** [CMakeFiles/Makefile2:73: CMakeFiles/surelog-bin.dir/all] Error 2
> make[2]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make[1]: *** [Makefile:163: all] Error 2
> make[1]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make: *** [Makefile:6: release] Error 2
> 

While the `make debug` is working fine.

I suspect an uninitialized variable usage like this one sound to me:

https://github.com/alainmarcel/Surelog/blob/897bced8b5b13b51a26789f4e706ebd778d29d22/third_party/antlr4/runtime/Cpp/runtime/src/atn/BasicState.cpp#L11

Have you consider to submit a PR for your antlr4 runtime change ? (ref https://github.com/antlr/antlr4/issues/2584)

alainmarcel, 2019-11-23:
classtype is initialized in:

ATNState::ATNState() {
  classtype = ATNStateClass;
}
BasicState inherits from ATNState and its constructor does the initialization of the field. Each subclass adds through the bit masking its specialization. This works. I suspect something else is going on. Run a valgrind --tool=memcheck on it.

As to do a PR, I think there is more to do with the runtime optimization before that, like replacing systematically all the remaining dynamic casting and taking a look at the synchronization code.

Thomasb81, 2019-11-23:
incorrect issue

Thomasb81, 2019-11-23:
> (gdb) r
> Starting program: /home/tom/prog/git/Surelog/build/dist/Release/surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> src/tcmalloc.cc:283] Attempt to free invalid pointer 0x7fffffffd920 
> 
> Program received signal SIGABRT, Aborted.
> __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
> 50	../sysdeps/unix/sysv/linux/raise.c: Aucun fichier ou dossier de ce type.
> (gdb) bt
> #0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
> #1  0x00007ffff7310899 in __GI_abort () at abort.c:79
> #2  0x00007ffff7854fb9 in tcmalloc::Log(tcmalloc::LogMode, char const*, int, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem, tcmalloc::LogItem) () from /lib/x86_64-linux-gnu/libtcmalloc.so.4
> #3  0x00007ffff7848d49 in ?? () from /lib/x86_64-linux-gnu/libtcmalloc.so.4
> #4  0x00007ffff78686b9 in tc_free () from /lib/x86_64-linux-gnu/libtcmalloc.so.4
> #5  0x0000555555a92502 in SURELOG::Compiler::createFileList_() [clone .cold] ()
> #6  0x0000555555b56442 in SURELOG::Compiler::compile() ()
> #7  0x0000555555b128c6 in SURELOG::start_compiler(SURELOG::CommandLineParser*) ()
> #8  0x0000555555af285b in executeCompilation(int, char const**, bool, bool) ()
> #9  0x0000555555adcabd in main ()
>

Thomasb81, 2019-11-23:
I can provide you the valgrind report but because the program has segfault ... there is 16345 errors.

alainmarcel, 2019-11-23:
Please send the header of the report up to the first couple of errors.

alainmarcel, 2019-11-23:
When running valgrind, you have to add the -nopython option on the command line.
Valgrind and Python don't work nicely in my setup.
I'm running on my end too.

Thomasb81, 2019-11-23:
> $ valgrind --tool=memcheck --log-file=valgrind_ovm_pkg.log  /home/tom/prog/git/Surelog/build/dist/Release//surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout -nopython
> Erreur de segmentation (core dumped)
[valgrind_ovm_pkg.log](https://github.com/alainmarcel/Surelog/files/3882693/valgrind_ovm_pkg.log)

alainmarcel, 2019-11-23:
Something is wrong in your installation/use of tcmalloc in surelog.Maybe you installed tcmalloc after finding you could not link?
tcmalloc has to be installed first on the system. Then make clean of everything, antlr, surelogMake sure you pull Surelog as I just fixed a problem with make clean.Also, make sure you don't have environment variables that are changing the ways GCC with deal with memory allocation.then:

cd Surelogmake cleanmake -j

This is my valgrind run, it is clean:
 valgrind --tool=memcheck dist/Release/surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp  -mt 0 -parse -nocomp -noelab -nopython==30841== Memcheck, a memory error detector==30841== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.==30841== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info==30841== Command: dist/Release/surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nopython==30841== ==30841== brk segment overflow in thread #1: can't grow to 0x4a28000==30841== Syscall param msync(start) points to uninitialised byte(s)==30841==    at 0x5459BA0: __msync_nocancel (syscall-template.S:84)==30841==    by 0x654E351: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x6551DF6: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x655233D: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x6552698: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x654ED60: _ULx86_64_step (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x5696442: ??? (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5696C4D: GetStackTrace(void**, int, int) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x568803F: tcmalloc::PageHeap::GrowHeap(unsigned long) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x56883A2: tcmalloc::PageHeap::New(unsigned long) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5686CB3: tcmalloc::CentralFreeList::Populate() (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5686EA7: tcmalloc::CentralFreeList::FetchFromOneSpansSafe(int, void**, void**) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==  Address 0xffefff150 is on thread 1's stack==30841== [INFO :CM0023] Creating log file ./slpp_all/surelog.log.
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:1208 Unused macro argument "FLAG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:1231 Unused macro argument "ARG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:1246 Unused macro argument "ARG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:1246 Unused macro argument "FLAG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:1470 Unused macro argument "T".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:2842 Unused macro argument "FLAG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:2910 Unused macro argument "ARG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_object_defines.svh:2910 Unused macro argument "FLAG".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_printer_defines.svh:368 Unused macro argument "KEY".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_callback_defines.svh:255 Unused macro argument "CB".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_callback_defines.svh:255 Unused macro argument "OPER".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_callback_defines.svh:256 Unused macro argument "CB".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_callback_defines.svh:256 Unused macro argument "OBJ".
[WARNI:PP0113] ovm-2.1.2/src/macros/ovm_callback_defines.svh:256 Unused macro argument "OPER".
[WARNI:PP0113] ovm-2.1.2/src/compatibility/urm_message_defines.svh:111 Unused macro argument "HIERARCHY".
==30841== Syscall param msync(start) points to unaddressable byte(s)==30841==    at 0x5459BA0: __msync_nocancel (syscall-template.S:84)==30841==    by 0x654E351: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x6551DF6: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x655233D: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x6552698: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x654ED60: _ULx86_64_step (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x5696442: ??? (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5696C4D: GetStackTrace(void**, int, int) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x568803F: tcmalloc::PageHeap::GrowHeap(unsigned long) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x56883A2: tcmalloc::PageHeap::New(unsigned long) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5686CB3: tcmalloc::CentralFreeList::Populate() (in /usr/lib/libtcmalloc.so.4.2.6)==30841==    by 0x5686EA7: tcmalloc::CentralFreeList::FetchFromOneSpansSafe(int, void**, void**) (in /usr/lib/libtcmalloc.so.4.2.6)==30841==  Address 0xffeffa000 is on thread 1's stack==30841==  264 bytes below stack pointer==30841== ==30841== Syscall param msync(start) points to uninitialised byte(s)==30841==    at 0x5459BA0: __msync_nocancel (syscall-template.S:84)==30841==    by 0x654E351: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x654E8CF: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x6551DA9: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30841==    by 0x655233D: ??? (in /usr/lib/x86_64-linux-gnu/libunwind.so.8.0.1)==30 [... cut ...]
[  FATAL] : 0[ SYNTAX] : 0[  ERROR] : 0[WARNING] : 16[   NOTE] : 0==30841== ==30841== HEAP SUMMARY:==30841==     in use at exit: 0 bytes in 0 blocks==30841==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated==30841== ==30841== All heap blocks were freed -- no leaks are possible==30841== ==30841== For counts of detected and suppressed errors, rerun with: -v==30841== Use --track-origins=yes to see where uninitialised values come from==30841== ERROR SUMMARY: 514 errors from 3 contexts (suppressed: 0 from 0)

Thomasb81, 2019-11-23:
May be I am doing it wrong.

$git pull
Already up to date.
$make clean
$make -j 20
crash
$make VERBOSE=1
[...]
[100%] Linking CXX executable dist/Release/surelog
make -f CMakeFiles/surelog-bin.dir/build.make CMakeFiles/surelog-bin.dir/build
make[3]: Entering directory '/home/tom/prog/git/Surelog/build'
[100%] Linking CXX executable dist/Release/surelog
/usr/bin/cmake -E cmake_link_script CMakeFiles/surelog-bin.dir/link.txt --verbose=1
/usr/bin/c++   -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free  -O3 -DNDEBUG -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -O3 -DNDEBUG   -rdynamic CMakeFiles/surelog-bin.dir/src/main.cpp.o  -o dist/Release/surelog libsurelog.a -lpython3.7m -L/home/tom/prog/git/Surelog/dist/ -Wl,-Bstatic -lantlr4-runtime -Wl,-Bdynamic -L/home/tom/prog/git/Surelog/build/third_party/flatbuffers -Wl,-Bstatic -lflatbuffers -Wl,-Bdynamic -ldl -lutil -lm -lrt -lpthread -ltcmalloc 
echo \ \ \ \ \ \ \ Creating\ staging\ for\ precompiled\ packages
       Creating staging for precompiled packages
mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/sv
mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/python
cp -f /home/tom/prog/git/Surelog/build/../src/API/builtin.sv /home/tom/prog/git/Surelog/build/dist/Release/sv
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/ovm-2.1.2 ovm-2.1.2
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/uvm-1.2 uvm-1.2
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/vmm-1.1.1a vmm-1.1.1a
rm -rf slpp_all slpp_unit
cp -f /home/tom/prog/git/Surelog/build/../src/API/slSV3_1aPythonListener.py /home/tom/prog/git/Surelog/build/dist/Release/python/
cp -f /home/tom/prog/git/Surelog/build/../src/API/slformatmsg.py /home/tom/prog/git/Surelog/build/dist/Release/python/slformatmsg.py
cp -f /home/tom/prog/git/Surelog/build/../src/API/slwaivers.py /home/tom/prog/git/Surelog/build/dist/Release/python/
echo \ \ \ \ \ \ \ Creating\ OVM\ precompiled\ package...
       Creating OVM precompiled package...
/home/tom/prog/git/Surelog/build/dist/Release//surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout
src/tcmalloc.cc:283] Attempt to free invalid pointer 0x7ffc406e2ae0 
make[3]: *** [CMakeFiles/surelog-bin.dir/build.make:99: dist/Release/surelog] Aborted (core dumped)
make[3]: *** Deleting file 'dist/Release/surelog'
make[3]: Leaving directory '/home/tom/prog/git/Surelog/build'
make[2]: *** [CMakeFiles/Makefile2:115: CMakeFiles/surelog-bin.dir/all] Error 2
make[2]: Leaving directory '/home/tom/prog/git/Surelog/build'
make[1]: *** [Makefile:163: all] Error 2
make[1]: Leaving directory '/home/tom/prog/git/Surelog/build'
make: *** [Makefile:6: release] Error 2

At this step I cannot found the surlog binary...
Probably the build system remove it.
So to progress I manually relaunch the command display by make:
$/usr/bin/c++   -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free  -O3 -DNDEBUG -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -O3 -DNDEBUG   -rdynamic CMakeFiles/surelog-bin.dir/src/main.cpp.o  -o dist/Release/surelog libsurelog.a -lpython3.7m -L/home/tom/prog/git/Surelog/dist/ -Wl,-Bstatic -lantlr4-runtime -Wl,-Bdynamic -L/home/tom/prog/git/Surelog/build/third_party/flatbuffers -Wl,-Bstatic -lflatbuffers -Wl,-Bdynamic -ldl -lutil -lm -lrt -lpthread -ltcmalloc

The command work and produce the binary.
But it crash on the tcmalloc.

hellosurlog is also crashing who is present before I manually link surelog binary.

I am on a fresh Ubuntu 19.10 install (less than 20 days) did not do anything special about tcmalloc library.

alainmarcel, 2019-11-23:
Alright, lets try not to use tcmalloc on your system, to rule out a few things.
You'll have to edit the CMakeLists.txt at the top level to:
- remove tcmalloc from the link line, and also 
- remove the CXX flags "-fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free"

Then make clean
Then recompile everything.....

The issue you are facing seems system related.
The build works fine on Travis and for my other user(s).
https://travis-ci.com/alainmarcel/Surelog/builds/137890562

Thomasb81, 2019-11-23:
Same:

> [100%] Linking CXX executable dist/Release/surelog
> /usr/bin/cmake -E cmake_link_script CMakeFiles/surelog-bin.dir/link.txt --verbose=1
> /usr/bin/c++    -O3 -DNDEBUG -O3 -DNDEBUG   -rdynamic CMakeFiles/surelog-bin.dir/src/main.cpp.o  -o dist/Release/surelog libsurelog.a -lpython3.7m -L/home/tom/prog/git/Surelog/dist/ -Wl,-Bstatic -lantlr4-runtime -Wl,-Bdynamic -L/home/tom/prog/git/Surelog/build/third_party/flatbuffers -Wl,-Bstatic -lflatbuffers -Wl,-Bdynamic -ldl -lutil -lm -lrt -lpthread -ltcmalloc 
> echo \ \ \ \ \ \ \ Creating\ staging\ for\ precompiled\ packages
>        Creating staging for precompiled packages
> mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/sv
> mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/python
> cp -f /home/tom/prog/git/Surelog/build/../src/API/builtin.sv /home/tom/prog/git/Surelog/build/dist/Release/sv
> ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/ovm-2.1.2 ovm-2.1.2
> ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/uvm-1.2 uvm-1.2
> ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/vmm-1.1.1a vmm-1.1.1a
> rm -rf slpp_all slpp_unit
> cp -f /home/tom/prog/git/Surelog/build/../src/API/slSV3_1aPythonListener.py /home/tom/prog/git/Surelog/build/dist/Release/python/
> cp -f /home/tom/prog/git/Surelog/build/../src/API/slformatmsg.py /home/tom/prog/git/Surelog/build/dist/Release/python/slformatmsg.py
> cp -f /home/tom/prog/git/Surelog/build/../src/API/slwaivers.py /home/tom/prog/git/Surelog/build/dist/Release/python/
> echo \ \ \ \ \ \ \ Creating\ OVM\ precompiled\ package...
>        Creating OVM precompiled package...
> /home/tom/prog/git/Surelog/build/dist/Release//surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout
> src/tcmalloc.cc:283] Attempt to free invalid pointer 0x7ffc4b8f3c80 
> make[3]: *** [CMakeFiles/surelog-bin.dir/build.make:99: dist/Release/surelog] Aborted (core dumped)
> make[3]: *** Deleting file 'dist/Release/surelog'
> make[3]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make[2]: *** [CMakeFiles/Makefile2:115: CMakeFiles/surelog-bin.dir/all] Error 2
> make[2]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make[1]: *** [Makefile:163: all] Error 2
> make[1]: Leaving directory '/home/tom/prog/git/Surelog/build'
> make: *** [Makefile:6: release] Error 2

I am using gcc-9 and the travis-CI you point seems to use gcc-7

Thomasb81, 2019-11-23:
> set(CMAKE_CXX_FLAGS  "${CMAKE_CXX_FLAGS} -g ${MY_CXX_WARNING_FLAGS}")
> set(CMAKE_CXX_FLAGS_DEBUG  "${CMAKE_CXX_FLAGS_DEBUG} -O0 -g ${MY_CXX_WARNING_FLAGS}")
> set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} -g -O3 -DNDEBUG ${MY_CXX_WARNING_FLAGS}")

Since it is still working in Debug but not in release ... I will end-up to toy with -DNDEBUG (did not look at what it control, and then -O2, -O1 ...)

Also I reproduce the issue with hellosureword binary. If I compare valgrind log between Release and Debug. The release log has about 300 additionnal lines ...

alainmarcel, 2019-11-23:
You need to remove  -ltcmalloc (The link line) to get rid of tcmalloc, you are still using it.

Thomasb81, 2019-11-23:
```
[100%] Linking CXX executable dist/Release/surelog
/usr/bin/cmake -E cmake_link_script CMakeFiles/surelog-bin.dir/link.txt --verbose=1
/usr/bin/c++   -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free  -O3 -DNDEBUG -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -O3 -DNDEBUG   -rdynamic CMakeFiles/surelog-bin.dir/src/main.cpp.o  -o dist/Release/surelog libsurelog.a -lpython3.7m -L/home/tom/prog/git/Surelog/dist/ -Wl,-Bstatic -lantlr4-runtime -Wl,-Bdynamic -L/home/tom/prog/git/Surelog/build/third_party/flatbuffers -Wl,-Bstatic -lflatbuffers -Wl,-Bdynamic -ldl -lutil -lm -lrt -lpthread 
echo \ \ \ \ \ \ \ Creating\ staging\ for\ precompiled\ packages
       Creating staging for precompiled packages
mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/sv
mkdir -p /home/tom/prog/git/Surelog/build/dist/Release/python
cp -f /home/tom/prog/git/Surelog/build/../src/API/builtin.sv /home/tom/prog/git/Surelog/build/dist/Release/sv
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/ovm-2.1.2 ovm-2.1.2
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/uvm-1.2 uvm-1.2
ln -fs /home/tom/prog/git/Surelog/build/../third_party/UVM/vmm-1.1.1a vmm-1.1.1a
rm -rf slpp_all slpp_unit
cp -f /home/tom/prog/git/Surelog/build/../src/API/slSV3_1aPythonListener.py /home/tom/prog/git/Surelog/build/dist/Release/python/
cp -f /home/tom/prog/git/Surelog/build/../src/API/slformatmsg.py /home/tom/prog/git/Surelog/build/dist/Release/python/slformatmsg.py
cp -f /home/tom/prog/git/Surelog/build/../src/API/slwaivers.py /home/tom/prog/git/Surelog/build/dist/Release/python/
echo \ \ \ \ \ \ \ Creating\ OVM\ precompiled\ package...
       Creating OVM precompiled package...
/home/tom/prog/git/Surelog/build/dist/Release//surelog -nobuiltin -createcache +incdir+ovm-2.1.2/src/ +incdir+vmm-1.1.1a/sv ovm-2.1.2/src/ovm_pkg.sv -writepp -mt 0 -parse -nocomp -noelab -nostdout
munmap_chunk(): invalid pointer
make[3]: *** [CMakeFiles/surelog-bin.dir/build.make:99: dist/Release/surelog] Aborted (core dumped)
make[3]: *** Deleting file 'dist/Release/surelog'
make[3]: Leaving directory '/home/tom/prog/git/Surelog/build'
make[2]: *** [CMakeFiles/Makefile2:115: CMakeFiles/surelog-bin.dir/all] Error 2
make[2]: Leaving directory '/home/tom/prog/git/Surelog/build'
make[1]: *** [Makefile:163: all] Error 2
make[1]: Leaving directory '/home/tom/prog/git/Surelog/build'
make: *** [Makefile:6: release] Error 2
```

No more ltcmalloc library, a little bit different message. But crash at the same place in surelog:
```
$ gdb -q --args ./build/dist/Release/hellosureworld dbuild/test.sv 
Reading symbols from ./build/dist/Release/hellosureworld...
(No debugging symbols found in ./build/dist/Release/hellosureworld)
(gdb) r
Starting program: /home/tom/prog/git/Surelog/build/dist/Release/hellosureworld dbuild/test.sv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[INFO :CM0023] Creating log file ./slpp_all/surelog.log.

munmap_chunk(): invalid pointer

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7582899 in __GI_abort () at abort.c:79
#2  0x00007ffff75ed38e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff77163a5 "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff75f54dc in malloc_printerr (str=str@entry=0x7ffff7717c10 "munmap_chunk(): invalid pointer") at malloc.c:5332
#4  0x00007ffff75f572c in munmap_chunk (p=<optimized out>) at malloc.c:2830
#5  0x000055555593c610 in SURELOG::Compiler::createFileList_() [clone .cold] ()
#6  0x00005555559d1692 in SURELOG::Compiler::compile() ()
#7  0x00005555559ac906 in SURELOG::start_compiler(SURELOG::CommandLineParser*) ()
#8  0x00005555559937ee in main ()
```

Thomasb81, 2019-11-23:
Got it !

There is a missing return value for this function definition :+1: 
https://github.com/alainmarcel/Surelog/blob/65fd78c13d99b2458421a51929e88a01303f44a1/src/SourceCompile/Compiler.cpp#L234

I add `return true;` at the end of the method and the issue seems gone.

alainmarcel, 2019-11-23:
Thanks! I'm fixing it and checkin in a few minutes.
I was missing -Wall in the cmake file which is why this got unnoticed. I also fixed that.
I fixed the Python problem too

Thomasb81, 2019-11-23:
Working for me.

PR #110 (2019-11-23): Guard for misuse of python API, fix example that had a misuse, add -DNO_TCMALLOC in case of tcmalloc issues
Fixes #108, Fixes #109
closed 2019-11-23 by Thomasb81