chipsalliance/Surelog #109: Crash when recursing visiting instance

Thomasb81, 2019-11-23:
Here attach a sv module pick-up from Ieee1800-2017 where recursive visit of children crash.

> $ gdb -q --args ./dist/Debug/surelog -nostdout -parse -pythonevalscript issue_scriptPerDesign.py test.sv
> Reading symbols from ./dist/Debug/surelog...
> (gdb) r
> Starting program: /home/tom/prog/git/Surelog/dbuild/dist/Debug/surelog -nostdout -parse -pythonevalscript issue_scriptPerDesign.py test.sv
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> 1
> work@top work@top
> B1[0] work@top.B1
> N1 work@M1
> 
> Program received signal SIGSEGV, Segmentation fault.
> 0x0000555555e39a1b in SURELOG::ModuleInstance::getChildren (this=0x555557560380, i=1) at /home/tom/prog/git/Surelog/src/Design/ModuleInstance.h:44
> 44	  ModuleInstance* getChildren(unsigned int i) { return m_childrens[i]; }
> (gdb) bt
> #0  0x0000555555e39a1b in SURELOG::ModuleInstance::getChildren (this=0x555557560380, i=1) at /home/tom/prog/git/Surelog/src/Design/ModuleInstance.h:44
> #1  0x0000555555e3979f in SURELOG::SLgetInstanceChildren (instance=0x555557560380, i=1) at /home/tom/prog/git/Surelog/src/API/SLAPI.cpp:664
> #2  0x0000555555e26eee in _wrap_SLgetInstanceChildren (args=0x7ffff6f2b5f0) at /home/tom/prog/git/Surelog/src/API/slapi_wrap.cxx:10953
> #3  0x00007ffff7d25169 in _PyMethodDef_RawFastCallKeywords () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #4  0x00007ffff7d25ac8 in _PyCFunction_FastCallKeywords () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #5  0x00007ffff7b36250 in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #6  0x00007ffff7b3807e in _PyEval_EvalFrameDefault () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #7  0x00007ffff7b3f94b in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #8  0x00007ffff7b36015 in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #9  0x00007ffff7b3807e in _PyEval_EvalFrameDefault () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #10 0x00007ffff7b3f94b in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #11 0x00007ffff7b36015 in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #12 0x00007ffff7b3807e in _PyEval_EvalFrameDefault () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #13 0x00007ffff7b3f94b in ?? () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #14 0x00007ffff7d259b8 in _PyFunction_FastCallDict () from /lib/x86_64-linux-gnu/libpython3.7m.so.1.0
> #15 0x0000555555e29ea8 in SURELOG::PythonAPI::evalScript (script="issue_scriptPerDesign.py", design=0x5555575581a0) at /home/tom/prog/git/Surelog/src/API/PythonAPI.cpp:389
> #16 0x0000555555ea1a30 in SURELOG::Compiler::compile (this=0x555557570140) at /home/tom/prog/git/Surelog/src/SourceCompile/Compiler.cpp:681
> #17 0x0000555555e3cdf7 in SURELOG::start_compiler (clp=0x555557636640) at /home/tom/prog/git/Surelog/src/API/Surelog.cpp:16
> #18 0x0000555555e0d6ad in executeCompilation (argc=6, argv=0x7fffffffe018, diff_comp_mode=false, fileunit=false) at /home/tom/prog/git/Surelog/src/main.cpp:43
> #19 0x0000555555e0da90 in main (argc=6, argv=0x7fffffffe018) at /home/tom/prog/git/Surelog/src/main.cpp:132
> (gdb)

[testcase.zip](https://github.com/alainmarcel/Surelog/files/3882710/testcase.zip)

Thomasb81, 2019-11-23:
I had few print in the python file:
```
LregisterNewErrorType("[WARN :PY0099]", "Instance \"%s\"", "");
SLoverrideSeverity("[WARN :PY0099]", "WARNING")

def recurseInstance(errors, design, instance):
    print ("enter instance",instance)
    mod = SLgetInstanceDefinition(instance)
    SLaddError(errors, "[WARN :PY0099]", SLgetInstanceFileName(instance), SLgetInstanceLine(instance), 0, SLgetInstanceName(instance))  
    nbdefs = SLgetnInstanceChildren(instance)
    for i in range(nbdefs):
        print(i,nbdefs,instance)
        instance = SLgetInstanceChildren(instance,i)
        recurseInstance(errors, design, instance)
    print ("exit instance",instance)

def slUserCallbackPerDesign (errors, design):
    nbdefs = SLgetnTopModuleInstance(design)
    for i in range(nbdefs):
        instance = SLgetTopModuleInstance(design,i)
        recurseInstance(errors, design, instance)
```
The execution trace look like:
```
gdb -q --args dbuild/dist/Debug/surelog -nostdout -parse -pythonevalscript dbuild/issue_scriptPerDesign.py dbuild/test.sv
Reading symbols from dbuild/dist/Debug/surelog...
(gdb) r
Starting program: /home/tom/prog/git/Surelog/dbuild/dist/Debug/surelog -nostdout -parse -pythonevalscript dbuild/issue_scriptPerDesign.py dbuild/test.sv
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
enter instance <Swig Object of type 'unknown' at 0x7ffff70d43c0>
0 2 <Swig Object of type 'unknown' at 0x7ffff70d43c0>
enter instance <Swig Object of type 'unknown' at 0x7ffff6dcf660>
0 3 <Swig Object of type 'unknown' at 0x7ffff6dcf660>
enter instance <Swig Object of type 'unknown' at 0x7ffff6dcf750>
exit instance <Swig Object of type 'unknown' at 0x7ffff6dcf750>
1 3 <Swig Object of type 'unknown' at 0x7ffff6dcf750>

Program received signal SIGSEGV, Segmentation fault.
0x0000555555d2346b in SURELOG::ModuleInstance::getChildren (this=0x5555573a0ca0, i=1) at /home/tom/prog/git/Surelog/src/Design/ModuleInstance.h:44
44	  ModuleInstance* getChildren(unsigned int i) { return m_childrens[i]; }
```

I observe that has soon I am going deep the address of the instance object is changing until reach one leaf. But come back from one level to process next branch the address of the object did not revert .

In the previous trace the last line is:
`1 3 <Swig Object of type 'unknown' at 0x7ffff6dcf750>`
while I would have expect:
`1 3 <Swig Object of type 'unknown' at 0x7ffff6dcf660>`

Somehow we are not returning to the correct instance...

Thomasb81, 2019-11-23:
There was an issue in the example script. fix now

Thomasb81, 2019-11-23:
To fast at closing.
https://github.com/alainmarcel/Surelog/blob/65fd78c13d99b2458421a51929e88a01303f44a1/tests/UnitPython/myscriptPerDesign.py#L11
code
```
        instance = SLgetInstanceChildren(instance,i)
        recurseInstance(errors, design, instance)
```
should be fix as:
        subInstance = SLgetInstanceChildren(instance,i)
        recurseInstance(errors, design, subInstance)

alainmarcel, 2019-11-23:
The example was bogus, please fix your copy to reflect the fix in the example:
.....
def recurseInstance(errors, design, instance):
    mod = SLgetInstanceDefinition(instance)
    print (SLgetInstanceName(instance), SLgetModuleName(mod))
    SLaddError(errors, "[WARN :PY0099]", SLgetInstanceFileName(instance), SLgetInstanceLine(instance), 0, SLgetInstanceName(instance))  
    nbdefs = SLgetnInstanceChildren(instance)
    for i in range(nbdefs):
        subinstance = SLgetInstanceChildren(instance,i)
        recurseInstance(errors, design, subinstance)
.....

alainmarcel, 2019-11-23:
Fixed by #110

PR #110 (2019-11-23): Guard for misuse of python API, fix example that had a misuse, add -DNO_TCMALLOC in case of tcmalloc issues
Fixes #108, Fixes #109
closed 2019-11-23 by alainmarcel