chipsalliance/Surelog #111: Question File API

Thomasb81, 2019-11-23:
Hello

Let say I have the following files:
pkg.sv:
```
package myPkg;
`include "myClassDefParent.svh"
`ifdef MY_CLASS
`include "myClassDef.svh"
`endif
endpackage
```

myClassDef.svh
```
class myClassDef : extend myClassDefParent;
endclass
```
myClassDefParent.svh
```
class myClassDefParent;
endclass
```

Would it be possible to query the API about myPkg object to retrieve the fileSet and the used macro that have been used to build the integrity of the object ?
It seems to me that no getter can do that.

alainmarcel, 2019-11-24:
Not thru the FileContent API,
There is no public API for that. The FileContent API accesses post preprocessor data. Serializing non preprocessed data seems feasible too.
2 ways to get to what you want through: create a specialized listener that listen to the preprocessing event s of the ANTLR AST, but you would have to reinvent a lot of code that is in the preprocessor class.
Or add a callback mechanism in the preprocessor code, so you get notified of the events you describe above.
If you have some time, I could guide you and would welcome your code contribution

Thomasb81, 2019-11-25:
Thanks for the answer.
So the idea would be to implement callback on preprocessor parser  similar to callback on listener of the main grammar ?

Time is like money, we can use it one time only :)

It is a tedious work to learn code of this kind of project (huge standard, huge code base, no code kickoff document  to efficiently dive in to the code ...  ) 
I already have a foot on hdlConvertor, which target more or less the same purpose, on which I start 2 year ago... Both project have strength and weakness.

alainmarcel, 2019-12-05:
I'm working on that, will be available soon. I had several inquires on this topic.

alainmarcel, 2019-12-09:
The C++ API is ready to support this kind of queries thru the Design and FileContent classes. 
The Python API can be added effortlessly.
Use hellosureworld.cpp as a starting point and query the Design class with:
Design::getAllPPFileContents()

closed 2019-12-09 by alainmarcel