chipsalliance/Surelog #135: compile error, please help

chenfengrugao, 2019-12-27:
Ｉ am compiling Surelog, but encounter the following error. I compile with devtoolset-8(gcc8) on Centos7. What should I do?

/home/billchen/work4/Surelog/src/Utils/StringUtils.cpp: In static member function ‘static void SURELOG::StringUtils::autoExpandEnvironmentVariables(std::string&)’:
/home/billchen/work4/Surelog/src/Utils/StringUtils.cpp:357:54: error: no matching function for call to ‘std::basic_string<char>::replace(const __gnu_cxx::__normal_iterator<const char*, std::basic_string<char> >&, const __gnu_cxx::__normal_iterator<const char*, std::basic_string<char> >&, std::string&)’
     text.replace(match[0].first, match[0].second, var);
                                                      ^
In file included from /opt/rh/devtoolset-8/root/usr/include/c++/8/string:52,
                 from /home/billchen/work4/Surelog/src/Utils/StringUtils.h:26,
                 from /home/billchen/work4/Surelog/src/Utils/StringUtils.cpp:24:
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/basic_string.h:4702:7: note: candidate: ‘std::basic_string<_CharT, _Traits, _Alloc>& std::basic_string<_CharT, _Traits, _Alloc>::replace(std::basic_string<_CharT, _Traits, _Alloc>::size_type, std::basic_string<_CharT, _Traits, _Alloc>::size_type, const std::basic_string<_CharT, _Traits, _Alloc>&) [with _CharT = char; _Traits = std::char_traits<char>; _Alloc = std::allocator<char>; std::basic_string<_CharT, _Traits, _Alloc>::size_type = long unsigned int]’
       replace(size_type __pos, size_type __n, const basic_string& __str)
       ^~~~~~~

alainmarcel, 2019-12-29:
Should be fixed now, pull the latest.

alainmarcel, 2019-12-31:
Please reopen if still an issue

closed 2019-12-31 by alainmarcel