chipsalliance/Surelog #161: Errror report: line numbers are off

hzeller, 2020-02-07:
Example: running the `earlgrey` test in https://github.com/SymbiFlow/sv-tests (need to run locally and first apply https://github.com/alainmarcel/UHDM/pull/119 ) there are error messages such as

```
[SYNTX:PA0207] third_party/cores/opentitan/hw/ip/prim/rtl/prim_assert.sv:39 Syntax error: no viable alternative at input 'InSelRange_A: assume property (@(posedge clk_i) disable iff (!rst_ni !== '0) (reg2hw.periph_insel[k].q < pinmux_reg_pkg::NMioPads + 2))         \n     else',
     else begin 
     ^-- ./slpp_all/work/third_party/cores/opentitan/hw/ip/pinmux/rtl/pinmux.sv:64 col:5.
```

... however, there is something different in prim_assert.sv:39 and pinmux.sv:64.

Maybe some counting skipping due to preprocess expansion ?

hzeller, 2020-02-08:
first investigation: yes, it looks like it points to the preprocessed file (this is why it points to slpp_all/). So what should happen of course is that it points to the line in the original `pinmux.sv`.

It looks like, right now the preprocessing does not emit
```
`line 123 "orig-filename.sv"
```
lines, so the parsing stage later has no idea what the original location was.

alainmarcel, 2020-02-08:
Both locations are reported, the first line points to the original file, the 2nd line points to the preprocessed file in sll_pp. It is a feature, not a bug :-)

hzeller, 2020-02-08:
Problem is that the relevant point, the original location in pinmux.sv is not revealed, so it is not possible to pinpoint where the problem starts.

So expectation would be something like
```
   syntax error ... pinmux.sv:<original location>  problem witth else
     ... epxanded from foo.sv:123 `ASSERT..()
```

alainmarcel, 2020-02-08:
Do we have the proper Verilog compilation command line for this test or is it still made up from a find in the directory?

alainmarcel, 2020-02-08:
The errors are pointing exactly where they are supposed to point (exact line):
The macro `define ASSERT_RPT(__name, __msg)                                         \
`ifdef UVM                                              \    <------------------------- Pointing here (after the carriage return which is the macro definition)
  assert_rpt_pkg::assert_rpt($sformatf("[%m] %s: %s (%s:%0d)",            \
                             __name, __msg, `__FILE__, `__LINE__));       \
`else                                                                     \
  $error("[ASSERT FAILED] [%m] %s: %s", __name, __msg);                   \
`endif

In the post processed file they point to the post processed macro, which is exactly what you need so you see the code that is not parsed properly:

 OutSelRange_A: assume property (@(posedge clk_i) disable iff (!rst_ni !== '0) (reg2hw.mio_outsel[k].q < pinmux_reg_pkg::NPeriphOut + 3))  
     else begin   <-------------------------------------------------- Pointing here
                                                                     
  $error("[ASSERT FAILED] [%m] %s: %s", "OutSelRange_A", "reg2hw.mio_outsel[k].q < pinmux_reg_pkg::NPeriphOut + 3");                   

 end 

I could find that way that I had an incomplete rule in the parser for assume property that I fixed.

The project file in sv-test for that design is incorrect too, it currently yields tons of undefine macro warnings. It misses the UVM package and UVM include dir and the file prim_assert.v should be parsed first, resulting in a command line like:

 -parse -verbose -noelab -timescale=1ns/1ns +incdir+.+../../UVM/uvm-1.2/src/ ../../UVM/uvm-1.2/src/uvm_pkg.sv
hw/vendor/lowrisc_ibex/shared/rtl/prim_assert.sv hw/ip/alert_handler/rtl/alert_handler_esc_timer.sv hw/ip/alert_handler/rtl/alert_handler_accu.sv hw/ip/alert_handler/rtl/alert_pkg.sv hw/ip/alert_handler/rtl/alert_handler_reg_wrap.sv hw/ip/alert_handler/rtl/alert_handler_ping_timer.sv hw/ip/alert_handler/rtl/alert_handler_class.sv hw/ip/alert_handler/rtl/alert_handler.sv hw/ip/pinmux/rtl/pinmux.sv hw/dv/sv/common_ifs/pins_if.sv hw/vendor/lowrisc_ibex/rtl/ibex_ex_block.sv hw/vendor/lowrisc_ibex/rtl/ibex_compressed_decoder.sv hw/vendor/lowrisc_ibex/rtl/ibex_multdiv_fast.sv hw/vendor/lowrisc_ibex/rtl/ibex_decoder.sv hw/vendor/lowrisc_ibex/rtl/ibex_register_file_ff.sv hw/vendor/lowrisc_ibex/rtl/ibex_core.sv hw/vendor/lowrisc_ibex/rtl/ibex_load_store_unit.sv hw/vendor/lowrisc_ibex/rtl/ibex_pmp.sv hw/vendor/lowrisc_ibex/rtl/ibex_fetch_fifo.sv hw/vendor/lowrisc_ibex/rtl/ibex_pkg.sv hw/vendor/lowrisc_ibex/rtl/ibex_cs_registers.sv hw/vendor/lowrisc_ibex/rtl/ibex_if_stage.sv hw/vendor/lowrisc_ibex/rtl/ibex_alu.sv hw/vendor/lowrisc_ibex/rtl/ibex_prefetch_buffer.sv hw/vendor/lowrisc_ibex/rtl/ibex_id_stage.sv hw/vendor/lowrisc_ibex/rtl/ibex_multdiv_slow.sv hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv hw/ip/prim_generic/rtl/prim_generic_clock_mux2.sv hw/ip/tlul/rtl/tlul_adapter_reg.sv hw/ip/tlul/rtl/tlul_socket_1n.sv hw/ip/tlul/rtl/tlul_err_resp.sv hw/ip/tlul/rtl/tlul_assert.sv hw/ip/tlul/rtl/tlul_fifo_async.sv hw/ip/tlul/rtl/tlul_assert_multiple.sv hw/ip/tlul/rtl/tlul_err.sv hw/ip/tlul/rtl/tlul_fifo_sync.sv hw/ip/tlul/rtl/tlul_adapter_sram.sv hw/ip/flash_ctrl/rtl/flash_ctrl_pkg.sv hw/ip/prim_generic/rtl/prim_generic_pad_wrapper.sv hw/top_earlgrey/ip/alert_handler/rtl/autogen/alert_handler_reg_pkg.sv hw/top_earlgrey/ip/alert_handler/rtl/autogen/alert_handler_reg_top.sv hw/top_earlgrey/ip/rv_plic/rtl/autogen/rv_plic.sv hw/top_earlgrey/ip/rv_plic/rtl/autogen/rv_plic_reg_pkg.sv hw/top_earlgrey/ip/rv_plic/rtl/autogen/rv_plic_reg_top.sv hw/top_earlgrey/ip/pinmux/rtl/autogen/pinmux_reg_pkg.sv hw/top_earlgrey/ip/pinmux/rtl/autogen/pinmux_reg_top.sv hw/top_earlgrey/ip/xbar_peri/rtl/autogen/xbar_peri.sv hw/top_earlgrey/ip/xbar_peri/rtl/autogen/tl_peri_pkg.sv hw/top_earlgrey/ip/xbar_main/rtl/autogen/xbar_main.sv hw/top_earlgrey/ip/xbar_main/rtl/autogen/tl_main_pkg.sv hw/top_earlgrey/rtl/autogen/top_earlgrey.sv hw/top_earlgrey/rtl/padctl.sv hw/ip/rv_timer/rtl/timer_core.sv hw/ip/rv_timer/rtl/rv_timer_reg_top.sv hw/ip/rv_timer/rtl/rv_timer_reg_pkg.sv hw/ip/rv_timer/rtl/rv_timer.sv hw/ip/nmi_gen/rtl/nmi_gen_reg_top.sv hw/ip/nmi_gen/rtl/nmi_gen_reg_pkg.sv hw/ip/nmi_gen/rtl/nmi_gen.sv hw/ip/prim/abstract/prim_clock_mux2.sv hw/ip/rv_dm/rtl/tlul_adapter_host.sv hw/ip/rv_dm/rtl/rv_dm.sv hw/top_earlgrey/rtl/top_pkg.sv hw/ip/flash_ctrl/rtl/flash_mp.sv hw/ip/flash_ctrl/rtl/flash_phy.sv hw/ip/flash_ctrl/rtl/flash_ctrl.sv hw/ip/flash_ctrl/rtl/flash_ctrl_reg_top.sv hw/ip/flash_ctrl/rtl/flash_prog_ctrl.sv hw/ip/flash_ctrl/rtl/flash_erase_ctrl.sv hw/ip/flash_ctrl/rtl/flash_ctrl_reg_pkg.sv hw/ip/flash_ctrl/rtl/flash_rd_ctrl.sv hw/ip/prim/abstract/prim_rom.sv hw/ip/prim_xilinx/rtl/prim_xilinx_ram_2p.sv hw/vendor/pulp_riscv_dbg/src/dm_mem.sv hw/vendor/pulp_riscv_dbg/src/dm_pkg.sv hw/vendor/pulp_riscv_dbg/src/dmi_jtag.sv hw/vendor/pulp_riscv_dbg/src/dmi_cdc.sv hw/vendor/pulp_riscv_dbg/src/dmi_jtag_tap.sv hw/vendor/pulp_riscv_dbg/src/dm_csrs.sv hw/vendor/pulp_riscv_dbg/src/dm_sba.sv hw/vendor/pulp_riscv_dbg/debug_rom/debug_rom.sv hw/ip/usbdev/rtl/usbdev_flop_2syncpulse.sv hw/ip/usbdev/rtl/usbdev_reg_pkg.sv hw/ip/usbdev/rtl/usbdev_reg_top.sv hw/ip/usbdev/rtl/usbdev_linkstate.sv hw/ip/usbdev/rtl/usbdev.sv hw/ip/usbdev/rtl/usbdev_iomux.sv hw/ip/usbdev/rtl/usbdev_usbif.sv hw/ip/usb_fs_nb_pe/rtl/usb_fs_tx_mux.sv hw/ip/usb_fs_nb_pe/rtl/usb_consts_pkg.sv hw/ip/usb_fs_nb_pe/rtl/usb_fs_nb_pe.sv hw/ip/u [... cut ...] n +incdir+hw/top_earlgrey/rtl +incdir+hw/ip/rv_timer/rtl +incdir+hw/ip/nmi_gen/rtl +incdir+hw/ip/prim/abstract +incdir+hw/ip/rv_dm/rtl +incdir+hw/ip/prim_xilinx/rtl +incdir+hw/vendor/pulp_riscv_dbg/src +incdir+hw/vendor/pulp_riscv_dbg/debug_rom +incdir+hw/ip/usbdev/rtl +incdir+hw/ip/usb_fs_nb_pe/rtl +incdir+hw/ip/prim/rtl +incdir+hw/ip/uart/rtl +incdir+hw/ip/rv_plic/rtl +incdir+hw/ip/rv_core_ibex/rtl +incdir+hw/ip/aes/rtl +incdir+hw/ip/hmac/rtl +incdir+hw/ip/gpio/rtl +incdir+hw/ip/spi_device/rtl

hzeller, 2020-02-08:
I know that technically they point to the place which causes the problem, but the line-number is post-preprocessing and shows the post-preprocessing file; it is not something the user can deal with as they will not be able to modify that file (it is neat to _also_ look at the post-processed file, but that is usually hard to navigate as the user needs references to the original file).

So this is why I am suggesting to report problems in macros the same way C-compilers do it: they say the line number where things happen, and say that it happens in macor XYZ and point to the original location where that problem is happening; all pointing to the original file where they then are able to modify it.

Consider the following example from a c-compiler
```
#define FOO(x) (x) else

int main() {
	FOO(32);
}
```

The compiler identifies the problem, and points to the original macro definition:

```
$ gcc -o foo foo.c 
foo.c: In function ‘main’:
foo.c:2:20: error: expected ‘;’ before ‘else’
 #define FOO(x) (x) else
                    ^~~~
foo.c:5:2: note: in expansion of macro ‘FOO’
  FOO(32);
  ^~~
```

Internally, the compilers can do so , as the preprocessor emits `#line` numbers pointing to the original location; the backtick-line in Verilog are meant for that as well. That way, they find the place to where to fix things (because they of course can't modify the preprocessed intermediate file).

The sv-test is probably not correct, can you send a pull request there ?

alainmarcel, 2020-02-08:
- The problem with Verilog is that there are many levels of macros invoking macros (UVM paradigm). The C style reporting doesn't help, you need to point to the actual unraveled post pre-processor code to make any sense of it. 
I'll keep is issue open anyhow until we come to an agreement, on how to proceed, a 3 lines error reporting might be needed. The post-processor report is a must for the tool developer at least. BTW HW engineers are used to look at the post-preprocessed code from Simulator.
 
- I don't know how sv-test is working, I'm not the best person to fix it. I opened an issue in sv-tests.

hzeller, 2020-02-08:
mmh, can't we unravel multiple expansion of expansions similarly ? XXX, exapanded from YYY ... etc.

```C
#define BAR(y) (y) else
#define FOO(x) if (x) ; BAR(x)

int main() {
	FOO(32);
}
```

```
foo.c: In function ‘main’:
foo.c:2:20: error: expected ‘;’ before ‘else’
 #define BAR(y) (y) else
                    ^~~~
foo.c:3:25: note: in expansion of macro ‘BAR’
 #define FOO(x) if (x) ; BAR(x)
                         ^~~
foo.c:6:2: note: in expansion of macro ‘FOO’
  FOO(32);
  ^~~
```
... then each error following step in emacs (or whatever IDE) will allow to point to the original.

Even if hardware engineers are used to look at the preprocessed content now, I think aiming for improved error reporting will help moving the state of the art forward. I also looked at proprocessed C-files in the 80ies because reporting was worse than today (but at least it did the `#line` annotation, so it was possible to jump to the place), but these days I appreciate that the compiler improved.

To make things less confusing for now, we should for now at least add the phrase "in the preprocessed file" or something when we point to `./slpp_all/`.

alaindargelas, 2020-05-15:
I found a more basic problem, I simply forgot to increment the line number for sections that the preprocessor removes:

line 1:    module blah
`ifdef NOT_DEFINED
...text...
`endif
assign foo = bar ; <<< The line number of the following code is 2 instead of being 5

alaindargelas, 2020-05-15:
That needs to be fixed urgently