chipsalliance/Surelog #177: 'make install' fails

hzeller, 2020-02-12:
Currently, 'make install' fails; it seems to not be able to find the generated pkg ?

It does seem to generate the packages:

```
make
#... all kinds of stuff
[ 98%] Building CXX object CMakeFiles/hellosureworld.dir/src/hellosureworld.cpp.o
[100%] Linking CXX executable dist/Release/surelog
       Creating staging for precompiled packages
       Creating OVM precompiled package...
       Creating UVM precompiled package...
       Packages created
make[3]: Leaving directory '/home/hzeller/src/Surelog/build'
# ...
```

But then this happens
```
sudo make install
# ...
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/bin/surelog
-- Installing: /usr/local/lib/surelog/libsurelog.a
-- Installing: /usr/local/include/surelog/surelog.h
-- Up-to-date: /usr/local/lib/surelog/python
-- Installing: /usr/local/lib/surelog/python/slSV3_1aPythonListener.py
-- Installing: /usr/local/lib/surelog/python/slwaivers.py
-- Installing: /usr/local/lib/surelog/python/slformatmsg.py
-- Up-to-date: /usr/local/lib/surelog/sv
-- Installing: /usr/local/lib/surelog/sv/builtin.sv
CMake Error at cmake_install.cmake:65 (file):
  file INSTALL cannot find
  "/home/hzeller/src/Surelog/build/dist/Release/pkg".

make[1]: *** [Makefile:74: install] Error 1
```
This is what is in build/dist/Release

```
$ find build/dist/Release/
build/dist/Release/
build/dist/Release/hellosureworld
build/dist/Release/uhdm-test2
build/dist/Release/uhdm-test1
build/dist/Release/python
build/dist/Release/python/slSV3_1aPythonListener.py
build/dist/Release/python/slwaivers.py
build/dist/Release/python/slformatmsg.py
build/dist/Release/sv
build/dist/Release/sv/builtin.sv
build/dist/Release/uhdm-test3
build/dist/Release/uhdm-dump
build/dist/Release/uhdm-listener
build/dist/Release/surelog
build/dist/Release/uhdm-test4
```

These are the only places where directories named `pkg/` exist:
```
$ find . -name "pkg"
./third_party/UVM/ovm-2.1.2/examples/basic_examples/pkg
./third_party/UVM/uvm-1.2/examples/simple/basic_examples/pkg
./third_party/UVM/uvm-1.1d/examples/simple/basic_examples/pkg
```

alainmarcel, 2020-02-13:
This was working fine until some recent change I guess.
The .travis contains all these steps that make sure the install is correct:
  - make release
  - sudo make install
  - make test_install
  - make test

alainmarcel, 2020-02-13:
This is actually working fine for me locally and in travis.
The only explanation is that the packages are not built in your local branch.
Pull the latest as I made changes to build the latest UVM package without error.

hzeller, 2020-02-13:
I have looked further what is happening on my machine: From earlier experiments, Surelog was installed in /usr/local/, so the surelog call in the build context was finding the precompiled version in /usr/local/lib/surelog/pkg and considered itself to be done and so didn't actually generate the pkg file.

The call is with `-nobuiltin` but apparently this does not prevent from looking at the pre-installed package file.

alainmarcel, 2020-02-13:
I see, you have to de-install to get the packages built

hzeller, 2020-02-13:
Yes, did de-install now and it workd.
For this to work, we essentially have to either make a `-nopkg` flag or make `-nobuiltin` don't use the installed packages.

hzeller, 2021-08-30:
Is anyone working on this ? It makes consistent compiling hard, because if there is already an installed version of `ovm_pkg.sv.slpa` or `uvm_pkg.sv.slpa`, the generation of these files will not happen in the build, thus failing to produce the artifacts needed for a new build. i.e. it is impossible to build and install a surelog on a machine that already has some surelog installed.