chipsalliance/Surelog #203: Surelog debug and release build don't coexist anymore

alainmarcel, 2020-03-05:
Hi Henner,
one of your last changes in the CMakeFile makes it almost impossible to have both 
- make 
and 
- make debug

work at the same time. This used to work! :-)

You either get to do complete rebuilds or you get undefined symbols.
Also, it seems not all sub libraries build coherently (The debug version does not seem to have UHDM in debug mode).

alainmarcel, 2020-03-05:
Example of thunk issue, typically happening when components don't get compiled with the same options:

[100%] Linking CXX executable dist/Debug/hellosureworld
libsurelog.a(SV3_1aLexer.cpp.o):(.rodata._ZTV11SV3_1aLexer[_ZTV11SV3_1aLexer]+0x210): undefined reference to `non-virtual thunk to antlr4::Lexer::getSourceName()'
libsurelog.a(SV3_1aPpLexer.cpp.o):(.rodata._ZTV13SV3_1aPpLexer[_ZTV13SV3_1aPpLexer]+0x210): undefined reference to `non-virtual thunk to antlr4::Lexer::getSourceName()'
collect2: error: ld returned 1 exit status
libsurelog.a(SV3_1aLexer.cpp.o):(.rodata._ZTV11SV3_1aLexer[_ZTV11SV3_1aLexer]+0x210): undefined reference to `non-virtual thunk to antlr4::Lexer::getSourceName()'
libsurelog.a(SV3_1aPpLexer.cpp.o):(.rodata._ZTV13SV3_1aPpLexer[_ZTV13SV3_1aPpLexer]+0x210): undefined reference to `non-virtual thunk to antlr4::Lexer::getSourceName()'

alainmarcel, 2020-03-05:
Mmmm it has to do with VSCODE if I build using the CMake built-in target of VSCODE.
If I build manually in the terminal, both builds coexist just fine.

hzeller, 2020-03-05:
Does it try to parse the cmake file? Maybe it has trouble with the third-party dependencies UHDM and capnproto?

alainmarcel, 2020-03-05:
I just switched to VSCODE from Netbeans. VSCODE does analyse the cmake and creates its own build target invoking cmake with many options. I can't find an easy way to make it use the "make debug" command.
I think that target does not generate object files that can be mixed and matched with the regular cmake invocation (make debug).
I hide the VSCODE button that makes builds for now, so I don't accidentally build incrementally,
because that yields linkage errrors systematically if the directory contains a fresh "make debug".
This requires a make clean and rebuild to fix (long...).

alainmarcel, 2020-03-07:
New evidence,
the "make debug" of Surelog (./dbuild) uses the libuhdm.a built in optimized mode (./build).

alainmarcel, 2020-03-14:
Fixed by #214

closed 2020-03-14 by alainmarcel