chipsalliance/Surelog #239: Windows/Mac Support?

hs-apotell, 2020-03-28:
Any plans to support Windows and Mac platforms?

alainmarcel, 2020-03-28:
No plans, but volunteers to add support are welcome!
We have to check the support from the following components used by Surelog first:
antlr
flattbuffers
capnp-proto
google-tests

hs-apotell, 2020-05-11:
All of the above mentioned libraries successfully build on Windows. Running into Surelog specific issues now.

src/API/SLAPI.h - There are two files with the same name, if casing is ignored. Windows is case-insensitive and so one file has to either be renamed or has to move into a subfolder.

SLAPI.h is generic api i.e. non-language specific. Can this be renamed to api.h?

src/API/slapi.h is auto-generated by src/API/embed_python_api.tcl and is very python specific. Can we call this file slpyapi.h or move into subfolder src/API/py/slapi.h? May be even move all the python related files in this folder into the subfolder?

Any other suggestion as to what's an acceptable resolution to move forward?

hs-apotell, 2020-05-11:
Is there a specific version of C++ Surelog supports? Is C++17 and above acceptable?

Reasoning - Some of the posix functions don't have equivalent functions under VC. However, C++17 added some of the functionality that can be used to replace and will work across platforms. For instance, use of file io functionality in src/Utils/FileUtils.cpp.

alaindargelas, 2020-07-15:
Windows support achieved, Thanks @hs-apotell!
Do we need a travis Windows build?

closed 2020-07-15 by alaindargelas