chipsalliance/Surelog #241: The test test_CompileHelper is too sensitive to updates to the UHDM model content filling

alainmarcel, 2020-03-28:
We are going to change the content of the object model on a daily basis. We cann have tests that need updating every time we support.
I am not talking about the model changing here, but simply when we are filling the members of the model with actual data.

rkapuscik, 2020-03-30:
I'm not sure I understand what changes do you mean, do you mean that there are significant changes expected to the Surelog AST? This is the hardest to maintain part, as you noted in #242.

If it is only the UHDM output from Surelog that is changing, I think this falls within the intended purpose of unit tests - to catch those changes and possible regressions. This is normal that the tests need to be updated (and expanded) if the expected result changed, i.e. because we support more argument types. I agree that the tests are not exactly easy to write right now, please let me know of any ideas on how to simplify them.

Or do you mean that current tests are too broad in scope? They do process whole `tf_call` at the moment, but I am unsure if this could be divided further - maybe we could separate testing argument parsing here somehow, but before we do that we would need unit tests for at least the expression builders.
@hzeller - do you have any suggestions on how we should approach this?

alainmarcel, 2020-03-30:
Hi Rafal,since we took a Depth First Approach to development (Not finishing one step first, then starting the next step, instead we push partial features all the way in the flow), we can expect the following:
- The UHDM Data Model is evolving (not too often, but mistakes have to be corrected on existing models) and new models yet have to be created,- Surelog populates the Data Model in a piecemeal fashion, so we can expect quite a few changes in already dumped data as we refine the semantic of the models (For instance a ref_obj to a signal can become an expression tree with a leaf node that is a ref_obj, I believe that is what changed in that instance).
So creating unit tests that are made of hardcoded schema using integer values is going to become unmanageable to maintain very quickly. Almost a N^2 maintenance problem. Imagine creating a large number of testcases that try to model pieces of the datamodel (Using this hard coded method), and each refinement of the data model/or the data populated requires to update a large number of these cases... you got the picture.
Some ideas: - The test should be built from snippet of AST text tree (using a string parser) instead of hardcoded numbers which I predict is going to be unmanageable.- The tests should look for invariant instead of strict string comparion so they don't break anytime the content of the dumped tree slightly change (For instance if intermediate nodes are added, that's OK). 

Alain

rkapuscik, 2020-03-31:
I think this is natural that the tests live and sometimes will need to be updated, and preferably this should be done by the same person that makes the code changes. Of course you are right that we should minimize the work required to change them.

> - The test should be built from snippet of AST text tree (using a string parser) instead of hardcoded numbers which I predict is going to be unmanageable.

I currently have a simple python script that creates brace-enclosed initializer lists for file content initialization. I suppose that this could be adapted to test cases. The question is, do we want to add this type of logic inside unit tests or maybe somewhere else. If you look at e.g. verible, the tree can be constructed using nested initializers [like here](https://github.com/google/verible/blob/master/verilog/formatting/tree_unwrapper_test.cc#L671), which I think is more readable than flat text AST. Maybe something similar could be added to Surelog?

> - The tests should look for invariant instead of strict string comparion so they don't break anytime the content of the dumped tree slightly change (For instance if intermediate nodes are added, that's OK). 

I'm less sure about this one, I think this would relax much of the accuracy that we aim for in unit tests. We want to make sure that the model is correct. Another way would be to offload that to UHDM by adding more advanced initializer functions - then even if the implementation in UHDM changes, our reference object in Surelog will be updated as well. What do you think?

alainmarcel, 2020-04-01:
Waiting for Henner to pitch in.
In the meantime I'm disabling the test which again fails because I updated the data to be more correct.
This test is getting in the way at this time.

closed 2020-07-15 by alaindargelas

hzeller, 2020-07-15:
Yeah, I disabled the test a while ago, but included it in the build to make sure it still compiles.
Will revisit at some point, but low prio right now.