chipsalliance/Surelog #269: Finite state machine

bsp13, 2020-04-10:
We trying to use Surelog  to identity finite state machine. Is there any exmaple walking the generated AST?
Our Goal is to  implement this feature using library only.

alainmarcel, 2020-04-10:
I am in the process of adding support of state machines in UHDM.
Give me a few days to put up a complete example. UHDM dump from Surelog already has always blocks, if else, case, assign statement support. I'll deal with enums shortly

alainmarcel, 2020-04-11:
Can you please attach a typical example(s) of the kind of state machines you are trying to analyze?

bsp13, 2020-04-11:
[data.zip](https://github.com/alainmarcel/Surelog/files/4464298/data.zip)

bsp13, 2020-04-11:
[data.zip](https://github.com/alainmarcel/Surelog/files/4464299/data.zip)

alainmarcel, 2020-04-11:
Thanks, I added the test in my regression: tests/FSMBsp13
I'll analyze what is missing in the uhdm object model. once I'm done you'll be able to query the model to actually analyze the FSM for the properties you are looking for using the UHDM API.
Check the README and the UHDM/README for details on how to use the API (As a library or as a separate executable).
You can already see what is supported in the UHDM model by looking at the text dump:
tests/FSMBsp13/FSMBsp13.log

alainmarcel, 2020-04-14:
@bsp13, FYI, none of these FSM can either compile in Yosys as-is or Simulate in Verilator.
I could transform them to compile in Yosys, but could not find a correct transformation for Verilator.
I'm still working on finishing the support in Surelog/UHDM regardless, so you'll be able to analyze them. But we won't be able to run them through the complete flow.

bsp13, 2020-04-15:
We are trying to elaborate Design "tests/FSP13/dut.v" file using hellosureworld.cpp. When we print out  something like Instance/netlist name, We got output like this "work@dut.F2.UNNAMED". This type of 'UNAMED' output is intentional or remain some developement.

alaindargelas, 2020-04-16:
The Pull request #279 completes the support of all the modeling of your examples in the UHDM.
If you run Surelog with that latest pull request, you'll find all the data in the UHDM data model.
If you run the hellosureworld.cpp on your example, you'll notice it dumps a slpp_all/surelog.uhdm file.
You can write a UHDM executable that will traverse the object model following the VPI standard which is much better than trying to browse the Surelog internal data model and AST.

I'll create a hellouhdm.cpp example that will make it easier for you tomorrow.
It will run Surelog compilation, create the UHDM data model and show you how to navigate the Data Model (Elaborated and non elaborated).

alaindargelas, 2020-04-16:
Surelog creates  unamed scopes for begin end and fork blocks  ("work@dut.F2.UNNAMED").
If you navigate the UHDM data model, you'll find a begin or fork in the model for the same object.

alaindargelas, 2020-04-17:
hellouhdm example is in #283
Please let me know if you need more help to navigate the data model.
One more question, how did you find Surelog? Google search? Some other github?

alaindargelas, 2020-04-17:
You can try:
 hellouhdm tests/FSP13/dut.v -d uhdm 
You'll see both debug output of the uhdm data model (-d uhdm)
and the output of the sample code traversing some pieces of the data model (hellouhdm.cpp)

bsp13, 2020-04-18:
We found Surelog through github using refrence to ANTLR4. We used hellouhdm.cpp for design file (dut.v) and we only got netlist but verilog file has register, wire and input or output which has multiple bit. How can We identityfy multi-bit of 'reg'(vpiReg, vpiRegArray) and 'wire' using Surelog. we also extract  surelog.uhdm using argument (-d uhdm) and also see but we can not find any 'reg' or 'wire' with multiple bit, inpute and output in it.

alaindargelas, 2020-04-18:
One thing to know is that the text output of UHDM does not repeat information. If you look at the UHDM/src/vpi_visitor.cpp file which generates the stdout output, you'll notice that if a node is visited once, it is not visited again. That is because that visitor code is automatically generated, and that is the only way to avoid loops. But when you program manually code that visit the tree as you do, you can visit the same node again coming from different paths, so even if the text output does not show some information, the object mode has more links (up and down the tree) than you can see in the output.

For your particular question, all the information is present, if you look at the sample of the output below, you'll notice that there is left and right range for a port instance and that the low con connects to a register (vpiNetType:48 is vpiReg in vpi_user.h) :

FSMbsp13.log line 10444:
    |vpiPort:
    \_port: (y), line:1                        <---- Port y of FSM2
      |vpiName:y
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (Fsm2Out)             <----  Hign conn connection to dut.Fsm2Out
        |vpiName:Fsm2Out
        |vpiActual:
        \_logic_net: (Fsm2Out), line:8
          |vpiName:Fsm2Out
          |vpiNetType:1
          |vpiLeftRange:                    <------------ Range [2:0]
          \_constant: 
            |INT:2
          |vpiRightRange:
          \_constant: 
            |INT:0
      |vpiLowConn:                       <----------------- Low conn to register y 
      \_ref_obj: 
        |vpiActual:
        \_logic_net: (y), line:4
          |vpiName:y
          |vpiNetType:48                    <--------- vpiNetType 48 == vpiReg

I'll double check why the register range does not print out though. I think that range was not printed before (visited before).

alaindargelas, 2020-04-18:
There is indeed a bug for the vpiReg, the range was not recorded in UHDM, it is fixed with #287, Thanks!

alaindargelas, 2020-04-18:
Also, one more note, the vpiDirection of the port gives the direction, from vpi_user.h:
#define vpiDirection             20   /* direction of port: */
#define vpiInput                   1   /* input */
#define vpiOutput                  2   /* output */
#define vpiInout                   3   /* inout */

alaindargelas, 2020-04-22:
@bsp13, is this working for you? Anymore question? Can I close this issue?

bsp13, 2020-04-23:
We are working on it. We are trying to retrieve state machine transition and assignment form it. Like truth table when perticular state machine execute on which condition. for example in 'fsm3.v' file, state machine trasition "Stop" -> "Slow" executed, when other parameter condition like "!(! keys) && (accelerate) && (Speed == Stop)". Have any information for it how we can find "!(! keys) && (accelerate) && (Speed == Stop)".

alaindargelas, 2020-04-24:
Certainly, you can write VPI code to retrieve the exact statement tree:
Please see in the log file:
  
  |vpiProcess:                           <----- vpi_iterate(vpiAlways, mod)
  \_always: , line:22
    |vpiAlwaysType:1
    |vpiStmt:                                <-------------vpi_get(vpiStmt, always_itr)
    \_event_control: , line:22                     gives you the event control
      |vpiCondition:
      \_operation: 
        |vpiOpType:35
        |vpiOperand:
        \_operation: 
          |vpiOpType:39
          |vpiOperand:
          \_ref_obj: (clock)
            |vpiName:clock
            |vpiParent:
            \_operation: 
        |vpiOperand:
        \_operation: 
          |vpiOpType:40
          |vpiOperand:
          \_ref_obj: (keys)
            |vpiName:keys
            |vpiParent:
            \_operation: 
      |vpiStmt:                                                   <------- vpi_get(vpiStmt, event_control)
      \_named_begin: (FSM3), line:22                              
        |vpiName:FSM3
        |vpiStmt:
        \_if_else: , line:23                                    <------   if (!keys)
          |vpiCondition:                                 <------ if condition
          \_operation:   
            |vpiOpType:3                            <------------- !
            |vpiOperand:
            \_ref_obj: (keys)                          <-------------- keys
              |vpiName:keys
              |vpiParent:
              \_operation: 
          |vpiStmt:                              <-----------      Speed = Stop;
          \_assignment: , line:24
            |vpiBlocking:1
            |vpiLhs:
            \_ref_obj: (Speed)
              |vpiName:Speed
            |vpiRhs:
            \_ref_obj: (Stop)
              |vpiName:Stop
          |vpiElseStmt:

.......

alaindargelas, 2020-04-24:
You can get the VPI relation in between objects from here:
https://github.com/alainmarcel/Surelog/blob/master/third_party/Verilog_Object_Model.pdf
It's an extract of the IEEE Standard which makes it easier to search for relations.

bsp13, 2020-04-26:
Finally, I got this output file after retrieve VPI code and exact statement tree relation  for finite state machine. Now, You can free to close issue.
[image]
[image]
[image]

alaindargelas, 2020-04-27:
Nice !Are you interested in providing your code as an example application of UHDM?Even if it's just the current version? Does not need to be updated. Or even better a link to a live GitHub.

closed 2020-04-27 by alaindargelas