chipsalliance/Surelog #273: Speedup Antlr runtime

alainmarcel, 2020-04-12:
Either contribute back to the Antlr main or create a fork maximized for performance.
the Surelog/third_party/antlr already contains some optimizations where dynamic casting has been replaced by type checking with a class member bitfield.

Relevant discussions with the Antlr community:

https://github.com/antlr/antlr4/issues/2584

https://github.com/antlr/antlr4/issues/1790

https://groups.google.com/forum/#!searchin/antlr-discussion/alainmarcel%7Csort:date

alainmarcel, 2020-04-12:
@hzeller can you share with @octaviis some of your findings when you looked at the Antlr runtime?
@octaviis has some time to dive into it.

alaindargelas, 2020-04-22:
Here are the things I know we can do (Low hanging fruits):

- Profile Surelog grammar some more to find new grammar level optimizations
- Replace Leaf grammar rules used for Lexer by AST 2 Serialization C++ code
- Systematically (by script?) replace all dynamic casting by bit field compare (the trick I already manually applied in some parts of the code to get 2x speedup)

alaindargelas, 2021-10-03:
Nothing happening on this front.

closed 2021-10-03 by alaindargelas