chipsalliance/Surelog #302: Should we use bigger machines for Surelog regression tests on Kokoro

mithro, 2020-04-21:
We have a pool of `16 vCPUs, 104 GB` (`n1-highmem-16`) machines to run Kokoro tasks on. Should we use them for running the tests on instead of the current smaller `4 vCPUs, 15 GB` (`n1-standard-4`) machines.

mithro, 2020-04-21:
@alaindargelas - If it makes sense to use the more powerful machines, rename the topic of this issue and assign it to @mithro so I know to make the change.

alaindargelas, 2020-04-22:
We should run with the multitrhread option on Koroko, we should also compile in parallel (-j 10 or more)

regression.tcl mt=8

alaindargelas, 2020-05-02:
Parallel regression is set to 4 in the makefile as this is the current number of kokoro machines. Move to bigger host and change the -j 4 to the according number of cpu. Only change that number in the makefile.

mithro, 2020-05-03:
@alaindargelas - So the 16vCPUs, 104 GB machine sounds good?

mithro, 2020-05-03:
I have updated the Kokoro configuration to use the mid pool (the `n1-highmem-16` sized machines). It should start being active in about 30mins to 1hour.

mithro, 2020-05-03:
When we see the CI running on the bigger machines we can increase the `-j` level (or maybe we can just use `nproc`?)

mithro, 2020-05-03:
The make file is currently hard coding `-j 4` and `-j 24` -- it should look for a `CPU_CORES` environment variable otherwise use something like `CPU_CORES := $(shell nproc)`.

closed 2020-07-15 by alaindargelas