chipsalliance/Surelog #303: Make kokoro see the test run outputs

mithro, 2020-04-21:
The test runner needs to produce junit style output logs. 

Then they need to be copied to the right location like so (https://github.com/SymbiFlow/prjxray/blob/master/.github/kokoro/tests.sh#L22-L33).
````
echo
echo "========================================"
echo "Copying tests logs"
echo "----------------------------------------"
(
	cat build/*test_results.xml
	mkdir build/py
	cp build/py_test_results.xml build/py/sponge_log.xml
	mkdir build/cpp
	cp build/cpp_test_results.xml build/cpp/sponge_log.xml
)
echo "----------------------------------------"
```

And the `.cfg` file needs to be updated to collect the xml files - see https://github.com/SymbiFlow/prjxray/blob/master/.github/kokoro/tests.cfg#L5-L14
```
action {
  define_artifacts {
    regex: "**/*.log"
    regex: "**/*sponge_log.xml"
    regex: "**/build/**/*.xml"
    regex: "**/build/*.xml"
    regex: "*result*.xml"
    strip_prefix: "github/symbiflow-prjxray-tests/"
  }
}
```

I'm not sure how you make ctest produce compatible output you use the following (https://github.com/SymbiFlow/prjxray/blob/master/Makefile#L54-L55);
```
	cd build && ctest --no-compress-output -T Test -C RelWithDebInfo --output-on-failure
	xsltproc .github/kokoro/ctest2junit.xsl build/Testing/*/Test.xml > build/cpp_test_results.xml
```
See https://github.com/SymbiFlow/prjxray/blob/master/.github/kokoro/ctest2junit.xsl for the xsl script.

mithro, 2020-04-21:
For python scripts, there is a [`import junit_xml` module](https://pypi.org/project/junit-xml/). 

See example usage at https://github.com/SymbiFlow/prjxray/blob/679f49812bebe4ef6189384e3e0dcfb8e7892c4f/fuzzers/run_fuzzer.py#L613-L667
```python
    test_case = junit_xml.TestCase(
        name=fuzzer_name,
        timestamp=time_start.timestamp(),
        elapsed_sec=(time_end - time_start).total_seconds(),
        stdout=success_log,
        stderr=error_log,
    )

    if retcode != 0:
        test_case.add_failure_info(
            'Fuzzer failed with exit code: {}'.format(retcode))

    with open(tsfilename, 'w') as f:
        ts = junit_xml.TestSuite(fuzzer_name, [test_case])
        junit_xml.TestSuite.to_file(f, [ts])

    if retcode != 0:
        test_case.add_failure_info(
            'Fuzzer failed with exit code: {}'.format(retcode), )
```

alaindargelas, 2021-04-11:
Is this issue still relevant?

closed 2021-10-03 by alaindargelas