chipsalliance/Surelog #338: Schematic diagrams

hs-apotell, 2020-04-30:
Hey -

I am attempting to build a schematic diagram visualizer tool for UHDM designs. To keep the scope small, layout of individual components is decided to be out of scope, at least for now. However, I still need to implement logic to identify higher level blocks like FF, Adders, Latch, etc. AFAIK, neither UHDM nor vpi provide such information. Just wondering if there's any work happening along this line or are there any examples of how build this on top of UHDM (or vpi).

Appreciate your continued support.

alaindargelas, 2020-05-01:
Inferencing (The process of identifying FF, Latch, combo logic) is out of the scope of UHDM for now.
And will be most likely for quite some time, as we plan to rely on Yosys (Open source Synthesis) to do the full Synthesis (Inferencing and optimization), using UHDM as the Design parser/elaborator.
See: https://github.com/alainmarcel/uhdm-integration -> Yosys integration (WIP)

That said, writing an Inference engine as a client application of UHDM is welcome as some applications like yours or Design-level Rule checking do not require full Synthesis, but only Inference.
If you are up for the task I would welcome the contribution and would be happy to guide you.

hs-apotell, 2020-05-01:
Are you proposing to extract the logic out of Yosys to build an independent application or build something from scratch? Apart from me, is there an active client/customer looking for such functionality?

Also, is the uhdm-yosys integration actively being worked on? I see some activity in the repository but it wasn't exactly clear whether or not it involved any active development. Is there a timeline for this to be ready?

mithro, 2020-05-01:
FYI @hs-apotell [netlistsvg](https://github.com/nturley/netlistsvg) can generate simple schematic diagrams from code that Yosys can read. We use it as part of the http://sphinxcontrib-verilog-diagrams.rtfd.io/ tooling. When yosys-uhdm integration works, that will include SystemVerilog systems. There is also some discussion around open source tooling to do this at https://github.com/SymbiFlow/ideas/issues/41

alaindargelas, 2020-05-01:
Let's start with some facts, that might help you make a judgment call:
- Surelog-UHDM-Yosys complete flow is probably 6 months out 
  It will Give Yosys full SystemVerilog support for the Synthesizable subset
- Current focus is the
  Surelog-UHDM-Verillator flow for simulation of SystemVerilog
- These priorities might change
- Existing Yosys has schematic generation capability as mentioned by @mithro with a documentation target
- Full fledged interactive schematic viewer (As in EDA tools) is not in scope of the projects I know of.
- Surelog-UHDM would benefit from an inference engine independent of Yosys for Design Analysis capabilies at RTL-level (Pre-Synthesis). I know I would like to have that capability, but not customer is waiting for it.

hs-apotell, 2020-05-02:
> Full fledged interactive schematic viewer (As in EDA tools) is not in scope of the projects I know of.

I assume you are referring to interactive UI here. Is full inference still in scope of these integration? Simulation is going to need that so I guess that would be true. Do you know how far along is Yosys in implementing inference logic.

Wouldn't this task be easier if executed post Yosys integration?

I am looking to focus on non-Unix platform support for Surelog and UHDM in immediate future. Let's hope that these integrations have moved further along so we can take advantage of that work to implement this.

alaindargelas, 2020-05-02:
I was referencing to this, correct:
https://edadirect.com/product/starvision-pro/?gclid=Cj0KCQjwtLT1BRD9ARIsAMH3BtVbfWpt_XDbnpbvZu2pqDJE2VV3PWNF0OgCNX5aNwMVL3ss0eWlb8waAkOfEALw_wcB

If you can wait for the UHDM-Yosys integration that would be the easier path, for sure.
Yosys has very good Inference, but you'll have to check if you can "stop" at the inference stage, and not the post-optimization to retrieve the netlist for the schematic, else you'll be literally looking at Gate-level instead of RTL which is almost undecipherable...
Have you checked if Yosys compiles on these other platforms?

mithro, 2020-05-02:
@alaindargelas - Yosys can be "stopped" as pretty much any stage (well actually it's more like you have to tell Yosys to move forward through the inferencing stages - most of the "synthesis scripts" list the various passes Yosys does in the documentation -> http://www.clifford.at/yosys/cmd_synth.html ).

alaindargelas, 2020-05-02:
@hs-apotell, can you try to play with Yosys scripts as mentioned by @mithro and see what do you get in the output netlist for adders, FFs, latches, state machines.... when you exit after Inference? See if that suits your needs for schematic? If not we can explore some more.

hs-apotell, 2020-05-03:
> Have you checked if Yosys compiles on these other platforms?

@alaindargelas Yes, Yosys does support both Windows & Mac. The download page here - http://www.clifford.at/yosys/download.html - has binaries for all.

> can you try to play with Yosys scripts ...

I fully intend to do that. Will loop back as soon as I have some information to share.

hs-apotell, 2020-07-08:
@alaindargelas @mithro Sent you guys an invitation to an private repository with our attempt at implementing "Inferencing" using Surelog. We have piggy backed on Yosys implementation with the exception of generating the RTLIL nodes. Please take a moment to review and provide feedback. Thanks!

mithro, 2020-07-08:
@hs-apotell Would you be willing to release this under the same license as Surelog / UDHM?

The Yosys + UHDM work is mainly been done by the team at @antmicro -- you can follow along it more closely at https://github.com/antmicro/uhdm-integration/commits/master

mithro, 2020-07-08:
FYI -- @kgugala @kbieganski -- You should probably reach out to @hs-apotell

hs-apotell, 2020-07-08:
> Would you be willing to release this under the same license as Surelog / UDHM?

That's the plan. I didn't want to throw it out in public with unfinished work. The implementation is very much in its infancy and would need time to mature. I just wanted to bring a few other members onboard to ensure we are on the "right" path and not running in opposite direction.

At the moment we have verified the implementation only against 25 test cases and have already identified at least 4 language features that are missing in Surelog parser. We are actively working to bring most, if not all, the test cases currently under Yosys for this repository too. Hopefully, that should give us a bigger/wider picture of the supported features.

@bsp13 @anadishukla Looping in developers who have actively worked on this feature.

mithro, 2020-07-08:
@hs-apotell You should most certainly talk to @kgugala 's team at Antmicro as it sounds like you could potentially be working on very similar things and could potentially combined forces!

Nic30, 2020-08-27:
[d3-hwschematic](https://github.com/Nic30/d3-hwschematic).
I had an interesting discussion with @drom about how we should design the input format and what features we actually want to display in such a schematic. (How to represent collapsible ports, logical blocks in pipelines, components with shapes predefined by some standard etc.). The schematic diagram editor turned to be a thing which takes ridiculous amount of time to finish.

I mean it is easy just to use yosys to extract some archetypes like FF, MUX, RAM... and display them in some existing visualizer. But such a raw conversion usually does not work for non trivial circuits because you will only obtain a sea of nodes which can not even be displayed. In addition you need to somehow search in the graph etc. In addition you can see such a result in Xilinx Vivado etc.

It would be great if we could somehow take existing projects and compile them in to some widely used fully working generic enought solution. Because maintenance of GUI project is not an easy task.

pka88, 2022-08-24:
Hi,

I came up with this thread while searching for a way to have an interactive diagram for analyzing the clock tree from RTL.
In general, the requirement is, the user could select a particular signal and keep tracing forward or backwards based on the RTL design instead of netlist.
Please suggest if such capability already exists in any of the tools.
Can SV source code -> Verible -> Kythe graph representations be used for such a representation instead of the default web UI?

Thanks in advance.

alaindargelas, 2022-08-24:
@hs-apotell, what is the status of this project?

Also,

@pka88, since my last comment in this thread, the Surelog/UHDM/Yosys integration has made big progresses:
https://github.com/antmicro/yosys-systemverilog

And https://github.com/chipsalliance/Surelog/issues/338#issuecomment-622392258 should give you a schematic of sort.

pka88, 2022-08-25:
@alaindargelas, @hs-apotell, thanks for the responses.
The suggested tool gives a full block diagram for the given module as I understand.

I was looking for more of a hierarchical schematic tracer which has options like
1. Startpoint signal
2. Hierarchical depth till which the signal needs to be traced down or up
3. This diagram is the base and it would be interactive so the user can further trace it as per his interest and add/delete nodes as needed.
4. The idea to reduce the scope of interactive diagram to a particular signal or data chain. This would get rid of the problems being mentioned by @Nic30 as part of [d3-hwschematic](https://github.com/Nic30/d3-hwschematic). right?

What would you suggest as the right approach for this? 
1. The Kythe already has intelligence of source code index, can this be leveraged for the diagraming UI? Since this tooling requirement falls under developer code analysis, I was thinking Verible might be the right choice due to performance.
2. Surelog -> UHDM -> Yosys intermediate representation and build the diagraming tool (as suggested in the project description of this thread)

alaindargelas, 2022-08-25:
UHDM will contain the elaborated netlist which I think you need for accurate signal tracing, Verible would only have a pre-elaboration view, potential issues: you'll trace more signals than you need, or you'll have the bus sizes wrong.
If you are not interested in drawing FFs, Latches for which you need the Yosys Inference, then you might want to use the UHDM data model and create the path tracer (Forward/Backward) based on the UHDM elaborated data model.

alaindargelas, 2022-08-25:
I'm suggesting UHDM should be a data server that generates incremental Json files on demand based on the user interaction with the GUI (Whatever the GUI might be).

Nic30, 2022-08-26:
@pka88 
> get rid of the problems being mentioned 

< 10K binary operator nodes displayed at once is ~15s of loading time, > 75K nodes -> out of default max memory size per tab in firefox/chrome

It depends on what you expect.

From what I see in this conversation I got the impression that you are trying to visualize something which is >100K FPGA LUT/ALM . If this is the case the elaboration can take significant time and it may defeat whole purpose of interactive loading.

As alaindargelas mentioned without elaboration there may be potential problems and thing would not be as simple as it seems.
There is also  parser [hdlConvertor](https://github.com/nic30/hdlconvertor) (test reports do contain parsing times and it is build for large designs (>2GB)) however as alaindargelas it UHDM/yosys will be better.