chipsalliance/Surelog #341: UHDM::ref_obj - missing reference?

hs-apotell, 2020-05-02:
Why doesn't UHDM::ref_obj include a reference to the UHDM::any that it refers to?

When walking the tree using UHDM (or vpi), we encounter, say an assignment statement like in the snippet below, The UHDM::assignment includes references to both operands LHS and RHS as UHDM::ref_obj instances. However, the references are merely names and not references to the modules parameters or local variables.

Am I missing something here or is this the expected behavior? If latter, then how are names to be de-referenced to get the UHDM::any reference?

Below is a snippet from log https://github.com/alainmarcel/Surelog/blob/master/tests/FSMBsp13/FSMBsp13.log 

```
          \_assignment: , line:16
            |vpiBlocking:1
            |vpiLhs:
            \_ref_obj: (CurState)  <= Shouldn't these ref_obj(s) also include a reference to the module/local variable/parameter???
              |vpiName:CurState
            |vpiRhs:
            \_ref_obj: (ST_Read)
              |vpiName:ST_Read
```

alaindargelas, 2020-05-02:
So far, the approach we have taken which slightly deviates from the standard (VPI is supposedly fully elaborated model), is to create a 2 part **Folded Model** (With some work for the client application to stitch it together):

1) The **non-elaborated** view that you are pointing to (With roots like |uhdmallModules: for each module)
  does represent the module taken independently and no binding from reference to object is performed
  That representation contains the processes and assignments of the module as you pointed out.
  Binding the nets at the module level is of little value because you need to actually bind the nets in 
  the design to see the complete picture, see  below:

2)  The **elaborated view** (With roots like |uhdmtopModules: for each top-level module)
  does represent the instance hierarchy and elaborated nets (Full paths like top.u1.n1 can be extracted).
  That representation does not contain the processes and assignments on the other hand.

- It is the responsibility of the client application to traverse the instantiated view and for each instance, retrieve the corresponding assignments, processes... from the corresponding module definition and do the binding to the actual nets in the instance. 

The Yosys and Verilator integration use this approach as both tools actually have their own binding mechanism.

We could create an other view (optional) that performs the full elaboration of the VPI model, but that will probably have to happen on the client side at runtime (not persisted on disk), because that would blow-up the size of the on-disk representation. In EDA tools, the VPI is elaborated at runtime.

hs-apotell, 2020-05-03:
Appreciate the detailed explanation. This very well coincides with what we are seeing in practice. Just wondering if there's any detailed documentation on how and why to guide us?

Having both non-elaborated and elaborated views being called ```UHDM::module``` is confusing. When looking at any snippet of code, especially out of context, it is not very obvious whether or not it is elaborated.

Also, the optional view you mentioned will surely help us a lot, especially, since we don't have a parser/compiler expert on our team (at least not yet). In the meantime, a working example of "how to bind" something very simple would be a good start.

alaindargelas, 2020-05-03:
"Having both non-elaborated and elaborated views being called UHDM::module is confusing." 
that is unfortunately in the Standard. I forgot to mention that the VPI Standard allows to traverse all the modules elaborated and non elaborated (as we do here) and they are both called modules (inheriting from instances). The way to distinguish is "how you got access to the pointer":
  -  from the vpiIterate(vpiModule, NULL) you get the top modules (Elaborated) and you can traverse the instance tree (See page 976: Top-level modules shall be accessed using vpi_iterate() with a NULL reference object.)
  - from the vpi_handle_by_name(NULL, "name") you get the non-elaborated modules (per Standard).  (BTW, our vpi_handle_by_name implementation at this time only supports getting objects from an existing handle - not from the NULL root)

For Binding:
Create a map of <name/module_def>
For each instance, you can 
      - create a map<name/net_or_params_object>, 
       - get the module definition from the module instance (VpiDefName and using the module map)
       - get the processes and assignments from the module_def
       - using the name/net map, bind all the ref_objs on lhs and rhs of assignments

hs-apotell, 2020-05-04:
> ... get the processes and assignments from the module_def

I see a ```VectorOfprocess* Process()``` function in ```UHDM::module``` class but I don't see a central list of all assignments of a module. What am I missing? Do I have to recursively walk the tree to find individual assignments?

Also, since the goal is to make the module instance a fully elaborated instance, shouldn't the assignments come from the module instance and not from the module definition?

> ... bind all the ref_objs on lhs and rhs of assignments

When you say, "bind" do you mean -

1. track this binding as a separate map per module? i.e. map of ```UHDM::ref_obj``` instance to ```UHDM::parameters``` (or ```UHDM::net```) instance, or
2. find the bound ```UHDM::parameters``` instance and set it back into the module instance object using ```UHDM::assignment::Lhs(...)``` or ```UHDM::assignment::Rhs(...)``` methods.

I tried the second approach and found that the Lhs of ```UHDM::assignment``` is an ```UHDM::expr``` and the bound variable is either ```UHDM::parameters``` or ```UHDM::net```. These two are mutually distinct trees in the class hierarchy.

Also, found that there are number of other references that needs to be fixed up -
* if / if else / else statement condition
* case statement condition
* case item expression

Appreciate your continued support.

alaindargelas, 2020-05-05:
Give me a couple of days to come up with a comprehensive solution.

alaindargelas, 2020-05-06:
I created a Listener example that illustrate how the Elaborator can be written from the Folded Model: 
https://github.com/alainmarcel/UHDM/blob/master/tests/listener_elab.cpp

I can create a similar example using a walker instead of the listener, but the mechanism is essentially the same.

The Folded Model on UHDM side works well when the client application has itself an elaborated model (Yosys/Verilator).

What I propose here is that you perform the full elaboration using the technique I propose and create your data structure while doing so, but that raised a question:
- Do you actually have your own data structure or were you planning to annotate some form of properties to the UHDM (That would explain more why you want the UHDM/VPI itself to be fully elaborated)?

alaindargelas, 2020-05-06:
More direct answers to your questions:

... I see a VectorOfprocess* Process() function in UHDM::module class but I don't see a central list of all assignments of a module. What am I missing? Do I have to recursively walk the tree to find individual assignments?

Correct, walk, or use the Listener

... Also, since the goal is to make the module instance a fully elaborated instance, shouldn't the assignments come from the module instance and not from the module definition?

The current Goal is not to create a fully elaborated VPI/UHDM model for the reason that the main client applications already have such a view and UHDM is used as a front-end to these applications.
That is why I need to understand more your usage model.

... When you say, "bind" do you mean -

track this binding as a separate map per module? i.e. map of UHDM::ref_obj instance to UHDM::parameters (or UHDM::net) instance, or
find the bound UHDM::parameters instance and set it back into the module instance object using UHDM::assignment::Lhs(...) or UHDM::assignment::Rhs(...) methods.
I tried the second approach and found that the Lhs of UHDM::assignment is an UHDM::expr and the bound variable is either UHDM::parameters or UHDM::net. These two are mutually distinct trees in the class hierarchy.

Also, found that there are number of other references that needs to be fixed up -

if / if else / else statement condition
case statement condition
case item expression
Appreciate your continued support.

[AD] Correct, all nets need to be bound.

hs-apotell, 2020-05-06:
@alaindargelas Your example is pretty close to what I have as a walker implementation. So far so good.

> Do you actually have your own data structure or were you planning to annotate some form of properties to the UHDM (That would explain more why you want the UHDM/VPI itself to be fully elaborated)?

Ideally, I don't want to maintain my own data structure. I would like to "annotate" UHDM tree itself so I have "that one thing" to answer my questions and support all my use cases. Having parallel data structures brings in a number of issues to deal with (synchronization, dirty flag management, incremental edits, object ownership management, etc to name a few) - I would rather not have to deal with all that across my application. All that complexity should be hidden behind the API.

Alternatively, I will be writing a wrapper around UHDM tree and provide a new API to serve the purpose. Whatever I do it won't be very different in functionality to what UHDM already does, except it might leave features out that I don't need/want. I would rather not take this route.

My forward looking goal is to use Surelog/UHDM for building an interactive FSM viewer while supporting features like being able to point and click to show snippets of code where a node/state variable was declared or where specific state transition was deduced. The longer term goal is to be able to implement a "FSM stepper" which is a simulation environment for debugging FSM(s) in large designs. One of the our tenet is to support "live editing" of source files and that is where a fully elaborated model (and "one" data structure) becomes critical.

Hope this helps answer your concerns/questions.

alaindargelas, 2020-05-07:
Got it. I'll develop an auto-generated full Elaborator to be invoked on-demand (not default) for client applications like yours.
I'll also add client data hooks to the BaseClass (any object) so the entire tree can be annotated, along with an access to an ID per object so you can have property annotation based on ID.
If you have better suggestions for the annotations, they are welcome.

alaindargelas, 2020-05-07:
For instance do annotations need to be persisted on disk?

Thomasb81, 2020-05-07:
> If you have better suggestions for the annotations, they are welcome.
> For instance do annotations need to be persisted on disk?

Isn't it the purpose of attributes ( chapter 5.12 / Ieee1800-2017.1) to attach vendor details on specific syntax construction and allow it to be store in the data model ?

Thomas

hs-apotell, 2020-05-07:
> If you have better suggestions for the annotations, they are welcome.

Actually, two.
1. Ensure type safety for annotations. May be provide a pure-virtual base class that all annotations need to subclass.
2. The ID per object would be great (In fact, I myself was about to propose this). One solution that has worked great for us in the past is a incremental number that is available as a public member and not as a function. Snippet below -

```
class BaseObject
{
private:
  static int sId = 0;

public:
  const int id = ++sId;
};
```
The advantage of this is that it can be used with the dumbest debuggers to add conditional breakpoints (assuming that there is a strict order of construction). Comes in very handy when working with large designs and you need some way to break into the code at specific conditions. And of course, the ID should be persisted.

> For instance do annotations need to be persisted on disk?

I can't envision a scenario where we would want to persist this data. Surelog/UHDM doesn't necessarily understand the model of this annotation data to guarantee an error-free serialization. Plus, the annotations are "user data" and so external to the core intent of Surelog/UHDM. As long as the IDs are persisted, the client can very easily write their own serialization for annotations. I would recommend leaving that as an exercise for the clients.

alaindargelas, 2020-05-07:
@Thomasb81, the vendor specific attributes are the ones end-users embed in their SystemVerilog code to direct the tool to do some specific actions like (* synthesis off/on *).
They are, as you mentioned integral part of the Data Model (37.81 Attribute), But their payload is typically a "string". BTW, They still need to be added to UHDM.

Here, we are discussing internal arbitrary annotations that the client application would need to decorate potentially all the nodes in the object tree, not just the nodes that the section 37.81 authorizes, and with arbitrary payload (sparse or dense), for instance, attach to an always block a C++ object that represents a Flipflop class from the client application.

@hs-apotell, Ill add the mechanisms you suggested and remove my void* ClientData.

Thomasb81, 2020-05-07:
> @Thomasb81, the vendor specific attributes are the ones end-users embed in their SystemVerilog code to direct the tool to do some specific actions like (* synthesis off/on *).

That's one possible usecase. But it seems to me, tools are use to use pragma pre-processor directive for this particular use-case...

> Here, we are discussing internal arbitrary annotations that the client application would need to decorate potentially all the nodes in the object tree, not just the nodes that the section 37.81 [...]

I probably miss something. For me an always block is a _process_ which is listed in section 37.81.
I don't understand the limitation you are pointing,  and consequently why attribute cannot be used in this context. I understand tha it will require to cast object into string, which could be sub-optimal.
Those attribute would not necessarily be read from the source code but could be feed by application context, and be use later by next application stage...

alaindargelas, 2020-05-08:
Some facts that should remove the confusion: 
- SystemVerilog attributes (* Synopsys synthesis off/on *) were added to the language to make it cleaner than just plain comments like // Synopsys translate off    and be attached (per standard following a strict BNF semantic) to a limited set of objects in the Data Model, attribute contents do not follow a standard themselves
- Pragma pre-processors directives ie: `expand_vectornets , `protect (from the original Verilog language) have a limited usage but which still fit in the Verilog writer use case and are essentially dealt with at the preprocessor stage and are mot query-able in the Data Model itself, they tend to be used in different compiler stages.

- Client code needs to annotate potentially all the objects in the Data Model (depending on the application) with arbitrary internal data structures. Attributes are supposed to be attached only to a subset of all the objects classes in the Data Model with a limited set a data, basically what the VPI value object can contain (a union of string, int, real), it just does not fit the bill for most custom data structures.

I'll be adding the "attributes", the ID and type safe pointer, they all fit different needs.

alaindargelas, 2020-05-09:
Elaboration is implemented, it needs testing so I'm leaving this issue open. It only performs net binding at this time, I need to think about parameters a bit more. The net binding is limited to direct nets in the current instance, also need to implement the full semantic of hierarchical binding described in the standard (sub instance, parent instance).

alaindargelas, 2020-05-09:
I'll be adding the ID and attributes shortly.

hs-apotell, 2020-05-09:
That's great. Happy to assist with testing.

Unfortunately, I need parameter binding (including parameter assignments) more than the net binding for the immediate task at hand. If this isn't too far in the future, I would wait for that to land and then pick it up.

alaindargelas, 2020-05-10:
Added UhdmId() persisted ID, attributes and CleintData pointer.
Last item remaining is the binding of parameter.

alaindargelas, 2020-05-11:
The original issue is now fixed in the elaborated tree (-elabuhdm command line option and look at the elaborated UHDM tree under topModules):
Same binding occurs for parameters.
Please open different issues when testing.

           \_assignment: , id:1704, line:16
              |vpiBlocking:1
              |vpiLhs:
              \_ref_obj: (CurState), id:1705
                |vpiName:CurState
                |vpiActual:
                \_logic_net: (CurState), id:1266, line:11
                  |vpiName:CurState
                  |vpiNetType:48
                  |vpiParent:
                  \_module: work@FSM1, id:1335, file:fsm1.v, line:1, parent:work@top

closed 2020-05-11 by alaindargelas