chipsalliance/Surelog #354: CMake install fails

hs-apotell, 2020-05-06:
Source paths for ```libflatbuffers.a``` and ```libuhdm.a``` is setup wrong in the CMakeLists.txt

https://github.com/alainmarcel/Surelog/blob/2190fa9a383d50b1fddf4bececeaeb21de33af90/CMakeLists.txt#L389
https://github.com/alainmarcel/Surelog/blob/2190fa9a383d50b1fddf4bececeaeb21de33af90/CMakeLists.txt#L390

```
        ${PROJECT_SOURCE_DIR}/build/third_party/flatbuffers/libflatbuffers.a
        ${PROJECT_SOURCE_DIR}/build/third_party/UHDM/libuhdm.a
```

should be
```
        ${CMAKE_BINARY_DIR}/third_party/flatbuffers/libflatbuffers.a
        ${CMAKE_BINARY_DIR}/third_party/UHDM/libuhdm.a
```

alaindargelas, 2020-05-06:
Can you make a push request?
Let me add you as a contributor.
Thanks
Alain

alaindargelas, 2020-05-06:
Mmm I can't find you. Is "hs" your user name?

hs-apotell, 2020-05-06:
> Can you make a push request?

Sure thing. I wasn't sure whether or not you were accepting external pull requests.

> Is "hs" your user name?

Yes. You can probably also find me with name "hs-apotell"

alaindargelas, 2020-05-07:
The more the merrier!
I added you.

mithro, 2020-05-07:
You don't need to be a collaborator to send pull requests.

mithro, 2020-05-07:
Instructions are at https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

closed 2020-05-07 by hs-apotell