chipsalliance/Surelog #364: Function calls do not store arguments in UHDM

rkapuscik, 2020-05-11:
For following code:
```verilog
assign next_state = fsm_function(state, req_0, req_1);
```
we get the following in UHDM:
```
|vpiContAssign:
\_cont_assign: , id:13, line:30
  |vpiRhs:                                    
  \_ref_obj: (fsm_function), id:12, line:30
    |vpiName:fsm_function
  |vpiLhs:
  \_ref_obj: (next_state), id:14
    |vpiName:next_state
    |vpiActual:
    \_logic_net: (next_state), id:203, line:28
      |vpiName:next_state
      |vpiNetType:1                           
      |vpiParent:
```
It looks like the arguments are skipped altogether.

alaindargelas, 2020-05-11:
Can you include a complete example that compiles (module .... endmodule)? Or point to an existing test in the regression.

alaindargelas, 2020-05-12:
Fixed by #367

PR #367 (2020-05-12): uhdm: support FSMFunction test FSM style
Fixes #364, #365
closed 2020-05-12 by alaindargelas