chipsalliance/Surelog #376: uhdm-dump not anymore where regression.tcl expects it (with fresh UHDM)

hzeller, 2020-05-14:
The regression.tcl expects that the uhdm-dump binary is at the same place installed where surelog is in the dist ( https://github.com/alainmarcel/Surelog/blob/master/tests/regression.tcl#L155 ).
(That would be `build/dist/Release/uhdm-dump`).

However, with latest UHDM, it is not there anymore, but only to be found in `./build/third_party/UHDM/uhdm-dump`

As a result, we see FAILDUMP results:
https://travis-ci.com/github/alainmarcel/Surelog/builds/165906929#L1899

Not sure what is different, so not clear about a good solution yet: maybe the more restrictive CMake change in https://github.com/alainmarcel/UHDM/pull/228 prevents it from being there ? Should we just point the `UHDM_DUMP_COMMAND` to the location in third_party/UHDM ?

alaindargelas, 2020-05-14:
@hs-apotell the changelist alainmarcel/UHDM#228 broke Surelog regression. The last stage of the regression is to do a uhdm-dump, the executable has changed place with your changelist.
I don't understand the rational for changing the location of the tests executable in UHDM. We need to reconcile what you are trying to achieve on Windows and how the overall system needs to be setup: UHDM and Surelog with UHDM as a subsystem.
I will override your changes until you give an explanation on why the previous setup did not work on Windows.

hs-apotell, 2020-05-15:
Sorry about the regression. I was supposed to follow up with accompanying surelog change but got sidetracked with the crash on UHDM serialization (Windows only). I have an open pull request to bring both projects back in sync and also fix the crash.

The rationale behind the change was that Visual Studio and QT Creator, both doesn't play well when the output files are moved around. They expect these files to be in specific location. Plus, on Windows, CMake works in multi-configuration mode which brings in its own challenges. The whole change started with me fixing the following scenario -
* Target surelog (library project) generates "surelog.lib" in the root output folder
* Target surelog_bin (binary/executable project) also generates "surelog.lib" in the root output folder.

The problem is that one of the library that surelog executable links to is exporting symbols which the linker exports as the library (unusable output but the linker still does it because it doesn't differentiate between a dll and executable - required the former but useless for the latter). If you haven't heard of DLL hell on windows, you should read up on it. It really is what the name implies.

I had two options - either rename the binary target or move it to a different folder. I chose the latter and do the work to solve the dev environment problem and this one. One added advantage of the change is that the library dependencies are setup right and so projects dependent on UHDM and/or Surelog get all the transitive dependencies automatically.

closed 2020-05-15 by hs-apotell