chipsalliance/Surelog #381: Support parsing and storing Ibex in UHDM

rkapuscik, 2020-05-15:
This issue intends to provide an overview to find out what elements are still missing in order to store [Ibex](https://github.com/lowRISC/ibex/) in UHDM as a long-term goal.

@alaindargelas, could you provide input on what you know is missing at the moment? I will edit this and link to related issues (I don't see any right now).

alaindargelas, 2020-05-16:
@rkapuscik 
1) You can take a look at the uhdm line coverage report:
https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/Opentitan/surelog.uhdm.chk

2) Then open the corresponding file to see what is the construct not supported.

Notes:
- generate blocks are supported, so only the active branch will be in uhdm, so ignore the sections of code in the inactive branches of generate statements as they are not in uhdm for a good reason.
- import statements are not supported and should not, I'll filter them out
- weird thinks like lines containing only constructs like  "  ) ( " are not in uhdm, ignore.

alaindargelas, 2020-07-16:
Assigning @rkapuscik to track when we are done.

alaindargelas, 2020-08-07:
This is tracked under https://github.com/alainmarcel/uhdm-integration

closed 2020-08-07 by alaindargelas