chipsalliance/Surelog #386: Compiling with swig4.0 crashes some of the regressions

hzeller, 2020-05-16:
On my system, the default swig is of version 4.0

```
swig -version

SWIG Version 4.0.1

Compiled with g++ [x86_64-pc-linux-gnu]
```

Compiling Surelog with this will create a binary that crashes on some inputs in the regression. 

```
make debug && cd tests/SimpleIncludeAndMacros/ && gdb -ex run --args ../../dbuild/bin/surelog   top*.v -d 0  +incdir+.+..  +libext+.v -writepp -y blah +incdir+foo -v lib.v  -fileunit +define+N=8  -mt max  -verbose  -pythonlistener -parse

#[... compiling ... ]

Starting program: /home/hzeller/src/my/Surelog/dbuild/bin/surelog top_1.v top_2.v top_3.v top_4.v top.v -d 0 +incdir+.+.. +libext+.v -writepp -y blah +incdir+foo -v lib.v -fileunit +define+N=8 -mt max -verbose -pythonlistener -parse
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[INFO :CM0023] Creating log file ./slpp_unit/surelog.log.

[INFO :CM0020] Separate compilation-unit mode is on.

[INFO :CM0024] Executing with 32 threads.

[WARNI:CM0007] Library path "blah" does not exist.

Program received signal SIGSEGV, Segmentation fault.
0x000055555568b9ff in SWIG_MangledTypeQueryModule (start=0x0, end=0x0, name=0x5555561337b3 "_p_char") at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:587
587         if (iter->size) {
(gdb) bt
#0  0x000055555568b9ff in SWIG_MangledTypeQueryModule (start=0x0, end=0x0, name=0x5555561337b3 "_p_char") at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:587
#1  0x000055555568bb19 in SWIG_TypeQueryModule (start=0x0, end=0x0, name=0x5555561337b3 "_p_char") at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:631
#2  0x000055555568e4db in SWIG_Python_TypeQuery (type=0x5555561337b3 "_p_char") at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:2454
#3  0x000055555568f471 in SWIG_pchar_descriptor () at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:4701
#4  0x000055555568f5f2 in SWIG_AsCharPtrAndSize (obj=0x5555566a61c0 <_Py_NoneStruct>, cptr=0x0, psize=0x0, alloc=0x0) at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:4800
#5  0x0000555555697944 in _wrap_SLsetWaiver (self=0x7ffff6bc95e0, args=0x7ffff6bc94f0) at /home/hzeller/src/my/Surelog/src/API/slapi_wrap.cxx:7714
#6  0x00007ffff7ceb457 in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#7  0x00007ffff7ceb8c7 in _PyObject_MakeTpCall () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#8  0x00007ffff7abd083 in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#9  0x00007ffff7ac4f96 in _PyEval_EvalFrameDefault () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#10 0x00007ffff7c0faab in _PyEval_EvalCodeWithName () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#11 0x00007ffff7ceaf00 in _PyFunction_Vectorcall () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#12 0x00007ffff7abcffd in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#13 0x00007ffff7abe6b5 in _PyEval_EvalFrameDefault () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#14 0x00007ffff7c0faab in _PyEval_EvalCodeWithName () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#15 0x00007ffff7c0fe2e in PyEval_EvalCodeEx () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#16 0x00007ffff7c101db in PyEval_EvalCode () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#17 0x00007ffff7bd15ac in ?? () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#18 0x00007ffff7bd16f2 in PyRun_FileExFlags () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#19 0x00007ffff7bd2e3b in PyRun_SimpleFileExFlags () from /usr/lib/x86_64-linux-gnu/libpython3.8.so.1.0
#20 0x00005555556a1dab in SURELOG::PythonAPI::loadScript_ (name=..., check=false) at /home/hzeller/src/my/Surelog/src/API/PythonAPI.cpp:113
#21 0x00005555556a1f59 in SURELOG::PythonAPI::loadScriptsInInterp_ () at /home/hzeller/src/my/Surelog/src/API/PythonAPI.cpp:152
#22 0x00005555556a2612 in SURELOG::PythonAPI::loadScripts () at /home/hzeller/src/my/Surelog/src/API/PythonAPI.cpp:203
#23 0x0000555555689d6f in executeCompilation (argc=23, argv=0x7fffffffdfb8, diff_comp_mode=false, fileunit=false, overallStats=0x0) at /home/hzeller/src/my/Surelog/src/main.cpp:55
#24 0x000055555568aba1 in main (argc=23, argv=0x7fffffffdfb8) at /home/hzeller/src/my/Surelog/src/main.cpp:223
(gdb) 
```

hzeller, 2020-05-16:
It looks like it happens deep in the generated swig code, attempting to go through some internal list which is pointing to null-pointers. Maybe there is something simple that needs to be done before we use it, like calling some initialization ?

alaindargelas, 2020-05-17:
Can you try to run the script in the regression first?

~/Surelog$ cat tests/UnitPython/UnitPython.sl
     -writepp -parse -verbose    -d ast +incdir+../../../UVM/ovm-2.1.2/src/ +incdir+../../../UVM/vmm-1.1.1a/sv     -mt max -fileunit  top.v   -nocache -pythonevalscriptperfile myscriptPerFile.py  -pythonevalscript myscriptPerDesign.py -pythonlistenerfile my_listener.py

hzeller, 2020-05-18:
This one seemingly runs fine:
```
Surelog/tests/UnitPython$ ../../build/bin/surelog $(cat UnitPython.sl)
[INF:CM0023] Creating log file ./slpp_unit/surelog.log.

[INF:CM0024] Executing with 32 threads.

[INF:CM0020] Separate compilation-unit mode is on.

[WRN:CM0005] Include path "../../../UVM/ovm-2.1.2/src/" does not exist.

[WRN:CM0005] Include path "../../../UVM/vmm-1.1.1a/sv" does not exist.

[INF:PP0122] Preprocessing source file "top.v".

[INF:PP0122] Preprocessing source file "builtin.sv".

[INF:PA0201] Parsing source file "builtin.sv".

[INF:PA0201] Parsing source file "top.v".

LIB:  work
FILE: top.v
n<> u<0> t<Null_rule> p<123> s<122> l<2>
n<pkg> u<1> t<StringConst> p<22> s<6> l<2>
n<Configuration1> u<2> t<StringConst> p<4> s<3> l<3>
n<> u<3> t<Endclass> p<4> l<4>
n<> u<4> t<Class_declaration> p<5> c<2> l<3>
n<> u<5> t<Package_or_generate_item_declaration> p<6> c<4> l<3>
n<> u<6> t<Package_item> p<22> c<5> s<20> l<3>
n<IDLE> u<7> t<StringConst> p<8> l<6>
n<> u<8> t<Enum_name_declaration> p<15> c<7> s<10> l<6>
n<WRITE> u<9> t<StringConst> p<10> l<6>
n<> u<10> t<Enum_name_declaration> p<15> c<9> s<12> l<6>
n<READ> u<11> t<StringConst> p<12> l<6>
n<> u<12> t<Enum_name_declaration> p<15> c<11> s<14> l<6>
n<DONE> u<13> t<StringConst> p<14> l<6>
n<> u<14> t<Enum_name_declaration> p<15> c<13> l<6>
n<> u<15> t<Data_type> p<17> c<8> s<16> l<6>
n<fsm_t_pkg> u<16> t<StringConst> p<17> l<6>
n<> u<17> t<Type_declaration> p<18> c<15> l<6>
n<> u<18> t<Data_declaration> p<19> c<17> l<6>
n<> u<19> t<Package_or_generate_item_declaration> p<20> c<18> l<6>
n<> u<20> t<Package_item> p<22> c<19> s<21> l<6>
n<> u<21> t<Endpackage> p<22> l<8>
n<> u<22> t<Package_declaration> p<23> c<1> l<2>
n<> u<23> t<Description> p<122> c<22> s<29> l<2>
n<Configuration2> u<24> t<StringConst> p<26> s<25> l<10>
n<> u<25> t<Endclass> p<26> l<11>
n<> u<26> t<Class_declaration> p<27> c<24> l<10>
n<> u<27> t<Package_or_generate_item_declaration> p<28> c<26> l<10>
n<> u<28> t<Package_item> p<29> c<27> l<10>
n<> u<29> t<Description> p<122> c<28> s<36> l<10>
n<> u<30> t<Module_keyword> p<34> s<31> l<13>
n<bottom> u<31> t<StringConst> p<34> s<33> l<13>
n<> u<32> t<Port> p<33> l<13>
n<> u<33> t<List_of_ports> p<34> c<32> l<13>
n<> u<34> t<Module_nonansi_header> p<35> c<30> l<13>
n<> u<35> t<Module_declaration> p<36> c<34> l<13>
n<> u<36> t<Description> p<122> c<35> s<121> l<13>
n<> u<37> t<Module_keyword> p<41> s<38> l<16>
n<top> u<38> t<StringConst> p<41> s<40> l<16>
n<> u<39> t<Port> p<40> l<16>
n<> u<40> t<List_of_ports> p<41> c<39> l<16>
n<> u<41> t<Module_nonansi_header> p<120> c<37> s<51> l<16>
n<pkg> u<42> t<StringConst> p<43> l<17>
n<> u<43> t<Package_import_item> p<44> c<42> l<17>
n<> u<44> t<Package_import_declaration> p<45> c<43> l<17>
n<> u<45> t<Data_declaration> p<46> c<44> l<17>
n<> u<46> t<Package_or_generate_item_declaration> p<47> c<45> l<17>
[... 182 lines cut ...]
builtin::string  0 0
builtin::system  0 0
pkg::Configuration1 top.v 3 4
work@Configuration2 top.v 10 26
work@mailbox /home/hzeller/src/my/Surelog/build/bin/sv/builtin.sv 4 100
work@process /home/hzeller/src/my/Surelog/build/bin/sv/builtin.sv 33 168
work@semaphore /home/hzeller/src/my/Surelog/build/bin/sv/builtin.sv 58 240
1
work@top work@top
bot work@bottom
[WRN:PY0099] top.v:16: Instance "work@top".

[WRN:PY0099] top.v:23: Instance "bot".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 3
[WARNING] : 7
[   NOTE] : 5
```

However, looking at stderr, there is this line which looks a bit suspicious; but it shows up if compiled with swig3.0 and swig4.0
```
AttributeError: module '__main__' has no attribute 'exitEveryRule'
```

alaindargelas, 2020-05-18:
Someone submitted a pretty large change around Python API, I guess they broke some things, like the option you used does not trigger Python initialization and potentially other issues like the missing command you report in the default test...

alaindargelas, 2020-05-18:
See #331

hzeller, 2020-06-24:
I had on that workstation reverted back to swig 3.0, but on a newer debian I don't have the option anymore for old swig. So I guess will have to look at the detail.

alaindargelas, 2021-04-11:
The Python API is not built by default anymore, and there are really no users for it, can we close this?

hzeller, 2021-04-13:
Sounds good.

Given that we now have UHDM, we probably should consider phasing out the Python API at some point anyway and request people to use UHDM as their stable interface to Surelog vs. allowing them to poke in Surelog internals which makes it harder to change them.

We can start by removing it from the cmake files and see if someone complains :)

alaindargelas, 2021-04-13:
It is "removed" as this week-end in the sense, you need to compile with a -DSURELOG_WITH_PYTHON=1 which is non-default.Without this compiler option, both Python include files and libraries are excluded from the build (cmake and source code).I still want to keep Python optional for users who simply want to customize messages or write simple Python linting rules. And I'll have some use for it in the future.

closed 2021-10-03 by alaindargelas