chipsalliance/Surelog #406: Command line input file paths

hs-apotell, 2020-05-22:
Is there an assumption that all input files on the command line will be filenames only i.e. non-relative, non-absolute paths?

If true, it doesn't seem to be enforced in code and "bad inputs" fails deep inside when attempting to create cache file. This was reproduced on windows with a relative file input.

alaindargelas, 2020-05-22:
- The -y <dir> Verilog option make it such that you have to find files of corresponding module names in the <dir>, it is a search directive for cell primitives (files not specified on the command line but searched for because their module of the same name are referenced in the design).

- The -v <file> is for cells like -y but this time an absolute or relative path has to be given for a single file 

- The -I<dir> and +incdir+<dir> are the search path for include files (files not specified on the command line but included in source files)

- All other -f <listfile>, .v, .sv, .cfg files have to be either relative or absolute, if a specified file does not exist there is a fatal error.

The cache file mechanism creates a slpp_all/cache/work/<modified file path>

The <modified file path> is a path where all ../ is replaced by __/ to keep the file structure local to the cache dir. That works nicely on Linux. Maybe Windows need to use a different character that _ ?

hs-apotell, 2020-05-23:
Appreciate the detailed explanation. It took some digging but I narrowed down the root cause of the problem I was having with absolute paths. Its the implementation of ```StringUtils::eliminateRelativePath``` which accounts for relative paths but not absolute paths. Note that absolute paths on windows has colon which is not a legal character except for when used to designate drive letter. I will create a pull request with appropriate fix.

closed 2020-05-29 by alaindargelas