chipsalliance/Surelog #410: FileUtils::collectFiles is not thread safe

hs-apotell, 2020-05-23:
Use of `chdir` in implementation is likely to cause non-reproducible errors, especially in multi-threaded environment.

Suggestion - FileUtils::collectFiles(std::string, SymbolTable) is making assumptions about what the base/starting path should be (and also changing to root folder when the input path is absolute). Why not take that the base, non-wildcard path as the argument?

alaindargelas, 2020-05-24:
Please propose another implementation. It has to match the spec page 904 of the 2017 SystemVerilog Standard:

Library map file details:
1) file_path_spec uses file-system-specific notation to specify an absolute or relative path to a
particular file or set of files. The following shortcuts/wildcards can be used:
? single character wildcard (matches any single character)
* multiple character wildcard (matches any number of characters in a directory/file name)
... hierarchical wildcard (matches any number of hierarchical directories)
.. specifies the parent directory
. specifies the directory containing the lib.map
Paths that end in / shall include all files in the specified directory. Identical to /*.
Paths that do not begin with / are relative to the directory in which the current lib.map file
is located.
2) The paths ./*.v and *.v are identical, and both specify all files with a .v suffix in the current
directory

alaindargelas, 2020-05-24:
To be noted, Library/Config file parsing should not happen anywhere but in the main thread, when processing the project command line.

alaindargelas, 2020-05-24:
Have you noticed actual runtime issues?

hs-apotell, 2020-05-25:
> Please propose another implementation.

Will do.

> To be noted, Library/Config file parsing should not happen anywhere but in the main thread, when processing the project command line.

That might be true within the scope of the Surelog code but you can't draw that line for all the applications using Surelog. This function is affecting the state of entire application. I could envision cases where I invoke the Surelog parser on its own thread, especially, true for interactive UI applications.

> Have you noticed actual runtime issues?

No, not yet. I just happen to look thru' other function implementations when working on another task in that file. That said, I hope that is not a criteria to fix something that has been identified to be a potential issue. This is an implementation with side-effects which aren't explicit from its function name. Plus, this code is likely to have permission issues as well when attempting to change working directory to root and also stepping into other user's home directory.

hs-apotell, 2020-06-05:
@alaindargelas I have a working version of the collectFiles and I was hoping to compare the results of my implementation with yours. Ran into some problems along the way and following up to ensure my understanding is correct (and its a bug in current implementation).

I found only one reference to this function in Surelog and that's [here](https://github.com/alainmarcel/Surelog/blob/8be10c70c85af820cd5d7b2d2d18b851e09b641a/src/Library/SVLibShapeListener.cpp#L78) Haven't yet ran thru' the regression tests to see if there's any example that actually goes thru' this code path.

Following no other easy way to test, I implemented my own unit test with following inputs -
* ./dir1
* ./dir1/*.v
* ./dir1/*.vhd
* ./dir1/abc?.v

The folder named dir1 has following 4 files.
* ./dir1/abc1.v
* ./dir1/abc2.v
* ./dir1/abc3.v
* ./dir1/cba4.v
* ./dir1/abc1234.vhd

I was expecting the current implementation to list 5, 4, 1 and 3 files for the inputs above. However, each of them returned 0 results. My working/current directory during this run was the folder that contained the folder named "dir1".

Little investigation into the current implementation tells me that the conversion from string literal to a valid regex expression has some issues.

At [this](https://github.com/alainmarcel/Surelog/blob/8be10c70c85af820cd5d7b2d2d18b851e09b641a/src/Utils/FileUtils.cpp#L182) location, 

regexp = "."
newregexp = "\."

This is wrong and ends up finding nothing. The result is an empty vector. I removed "./" prefix from all inputs and that resulted in the expected output.

I will create a pull request later today once I validate all the regression tests locally.

alaindargelas, 2020-06-05:
Your approach seems correct.The tests to watch for are:LibraryIntercon
UnitLibrary

The ./ as in ./dir1 is superfluous and is not expected to be used in the regexp specification.

alaindargelas, 2020-06-06:
Fixed by #479

closed 2020-06-06 by alaindargelas