chipsalliance/Surelog #418: Split surelog library (Surelog and PySurelog)

hs-apotell, 2020-05-25:
Not sure if this is feasible but throwing it out there to evaluate potential viability.

**Suggestion** - Split surelog into two distinct libraries
* surelog - native only with no python dependencies
* pysurelog - pulls in python as a dependency and be dependent on surelog

**Rationale** - Our application uses embedded python (might even switch to stack-less python, IronPython, for ease of use) and that conflicts with Surelog's use of python.

alaindargelas, 2020-05-26:
I have no bandwidth to pull out Python in the foreseeable future. But it should be fairly easy to add #ifdef to conditionally compile all the Python code. Please propose a changelist that keeps Python by default and only if you define something like -DNO_PYTHON to Cmake, then Python should not be compiled.

alaindargelas, 2021-04-11:
Python support is not built in by default: #1188

closed 2021-04-11 by alaindargelas