chipsalliance/Surelog #420: Reference objects are mistaken for complex function calls

rkapuscik, 2020-05-25:
For test case:
```verilog
module dut (a, b);
    input [3:0] a;
    output [3:0] b;
    reg [3:0] b;
    assign b = {~(|a[3:2]), a[3:2]};
endmodule
```
In UHDM:
```
  |vpiContAssign:
  \_cont_assign: , id:3, line:5
    |vpiRhs:
    \_operation: , id:0, line:5
      |vpiOpType:33
      |vpiOperand:
      \_operation: , id:1, line:5
        |vpiOpType:4
      |vpiOperand:
      \_func_call: (a), id:2, line:5
        |vpiName:a
    |vpiLhs:
    \_ref_obj: (b), id:4
      |vpiName:b
      |vpiActual:
      \_logic_net: (b), id:31, line:4
```
This also results in range selection not being evaluated properly.

rkapuscik, 2020-05-25:
I've tried amending that in `compileExpression` in a way similar to the dotted names, but I'm unsure how to differentiate the reference objects from actual functions.

alaindargelas, 2020-05-26:
The Sibling of "a" is a range, you need to key on that and instead of making a "function call", make a "part select". The Antlr AST does not match the correct rule it looks like, might be a grammar ambiguity, but it can be fixed at the compilation time. If not, I'll check the grammar.

See section in CompileExpression that already creates a part select:

                } else if (fC->Type(Bit_select) ==
                           VObjectType::slPart_select_range) {
                  NodeId Constant_range = fC->Child(Bit_select);
                  result = compilePartSelectRange(component, fC, Constant_range, name, compileDesign, pexpr, instance);               
                  break;
                }

PR #426 (2020-05-26): Fix complex_func_call by keying on select node
Fixes #420.
closed 2020-05-26 by alaindargelas