chipsalliance/Surelog #433: What does VpiParent provide?

hs-apotell, 2020-05-27:
Little confused with the concept of VpiParent and what it provides.

* Should the statements within an if clause have the `UHDM::if_stmt` as its parent?
* What about `UHDM::case_item` parent? Should it be pointing to `UHDM::case_stmt`?
* In general, is the `VpiParent` dictated by code/block/scope hierarchy?

The specific problem I am trying to figure out is the second item above. How do I get to the `UHDM::case_stmt` given a `UHDM::case_item` pointer?

alaindargelas, 2020-05-27:
All of the above is correct.
I have not been super methodical in filling the VpiParent property for the statements and expressions.
I'll add the missing links as I find them.
On the other hand, the VpiParent should be correct for all instantiation, generate stmts, nets, ports (all objects that need to be elaborated in the Folded Model).

hs-apotell, 2020-05-27:
The specific case of `case_stmt` and `case_item` fails in elaborated model.

Would the `if_else` be considered the parent of corresponding/recurive child `if_else` statement as well?

alaindargelas, 2020-05-27:
I'll be working on this today.

alaindargelas, 2020-05-27:
#437 should fix this problem. Let me know if not.

hs-apotell, 2020-05-29:
Post elaboration (i.e. using `ElaboratorListener`) `case_item->VpiParent()->VpiCondition()->Actual_group()` is `nullptr`

alaindargelas, 2020-05-29:
Please add a minimalist Verilog test here.

hs-apotell, 2020-05-30:
The issue is easily reproducible with dut.v file under tests folder. But here's a smaller version of something similar.

```
module FSM(clock, keys, brake, accelerate, Speed);
input clock, keys, brake, accelerate;
output [3:0] Speed;
reg [3:0] Speed;

reg Ctrl1, Ctrl2, Ctrl3;

parameter Stop   =  4'b0000,
          Move   =  4'b0001,
          Turn   =  4'b0010,
          Slow   =  4'b0011,
          Medium =  4'b0100,
          Fast   =  4'b0101,
          Faster =  4'b0110;

always @(posedge clock or negedge keys) begin:FSM3
  
if (keys) begin
  Ctrl1 <= 0;
  Ctrl2 <= 0;
  Ctrl3 <= 0;

  if (accelerate)
    case (Speed)
    Stop:   Speed <= Slow;
    Move:   Speed <= Faster;
    Turn:   Speed <= Move;
    Slow:   Speed <= Medium;
    Medium: Speed <= Fast;
    Fast:   Speed <= Fast;
    Faster: Speed <= Stop;
    default:begin
    end   
    endcase
  end
else if (brake)
  case (Speed)
    Stop:   Speed <= Faster;
    Move:   Speed <= Faster;
    Turn:   Speed <= Move;
    Slow:   Speed <= Stop;
    Medium: Speed <= Slow;
    Fast:   Speed <= Medium;
    Faster: Speed <= Slow;
    default:begin
    end  
  endcase
else begin
  Speed <= Speed;
end 

end

endmodule

```

alaindargelas, 2020-05-30:
Fixed, see:
https://github.com/alainmarcel/Surelog/blob/master/tests/CaseFullElab/top.sv
https://github.com/alainmarcel/Surelog/blob/master/tests/CaseFullElab/CaseFullElab.log#L807

closed 2020-05-30 by alaindargelas