chipsalliance/Surelog #455: Bounds check failing

hs-apotell, 2020-05-31:
Hit a crash when running `TestBatchMode` test.

Root cause -
`param` == 1209
`fC->m_objects.size()` == 1209
`fC->m_objects[param].m_line` [here](https://github.com/alainmarcel/Surelog/blob/e9d2b482adb7c7920ce701f4797e717f9fe53118/src/Design/FileContent.cpp#L206) is outside the bounds of the vector.

Here's the callstack -
```
surelog.exe!std::vector<SURELOG::VObject,std::allocator<SURELOG::VObject>>::operator[](const unsigned __int64 _Pos=1209) Line 1500	C++
surelog.exe!SURELOG::FileContent::Line(unsigned int index=1209) Line 206	C++
surelog.exe!SURELOG::DesignElaboration::collectParams_(std::vector<std::string,std::allocator<std::string>> & params={...}, SURELOG::FileContent * fC=0x000002213da1f530, unsigned int nodeId=793, SURELOG::ModuleInstance * instance=0x00000221404a2b40, unsigned int parentParamOverride=0) Line 1240	C++
surelog.exe!SURELOG::DesignElaboration::elaborateInstance_(SURELOG::FileContent * fC=0x000002213da1f530, unsigned int nodeId=793, unsigned int parentParamOverride=0, SURELOG::ModuleInstanceFactory * factory=0x00000221403bb2c0, SURELOG::ModuleInstance * parent=0x00000221404a2b40, SURELOG::Config * config=0x0000000000000000) Line 511	C++
surelog.exe!SURELOG::DesignElaboration::elaborateModule_(std::string moduleName={...}, SURELOG::FileContent * fC=0x000002213da1f530, bool onlyTopLevel=false) Line 453	C++
surelog.exe!SURELOG::DesignElaboration::elaborateAllModules_(bool onlyTopLevel=false) Line 405	C++
surelog.exe!SURELOG::DesignElaboration::elaborate() Line 71	C++
surelog.exe!SURELOG::CompileDesign::elaboration_() Line 327	C++
surelog.exe!SURELOG::CompileDesign::elaborate() Line 252	C++
surelog.exe!SURELOG::Compiler::compile() Line 807	C++
surelog.exe!SURELOG::start_compiler(SURELOG::CommandLineParser * clp=0x0000022139959cc0) Line 32	C++
surelog.exe!executeCompilation(int argc=19, const char * * argv=0x00000221390456b0, bool diff_comp_mode=false, bool fileunit=false, SURELOG::ErrorContainer::Stats * overallStats=0x00000038383ff668) Line 70	C++
surelog.exe!batchCompilation(const char * argv0=0x00000221373a6d80, std::string batchFile={...}) Line 157	C++
surelog.exe!main(int argc=3, const char * * argv=0x00000221373a6d60) Line 241	C++
```

And here are the local variables responsible for the crash -
```

  | Name | Value | Type
-- | -- | -- | --
▶ | def | 0x000002213d2ef970 {m_name="uvm_pkg" m_library=0x00000221395c5970 {m_name="work" m_fileIds={ size=3 } ...} ...} | SURELOG::Package *
▶ | defParams | { size=0 } | std::vector<unsigned int,std::allocator<unsigned int>>
▶ | design | 0x0000022139970e50 {m_errors=0x0000022139910e90 {m_errors={ size=20 } m_errorSet={ size=20 } m_clp=0x0000022139959cc0 {...} ...} ...} | SURELOG::Design *
▶ | errors | 0x0000022139910e90 {m_errors={ size=20 } m_errorSet={ size=20 } m_clp=0x0000022139959cc0 {m_libraryPaths=...} ...} | SURELOG::ErrorContainer *
▶ | fC | 0x000002213da1f530 {m_elements={ size=22 } m_objects={ size=1209 } m_definitionFiles={ size=0 } ...} | SURELOG::FileContent *
  | i | 14 | unsigned int
  | ident | 1193 | unsigned int
▶ | instance | 0x00000221404a2b40 {m_definition=0x000002213d2ef6b0 {m_name="work@top" m_modportSignalMap={ size=0 } ...} ...} | SURELOG::ModuleInstance *
▶ | module | 0x000002213d2ef6b0 {m_name="work@top" m_modportSignalMap={ size=0 } m_modportClockingBlockMap={ size=0 } ...} | SURELOG::DesignComponent * {SURELOG::ModuleDefinition}
▶ | name | "UVM_NOCOMPARE" | std::string
  | nodeId | 793 | unsigned int
  | pack_id | 1 | unsigned int
▶ | pack_import | {fC=0x000002213da1f530 {m_elements={ size=22 } m_objects={ size=1209 } m_definitionFiles={ size=0 } ...} ...} | SURELOG::FileCNodeId
▶ | pack_imports | { size=3 } | std::vector<SURELOG::FileCNodeId,std::allocator<SURELOG::FileCNodeId>>
▶ | pack_name | "uvm_pkg" | std::string
▶ | packageFile | 0x000002213da1e730 {m_elements={ size=328 } m_objects={ size=431475 } m_definitionFiles={ size=0 } ...} | SURELOG::FileContent *
  | param | 1209 | unsigned int
▶ | params | { size=14 } | std::vector<std::string,std::allocator<std::string>> &
▶ | paramSet | { size=41 } | std::vector<SURELOG::FileCNodeId,std::allocator<SURELOG::FileCNodeId>> &
  | parentParamOverride | 0 | unsigned int
▶ | st | 0x000002213988bd00 {m_idCounter=2945 m_id2SymbolMap={ size=2945 } m_symbol2IdMap={ size=2945 } } | SURELOG::SymbolTable *
▶ | stopPoints | { size=0 } | std::vector<enum VObjectType,std::allocator<enum VObjectType>>
▶ | types | { size=0 } | std::vector<enum VObjectType,std::allocator<enum VObjectType>>
▶ | value | 0x000002213f2856d0 {m_type=Unsigned (4) m_nbWords=1 m_valueArray=0x00000221403e2508 {m_value=184549376 ...} ...} | SURELOG::Value * {SURELOG::LValue}

```

alaindargelas, 2020-05-31:
Fixed by #456

alaindargelas, 2020-05-31:
The issue reported here for this testcase is fixed. But there are many other occurrences in other tests due most likely to other misuse of FileContent::Line function.
I added a debug error message to track these.

alaindargelas, 2020-06-01:
Fixed by #459

closed 2020-06-01 by alaindargelas