chipsalliance/Surelog #462: Part select does not store parent expression

rkapuscik, 2020-06-01:
For code:
```verilog
module dut (a, o);
    input [3:0] a;
    output [2:0] o;
    wire [3:0] a;
    reg [2:0] o;
  assign o = a[3:1];
endmodule
```
Relevant part of UHDM:
```
  |vpiContAssign:
  \_cont_assign: , id:8, line:7
    |vpiRhs:
    \_part_select: , id:6, line:7
      |vpiConstantSelect:1
      |vpiLeftRange:
      \_constant: , id:4, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:3
      |vpiRightRange:
      \_constant: , id:5, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:1
    |vpiLhs:
    \_ref_obj: (o), id:9
      |vpiName:o
      |vpiFullName:@@BAD_SYMBOL@@
      |vpiActual:
      \_logic_net: (o), id:46, line:6
```
There is no information to what signal does the part_select refer (but it was present at some point, I did not see this issue when working on #443).

alaindargelas, 2020-06-01:
The VpiParent is not longer getting printed by the vpi_visitor printer, that is all. It was making the dumps too big and hard to read to have the parent info for each object. The data is there and can be retrieved with the API though.
i notice that you have a BAD_SYMBOL in your dump, which is not there in the main branch.

`design: (work@dut), id:22
|vpiName:work@dut
|uhdmallPackages:
\_package: builtin, id:23, parent:work@dut
  |vpiDefName:builtin
  |vpiFullName:builtin
|uhdmallClasses:
\_class_defn: (builtin::array), id:24
  |vpiName:builtin::array
  |vpiFullName:builtin.builtin::array
|uhdmallClasses:
\_class_defn: (builtin::queue), id:25
  |vpiName:builtin::queue
  |vpiFullName:builtin.builtin::queue
|uhdmallClasses:
\_class_defn: (builtin::string), id:26
  |vpiName:builtin::string
  |vpiFullName:builtin.builtin::string
|uhdmallClasses:
\_class_defn: (builtin::system), id:27
  |vpiName:builtin::system
  |vpiFullName:builtin.builtin::system
|uhdmallModules:
\_module: work@dut, id:28, file:tests/ScratchPad.sv, line:2, parent:work@dut
  |vpiDefName:work@dut
  |vpiFullName:work@dut
  |vpiPort:
  \_port: (a), id:29, line:2
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: , id:35
      |vpiActual:
      \_logic_net: (a), id:31, line:5
        |vpiName:a
        |vpiFullName:work@dut.a
        |vpiNetType:1
  |vpiPort:
  \_port: (o), id:30, line:2
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: , id:36
      |vpiActual:
      \_logic_net: (o), id:33, line:6
        |vpiName:o
        |vpiFullName:work@dut.o
        |vpiNetType:48
  |vpiContAssign:
  \_cont_assign: , id:8, line:7
    |vpiRhs:
    \_part_select: , id:6, line:7
      |vpiConstantSelect:1
      |vpiLeftRange:
      \_constant: , id:4, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:3
      |vpiRightRange:
      \_constant: , id:5, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:1
    |vpiLhs:
    \_ref_obj: (o), id:9
      |vpiName:o
      |vpiFullName:work@dut.o
      |vpiActual:
      \_logic_net: (o), id:33, line:6
  |vpiNet:
  \_logic_net: (a), id:31, line:5
  |vpiNet:
  \_logic_net: (o), id:33, line:6
|uhdmtopModules:
\_module: work@dut (work@dut), id:37, file:tests/ScratchPad.sv, line:2
  |vpiDefName:work@dut
  |vpiName:work@dut
  |vpiPort:
  \_port: (a), id:18, line:2, parent:work@dut
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: , id:19
      |vpiActual:
      \_logic_net: (a), id:13, line:5, parent:work@dut
        |vpiName:a
        |vpiFullName:work@dut.a
        |vpiNetType:1
        |vpiRange:
        \_range: , id:12
          |vpiLeftRange:
          \_constant: , id:10
            |INT:3
          |vpiRightRange:
          \_constant: , id:11
            |INT:0
  |vpiPort:
  \_port: (o), id:20, line:2, parent:work@dut
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: , id:21
      |vpiActual:
      \_logic_net: (o), id:17, line:6
        |vpiName:o
        |vpiFullName:work@dut.o
        |vpiNetType:48
  |vpiContAssign:
  \_cont_assign: , id:38, line:7
    |vpiRhs:
    \_part_select: , id:39, line:7
      |vpiConstantSelect:1
      |vpiLeftRange:
      \_constant: , id:40, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:3
      |vpiRightRange:
      \_constant: , id:41, line:7
        |vpiConstType:7
        |vpiSize:32
        |INT:1
    |vpiLhs:
    \_ref_obj: (o), id:42
      |vpiName:o
      |vpiFullName:work@dut.o
      |vpiActual:
      \_logic_net: (o), id:17, line:6
  |vpiNet:
  \_logic_net: (a), id:13, line:5, parent:work@dut
  |vpiNet:
  \_logic_net: (o), id:17, line:6
===================`

rkapuscik, 2020-06-02:
Sorry, the problem is not that they're not printed, but that it does not reference the `a` object. Just checked with latest master (https://github.com/alainmarcel/Surelog/commit/68e5ac69df4b8a1c33e5199d489f3359d6cb15c0) and for the code at [Verilator frontend](https://github.com/antmicro/verilator/blob/uhdm-verilator/src/UhdmAst.cpp#L1250) with some debug prints added we get:
```
Object: work_dut of type 32              @ 1 in ../tests/PartSelect/dut.v
Object:  of type 8               @ 6 in ../tests/PartSelect/dut.v
Object:  of type 42              @ 6 in ../tests/PartSelect/dut.v
        Entering Part select
Object:  of type 7               @ 6 in ../tests/PartSelect/dut.v
        ^ Left range
Object:  of type 7               @ 6 in ../tests/PartSelect/dut.v
        ^ Right range
Object:  of type 8               @ 6 in ../tests/PartSelect/dut.v
        No item returned
        Part select end
```
Object of type 8 at line 6 is `vpiContAsssign`, which the `vpiParent` in part select points to instead of a `vpiRefObj` to `a`.

alaindargelas, 2020-06-02:
Got it

PR #465 (2020-06-03): diff uhdm line count, preserve part_select parent
Fixes #462
closed 2020-06-03 by alaindargelas