chipsalliance/Surelog #515: Conda package build broken

tgorochowik, 2020-06-16:
Conda package (linux build) started failing after introducing this change: https://github.com/alainmarcel/Surelog/commit/e63ebbecd#diff-af3b638bc2a3e6c650974192a53c7291R45-R50

The error we get is:
```
CMake Error at /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/surelog_1592301850162/_build_env/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
3777  Could NOT find Python3 (missing: Python3_EXECUTABLE Interpreter) (found
3778  suitable version "3.7", minimum required is "3.3")
3779Call Stack (most recent call first):
3780  /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/surelog_1592301850162/_build_env/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
3781  /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/surelog_1592301850162/_build_env/share/cmake-3.17/Modules/FindPython/Support.cmake:2398 (find_package_handle_standard_args)
3782  /tmp/really-really-really-really-really-really-really-really-really-really-really-really-really-long-path/conda/conda-bld/surelog_1592301850162/_build_env/share/cmake-3.17/Modules/FindPython3.cmake:311 (include)
3783  CMakeLists.txt:45 (find_package)
```

The whole log can be found in the [conda-packages CI](https://travis-ci.com/github/SymbiFlow/conda-packages/jobs/349642231)

@hs-apotell could you please take a look at this?

Ref: https://github.com/SymbiFlow/conda-packages/issues/105
CC: @hzeller

hs-apotell, 2020-06-17:
'python3-dev' is missing in the list of packages called out for installation. Could you include that and report back please.

alaindargelas, 2020-06-24:
@tgorochowik  have you tried the suggestion above?

hzeller, 2020-06-24:
I have various experiments around
 https://github.com/SymbiFlow/conda-packages/pull/106 (fudging the Python issue, still missing a library)
 https://github.com/SymbiFlow/conda-packages/pull/111 (attempt to proper fix Python issue, but guessing how the conda environment works makes it hard)

Long story short, someone better knowing Conda should do this proper :)

tgorochowik, 2020-06-26:
Resolved in https://github.com/SymbiFlow/conda-packages/pull/107

closed 2020-06-26 by tgorochowik