chipsalliance/Surelog #527: Unpacked arrays are not recognized correctly

rkapuscik, 2020-06-22:
Seen at https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/Ibex/ibex/rtl/ibex_core.sv#L188.
```verilog
  logic [33:0] csr_pmp_addr [PMPNumRegions];
  pmp_cfg_t    csr_pmp_cfg  [PMPNumRegions];
  logic        pmp_req_err  [PMP_NUM_CHAN];
```
`csr_pmp_cfg` is stored in UHDM correctly:
```
|vpiVariables:                                                    
\_array_var: (csr_pmp_cfg), parent:u_ibex_core
  |vpiArrayType:1
  |vpiName:csr_pmp_cfg
  |vpiFullName:work@core_ibex_tb_top.dut.u_ibex_core.csr_pmp_cfg
  |vpiRandType:1
  |vpiVisibility:1
  |vpiSize:4
  |vpiReg:
  \_struct_var: , line:188, parent:csr_pmp_cfg
    |vpiFullName:work@core_ibex_tb_top.dut.u_ibex_core.csr_pmp_cfg
    |vpiTypespec:
    \_struct_typespec: (pmp_cfg_t), line:212
```
but both `csr_pmp_addr` and `pmp_req_err` are stored only as logic nets:
```
\_logic_net: (pmp_req_err), line:189, parent:u_ibex_core
  |vpiName:pmp_req_err
  |vpiFullName:work@core_ibex_tb_top.dut.u_ibex_core.pmp_req_err
  |vpiNetType:36
```

rkapuscik, 2020-06-22:
Another test case:
```verilog
module dut;

  typedef struct packed {
     bit   [7:0]   addr;
     bit   [7:0]   data;
  } mem_s;

  mem_s mem;
  mem_s memArr [20];
  mem_s memMulti [20][30];
endmodule
```
```
|vpiVariables:
\_struct_var: (memArr), line:13, parent:work@dut
  |vpiName:memArr
  |vpiFullName:work@dut.memArr
  |vpiTypespec:
  \_struct_typespec: (mem_s), line:6
|vpiVariables:
\_struct_var: (memMulti), line:14, parent:work@dut
  |vpiName:memMulti
  |vpiFullName:work@dut.memMulti
  |vpiTypespec:
  \_struct_typespec: (mem_s), line:6
```

alaindargelas, 2020-06-23:
This is fixed:

module dut;

  typedef struct packed {
     bit   [7:0]   addr;
     bit   [7:0]   data;
  } mem_s;

  mem_s mem;
  mem_s memArr [20];
  mem_s memMulti [20][30];
endmodule

Still working on:

  logic [33:0] csr_pmp_addr [PMPNumRegions];
  pmp_cfg_t    csr_pmp_cfg  [PMPNumRegions];
  logic        pmp_req_err  [PMP_NUM_CHAN];

alaindargelas, 2020-06-24:
Fixed by #531

closed 2020-06-24 by alaindargelas