chipsalliance/Surelog #544: Gate primitives not supported in UHDM Surelog

anadishukla, 2020-06-25:
Verilog has built-in gate primitives like and,or,nand,xor etc. 
See below example: 
module LogicGates(a,b,y1,y2,y3,y4,y5,y6,y7);
	input a,b;
	output y1;//y2,y3,y4,y5,y6,y7;
	and(y1,a,b);
	//or(y2,a,b);
	//not(y3,a);
	//nand(y4,a,b);
	//nor(y5,a,b);
	//xor(y6,a,b);
	//xnor(y7,a,b);
endmodule
**UHDM output :** 
Actual Problematic Output --> Verilog Code  : and_gate_primitive.txt, UHDM output is : uhdm_exp.txt
Expected Output --> Verilog Code : and_gate.txt, UHDM output is : uhdm_actual.txt

[uhdm_exp.txt](https://github.com/alainmarcel/Surelog/files/4829989/uhdm_exp.txt)
[uhdm_actual.txt](https://github.com/alainmarcel/Surelog/files/4829990/uhdm_actual.txt)
[and_gate_primitive.txt](https://github.com/alainmarcel/Surelog/files/4830028/and_gate_primitive.txt)
[and_gate.txt](https://github.com/alainmarcel/Surelog/files/4830029/and_gate.txt)

Gate level primitives are supported in YOSYS and other industry standard EDA tools. 
But seems it is not supported in UHDM VPI interpreted output. 

In case more details are required, please inform me.

alaindargelas, 2020-06-25:
Correct, UHDM does not support them either yet, it is not in the priority list to support the designs we need to support first.
Can you pick up the task of creating the model in UHDM first?
The models on page 1012 of the 2017 standard need to be captured properly.
Surelog will need some changes too to propagate them in UhDM. Surelog has an internal representation of the gates already, it needs the translation step, I can also guide you to implement that.

anadishukla, 2020-06-28:
I am investigating same and will get back to you by end of next week.

alaindargelas, 2020-08-05:
I should be done shortly with this task

PR #719 (2020-08-07): Initial support of gate level
Fixes #544.

@anadishukla, please confirm.

Still need to model delays, and input/output primterm for transistor level and bufn/notn
closed 2020-08-07 by alaindargelas