chipsalliance/Surelog #545: BAD_SYMBOL in UHDM o/p while using Concatenations in LHS

anadishukla, 2020-06-25:
Tried same example with yosys and industry standard simulator and it is functional. 

Attached example and UHDM output: 
[multiBitAdder.txt](https://github.com/alainmarcel/Surelog/files/4830142/multiBitAdder.txt)
[UHDM_adder_op.txt](https://github.com/alainmarcel/Surelog/files/4830143/UHDM_adder_op.txt)

Please help us to rectify BAD_SYMBOL error.

alaindargelas, 2020-06-25:
I'm not seeing that problem, can you update?
On the other hand I see 2 other issues: 
- the ranges are not captured correctly as the wire type is implicit
- the vpiFullName of cout and sum does not show in the assigment.

|uhdmtopModules:
\_module: work@MultibitAdder (work@MultibitAdder), file:tests/ScratchPad.sv, line:1
  |vpiDefName:work@MultibitAdder
  |vpiName:work@MultibitAdder
  |vpiPort:
  \_port: (a), line:1, parent:work@MultibitAdder
    |vpiName:a
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (a), line:1, parent:work@MultibitAdder
        |vpiName:a
        |vpiFullName:work@MultibitAdder.a
  |vpiPort:
  \_port: (b), line:1, parent:work@MultibitAdder
    |vpiName:b
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (b), line:1, parent:work@MultibitAdder
        |vpiName:b
        |vpiFullName:work@MultibitAdder.b
  |vpiPort:
  \_port: (cin), line:1, parent:work@MultibitAdder
    |vpiName:cin
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (cin), line:1, parent:work@MultibitAdder
        |vpiName:cin
        |vpiFullName:work@MultibitAdder.cin
  |vpiPort:
  \_port: (sum), line:1, parent:work@MultibitAdder
    |vpiName:sum
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (sum), line:1, parent:work@MultibitAdder
        |vpiName:sum
        |vpiFullName:work@MultibitAdder.sum
  |vpiPort:
  \_port: (cout), line:1, parent:work@MultibitAdder
    |vpiName:cout
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (cout), line:1, parent:work@MultibitAdder
        |vpiName:cout
        |vpiFullName:work@MultibitAdder.cout
  |vpiContAssign:
  \_cont_assign: , line:6
    |vpiRhs:
    \_operation: , line:6
      |vpiOpType:24
      |vpiOperand:
      \_operation: , line:6
        |vpiOpType:24
        |vpiOperand:
        \_ref_obj: (a), line:6
          |vpiName:a
          |vpiFullName:work@MultibitAdder.a
          |vpiActual:
          \_logic_net: (a), line:1, parent:work@MultibitAdder
        |vpiOperand:
        \_ref_obj: (b), line:6
          |vpiName:b
          |vpiFullName:work@MultibitAdder.b
          |vpiActual:
          \_logic_net: (b), line:1, parent:work@MultibitAdder
      |vpiOperand:
      \_ref_obj: (cin), line:6
        |vpiName:cin
        |vpiFullName:work@MultibitAdder.cin
        |vpiActual:
        \_logic_net: (cin), line:1, parent:work@MultibitAdder
    |vpiLhs:
    \_operation: , line:6
      |vpiOpType:33
      |vpiOperand:
      \_ref_obj: (cout), line:6
        |vpiName:cout
        |vpiActual:
        \_logic_net: (cout), line:1, parent:work@MultibitAdder
      |vpiOperand:
      \_ref_obj: (sum), line:6
        |vpiName:sum
        |vpiActual:
        \_logic_net: (sum), line:1, parent:work@MultibitAdder
  |vpiNet:
  \_logic_net: (a), line:1, parent:work@MultibitAdder
  |vpiNet:
  \_logic_net: (b), line:1, parent:work@MultibitAdder
  |vpiNet:
  \_logic_net: (cin), line:1, parent:work@MultibitAdder
  |vpiNet:
  \_logic_net: (sum), line:1, parent:work@MultibitAdder
  |vpiNet:
  \_logic_net: (cout), line:1, parent:work@MultibitAdder

alaindargelas, 2020-06-26:
Sub problems fixed by #552, original problem not reproducible.

closed 2020-06-26 by alaindargelas