chipsalliance/Surelog #568: Test Surelog against XilinxUnisimLibrary

mithro, 2020-06-30:
I would like to know if Surelog is able to parse the files found in the library at https://github.com/SymbiFlow/XilinxUnisimLibrary

alaindargelas, 2020-06-30:
Are they in sv-test?

alaindargelas, 2020-06-30:
It does parse all of them except 2 syntax errors of the same type in a specify block.

https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/XilinxUnisimLibrary/Unisim.log

[  FATAL] : 0
[ SYNTAX] : 2
[  ERROR] : 4
[WARNING] : 0
[   NOTE] : 256

closed 2020-06-30 by alaindargelas

mithro, 2020-06-30:
See https://github.com/SymbiFlow/sv-tests/issues/903 for the sv-test issue

mithro, 2020-07-22:
@alaindargelas - The upstream sim models has been updated with the missing `verilog/src/glbl.v` file which should make it possible to get further along in the parsing.