chipsalliance/Surelog #590: Parameter values are no longer elaborated

rkapuscik, 2020-07-08:
For the following test case: https://github.com/alainmarcel/uhdm-integration/blob/master/tests/OnePackage/dut.sv
UHDM no longer contains the value of the parameter, but the parameter itself: https://travis-ci.com/github/antmicro/uhdm-integration/jobs/358711569#L2749.
Is this an intended change (I can work around that by using `vpiFullname` to get the hierarchy) or an uncaught regression?

alaindargelas, 2020-07-09:
Yes, that was a regression when hadling more complex parameter, I was not binding to the value and keeping the parameter formal defintion. I changed the behavior that for trivial paramaters, I do the binding and only for more complex forms, I keep the algebraic form.
I added your test in my regression.

alaindargelas, 2020-07-09:
Fixed by #592

closed 2020-07-09 by alaindargelas