chipsalliance/Surelog #595: vpiParent is missing in vpiPartSelect

rkapuscik, 2020-07-10:
When parsing following line:
https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/Earlgrey_0_1/src/lowrisc_prim_all_0.1/rtl/prim_prince.sv#L81
UHDM looks as follows:
```
|vpiStmt:
\_assignment: , line:81, parent:p_pre_round_xor
  |vpiOpType:82
  |vpiBlocking:1
  |vpiLhs:
  \_bit_select: (data_state), line:81
    |vpiName:data_state
    |vpiFullName:work@prim_prince.p_pre_round_xor.data_state
    |vpiIndex:
    \_constant: , line:81
      |vpiConstType:7
      |vpiDecompile:0
      |vpiSize:32
      |INT:0
  |vpiRhs:
  \_var_select: , line:81, parent:PRINCE_ROUND_CONST
    |vpiFullName:prim_cipher_pkg::PRINCE_ROUND_CONST
    |vpiIndex:
    \_constant: , line:81
      |vpiConstType:7
      |vpiDecompile:0
      |vpiSize:32
      |INT:0
    |vpiIndex:
    \_part_select:~               <-- missing information
      |vpiConstantSelect:1
      |vpiLeftRange:
      \_operation: , line:81
        |vpiOpType:11
        |vpiOperand:
        \_ref_obj: (DataWidth), line:81
          |vpiName:DataWidth
        |vpiOperand:
        \_constant: , line:81
          |vpiConstType:7
          |vpiDecompile:1
          |vpiSize:32
          |INT:1
      |vpiRightRange:
      \_constant: , line:81
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
```
`vpiParent` relation for the part select is missing (not only not printed, it does not return a handle).

rkapuscik, 2020-07-10:
Looking again, it seems that it's missing because the parent would be a bit select which does not have its proper node. I'm not sure what the handling should be in this case.

alaindargelas, 2020-07-11:
Fixed by #600

closed 2020-07-11 by alaindargelas