chipsalliance/Surelog #631: Add Travis (and Kokoro?) build for Windows

mithro, 2020-07-15:
_Originally posted by @alaindargelas in https://github.com/alainmarcel/Surelog/issues/239#issuecomment-658534307_

> Windows support achieved, Thanks @hs-apotell!
> Do we need a travis Windows build?

We should have CI running on Windows so that we don't break the Windows build accidentally.

hs-apotell, 2020-07-16:
Absolutely, Yes. I started working on it and then got stuck with the requirement for tcl for windows. It was a while back and might have gotten stale.

Related issue - https://github.com/alainmarcel/UHDM/issues/229

hs-apotell, 2020-10-15:
We have a basic windows CI support working. However, we are running into some issues.

1. We have about 16 test failures. Attached are the logs for failing tests and an excel document summarizing them.
2. Our builds keep timing out on travis. This is happening for Unix builds also - which is more troubling because the same builds under @alaindargelas account seem to be finishing much faster.

Any assistance in identifying the root causes and/or fixing these issues or sharing any pointers helping us do it is highly appreciated.
[logdir.tar.gz](https://github.com/alainmarcel/Surelog/files/5383765/logdir.tar.gz)
[surelog_test_diff.xlsx](https://github.com/alainmarcel/Surelog/files/5383766/surelog_test_diff.xlsx)

umarcor, 2020-10-16:
I tried creating a GitHub Actions workflow: https://github.com/umarcor/Surelog/blob/ci/add-gha-workflow/.github/workflows/test.yml

However, I'm working on Windows, and it seems that this repository contains several files which have the same name but with different capitalisation: https://github.com/umarcor/Surelog/commit/752b17dfd2d9c0a014f50408036375ffdc49e54f. That's unfortunate, because it produces many conflicts when trying to rebase commits.

On the other hand, when cmake is executed, the following error is produced:

```
 CMake Error at third_party/UHDM/CMakeLists.txt:29 (add_subdirectory):
  The source directory

    /home/runner/work/Surelog/Surelog/third_party/UHDM/third_party/capnproto

  does not contain a CMakeLists.txt file.

CMake Error at third_party/UHDM/CMakeLists.txt:249 (install):
  install TARGETS given target "capnp" which does not exist.
```

Hence, it seems that an issue is coming from a submodule.

umarcor, 2020-10-16:
Apart from that, checks in https://github.com/alainmarcel/Surelog/commits/master show that GitHub Pages is being used, but https://alainmarcel.github.io/Surelog/ seems to be empty. There is no `gh-pages` branch in the repo, nor any `doc` subdir.

alaindargelas, 2020-10-16:
For the missing submodule files, did the git submodule update --init --recursive
work correctly?

alaindargelas, 2020-10-16:
@hs-apotell for the tests that have very few errors compared to Linux,
I think this function:
std::string StringUtils::getLineInString(std::string_view bulk,
                                         unsigned int line); 

is not doing its job properly on Windows (count the lines).

alaindargelas, 2020-10-16:
@hs-apotell one of your Gold files is actually obsolete:
golden_dir_logs/Sky130Udp.log
You need to get the latest.

hs-apotell, 2020-10-16:
https://github.com/anadishukla/Surelog/blob/master/.github/workflows/cmake.yml

Here's our attempt to get Unix build working with Github actions. Something very weird happening. The process rebuilds everything between each command and so ends up taking over 90 minutes before it ultimately fails.

Could someone more experienced with Makefile do a quick review of the changes in here - https://github.com/anadishukla/Surelog/blob/master/Makefile

hs-apotell, 2020-10-16:
> that this repository contains several files which have the same name but with different capitalisation

@umarcor Yes but all those issues are in tests and we haven't yet been able to build successfully to stumble into this issue.

> For the missing submodule files, did the git submodule update --init --recursive work correctly?

@alaindargelas There is no solution for duplicate name problem for the platform. The files just have to be named differently. Case difference alone is not acceptable on Windows.

> I think this function:
> std::string StringUtils::getLineInString(std::string_view bulk, unsigned int line);
> is not doing its job properly on Windows (count the lines).

@alaindargelas Appreciate the input. We had just started looking into this earlier today.

> one of your Gold files is actually obsolete:
Yes I merged the latest earlier yesterday.

hs-apotell, 2020-10-16:
Most builds are IO bound. Have we tried moving away from Makefile and using Ninja instead as the build generator option for Linux? The later is supposed to be capable of building in parallel so we should see noticeable improvement in build times - at least, in theory.

hs-apotell, 2020-10-16:
For reference, here's the list of files that would need to be renamed. They are all part of Yosys tests.
```
        third_party/tests/YosysTests/misc/scripts/select_%C.ys
        third_party/tests/YosysTests/misc/scripts/select_%M.ys
        third_party/tests/YosysTests/misc/scripts/select_%x_%D.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A_eq.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A_less.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A_lesseq.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A_more.ys
        third_party/tests/YosysTests/misc/scripts/select_add_A_moreeq.ys
        third_party/tests/YosysTests/misc/scripts/select_one_elem_for__C.ys
        third_party/tests/YosysTests/misc/scripts/select_one_elem_for__D.ys
        third_party/tests/YosysTests/misc/scripts/select_one_elem_for__M.ys
```

alaindargelas, 2020-10-16:
Just submit a PR and delete them.These files are of no use for Surelog regression.

umarcor, 2020-10-16:
@alaindargelas, my bad, submodules were not being cloned recursively. I fixed that yesterday but then some mirror availability issues (unrelated to Surelog) prevented the CI from running: https://github.com/umarcor/Surelog/runs/1262771362?check_suite_focus=true#step:4:21.

Fortunately, it's working now: #845 https://github.com/umarcor/Surelog/runs/1264504759?check_suite_focus=true

> Yes but all those issues are in tests and we haven't yet been able to build successfully to stumble into this issue.

@hs-apotell see #845

umarcor, 2020-10-16:
> We have a basic windows CI support working.

@hs-apotell, I could not find the sources of the current Windows CI procedure. Are those public?

My approach on GitHub Actions is using MSYS2. That'll soon be the default `bash` in the `windows-latest` environment provided. See https://github.com/umarcor/Surelog/blob/ci/gha-windows/.github/workflows/test.yml (which is based on #845).

hs-apotell, 2020-10-16:
> I could not find the sources of the current Windows CI procedure. Are those public?

@umarcor We were still attempting it on travis. Here's the last updated revision - https://github.com/anadishukla/Surelog/blob/master/.travis.yml and here's the last unsuccessful run https://travis-ci.com/github/anadishukla/Surelog/builds/189897381

We were having issues with timeouts for linux builds and test failures for windows.

umarcor, 2020-10-16:
> We were having issues with timeouts for linux builds and test failures for windows.

Interesting. In GitHub Actions Linux builds seem to finish in 30-35 min: https://github.com/umarcor/Surelog/actions

hs-apotell, 2020-10-16:
@umarcor I have also been playing around with Github Actions and here's my attempt at it. 
https://github.com/anadishukla/Surelog/blob/actions/.github/workflows/cmake.yml

I just compared my attempt with yours and I don't see any meaningful differences and yet my attempt doesn't succeed. In all my attempts so far, every make command ends up rebuilding the whole source code again. And, after ~90 minutes it just fails. Here's the link to the most recent attempt - https://github.com/anadishukla/Surelog/actions/runs/311323253

umarcor, 2020-10-16:
@hs-apotell, what's the motivation for the following?

- https://github.com/anadishukla/Surelog/blob/actions/.github/workflows/cmake.yml#L9-L11
- https://github.com/anadishukla/Surelog/blob/actions/.github/workflows/cmake.yml#L16-L18
- https://github.com/anadishukla/Surelog/blob/actions/.github/workflows/cmake.yml#L40-L41

> I just compared my attempt with yours and I don't see any meaningful differences and yet my attempt doesn't succeed.

I think that the only relevant difference is https://github.com/anadishukla/Surelog/blob/actions/.github/workflows/cmake.yml#L16-L17. Likely a different version of cmake is used.

- Mine uses cmake 3.17.0: https://github.com/umarcor/Surelog/runs/1265055624?check_suite_focus=true#step:5:12
- Yours uses cmake 3.18.4: https://github.com/anadishukla/Surelog/runs/1263687462?check_suite_focus=true#step:4:101

hs-apotell, 2020-10-16:
@umarcor I have minimal CI experience and literally zero on Github Actions itself. I have merely been following working examples to model mine. Some of it is mere _desperation_ in trying to narrow down what's wrong.

As it turns out setting the working directory is the problem. Between the following two runs -
https://github.com/anadishukla/Surelog/actions/runs/311323253 (succeeds, not just yet but much farther than I have had so far)
https://github.com/anadishukla/Surelog/actions/runs/310226185 (failed)

the only _major_ difference is the one that succeeds doesn't have the working directory set. This specific line was in the default CMake yml file generated by Github.

hs-apotell, 2020-10-16:
> Did you succeed in building/using Surelog locally on Windows 10?

Yes and No. The build itself succeeds but the tests fail. If you deliver on the CI, I am happy to go back to my happy world of programming and debugging :) I can focus on fixing the tests for windows.

I have been working off of a fork of Surelog - https://github.com/anadishukla/Surelog

umarcor, 2020-10-16:
Since the default `bash` is window-latest will soon be MSYS2's MINGW64 and because I'm familiar with building other tools in that shell, this is my attempt:

- https://github.com/umarcor/Surelog/blob/ci/gha-windows/.github/workflows/test.yml#L52-L92
- https://github.com/umarcor/Surelog/runs/1266684330?check_suite_focus=true

As you see, it fails at:

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
TCMALLOC_LIBRARY
    linked by target "surelog" in directory D:/a/Surelog/Surelog

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Any guess?

In https://github.com/anadishukla/Surelog/blob/master/.travis.yml, you seem to be using choco and visualstudio instead of GCC/clang. Is that the toolchain you tested locally?

hs-apotell, 2020-10-16:
> TCMALLOC_LIBRARY

Add an environment variable "NO_TCMALLOC=On" (or, optionally -DNO_TCMALLOC=On for cmake command line). We have kept this turned off for Windows for now.

> you seem to be using choco and visualstudio instead of GCC/clang. Is that the toolchain you tested locally?

Visual Studio + Ninja is what I have tested locally. Choco is just for installing dependencies which I am not doing locally - only for travis. For travis we were attempting to build it under MinGW bash shell (Git bash shell, to be precise) and we have verified locally that the builds do succeed in this environment.

For Windows, the preferred compiler would be VS/cl. Longer term, we would want to switch to using clang for all platforms (low on the long list of wishes).

Also, ideally we would switch to using Ninja for ubuntu builds as well to take full advantage of parallel builds. Also, we had to make a few changes to Makefile in the repository to keep the commands compatible across platforms. Check the file https://github.com/anadishukla/Surelog/blob/master/Makefile. In summary, remove the assumption that cmake is using Makefile as generator.

umarcor, 2020-10-16:
This is as far as I could get without futher modification in the codebase: https://github.com/umarcor/Surelog/runs/1266974896?check_suite_focus=true. I will keep Windows CI on hold until support for Windows is merged upstream.

hs-apotell, 2020-10-17:
The default cmake generator on Windows is Visual Studio and the Makefile is not authored to account for it. That was the reason to modify the Makefile as I mentioned in my earlier comment. You will need the Makefile here https://github.com/anadishukla/Surelog/blob/master/Makefile to get past the current error.

Give me next 24 hours and I will have a more complete solution with tests ready for CI support.

hs-apotell, 2020-10-17:
@umarcor You are good to go. With #849 change locally Windows builds successfully. I have provided detailed description of my environment as well.

@alaindargelas #849 lists a few regression test failures which _seems_ logical errors in code and not just platform differences. Could you please take a moment to either diagnose yourself or provide some direction to proceed. Let me know if you need full log files to assist.

alaindargelas, 2020-10-17:
@hs-apotell, @umarcor, let me know when I can merge the 2 DRAFT PRs.

I'll look at the black list of tests on Windows once you guys have a working solution that can be merged. Re-enabling the black list of tests should be easy and not blocking your check in.

hs-apotell, 2020-10-18:
@alaindargelas I have the split the draft PR into a few smaller more focused changes. These changes are numbered. Please merge them in incremental order. Also, one of the PR (3rd in the series) is expected to fail on regression tests. This change requires updating the golden logs. Please make sure to do that so 4th and 5th can be merged after they get green.

alaindargelas, 2020-10-18:
@hs-apotell, @umarcor, Somehow the build on Travis for Linux is not running anymore.
@hs-apotell, please fix the Makefile and cmake_gen.tcl for Linux, your change for Windows broke the Linux regression.
I did undo massively, not trying to find the root cause.

umarcor, 2020-10-19:
> @umarcor You are good to go. With #849 change locally Windows builds successfully. I have provided detailed description of my environment as well.

@hs-apotell, I rebased #847 on top of master. That is failing due to some issue with the paths: https://github.com/alainmarcel/Surelog/runs/1273143748?check_suite_focus=true#step:6:605. Then, I rebased the same CI addition on top of #849, but I get the same result: https://github.com/umarcor/Surelog/runs/1273157374?check_suite_focus=true#step:6:604.

> let me know when I can merge the 2 DRAFT PRs.

@alaindargelas, this will likely need a few iterations yet. We'll let you know explicitly when it's ready.

hs-apotell, 2020-10-19:
@umarcor The reported issue on windows build is more in Alain's court - something to do with tcl.
@alaindargelas Do you mind taking a look please!

hs-apotell, 2020-10-19:
@umarcor I just noticed that for Windows you still have msys2 as the default shell. Shouldn't that be bash?

umarcor, 2020-10-19:
> @umarcor I just noticed that for Windows you still have msys2 as the default shell. Shouldn't that be bash?

The default bash in `windows-latest` is MSYS2: https://github.com/actions/virtual-environments/pull/1648. However, the built-in installation is flawed: https://github.com/actions/virtual-environments/issues/1572. That's why msys2/setup-msys2 is used instead.

hs-apotell, 2020-10-19:
I haven't tested the flow with msys in particular. I have been using Gitbash locally. Would that be an option or is that too much to deviate from the norm?

umarcor, 2020-10-19:
It might be done. Indeed, both options can be tested. However, using Git Bash requires installing all the dependencies manually, because there is no package manager available. Hence `envvar.txt` in #849 would need to be replaced with the sources of each of those tools. Conversely, in MSYS2 dependencies are installed through pacman.

Overall, Git Bash is not expected to be used for anything other than using git. See https://github.com/actions/virtual-environments/pull/1435#issue-470068824.

alaindargelas, 2020-10-19:
The file path seems very wrong:
**/d/a/**Surelog/Surelog/third_party/UHDM**/D:/a/**Surelog/Surelog/third_party/UHDM/model_gen/file_utils.tcl 
In model_gen.tcl this file is sourced from the executable (model_gen.tcl) current location. The current location in Windows returns that funny looking path.

umarcor, 2020-10-19:
I tried the following based on https://github.com/alainmarcel/UHDM/blob/master/model_gen/model_gen.tcl:

```
variable myLocation [file normalize [info script]]

proc exec_path {} {
    variable myLocation
    return [file dirname $myLocation]
}

puts $myLocation
puts [exec_path]
```

```
# tclsh test.tcl
T:/Surelog/test.tcl
T:/Surelog
```

Is that the expected output?

alaindargelas, 2020-10-19:
Yes, now, can you try this more in the context of the build?
Execute that mini script in place of the invocation of model_gen.tcl

umarcor, 2020-10-19:
In fact, I believe the solution is more straightforward. I tried that locally, where `mingw-w64-x86_64-tcl` is installed. That was not installed in CI, so the script was probably using some other tcl shell. Now, I added the package explicitly, and it seems to pass that step. It fails later, but I'm not sure about the reason: https://github.com/alainmarcel/Surelog/runs/1273557211?check_suite_focus=true#step:6:691.

The only obvious error I saw is https://github.com/alainmarcel/Surelog/runs/1273557211?check_suite_focus=true#step:6:323. Is java a requirement?

hs-apotell, 2020-10-19:
> Is java a requirement?

Yes.

umarcor, 2020-10-19:
Java is available in the `windows-latest` environment, and `JAVA_HOME` envvar is set. I added it to the PATH. cmake finds java, as shown in https://github.com/umarcor/Surelog/runs/1274055773?check_suite_focus=true#step:7:293. However, it is still failing later because it cannot find `java`: https://github.com/umarcor/Surelog/runs/1274055773?check_suite_focus=true#step:7:512.

hs-apotell, 2020-10-19:
Try replacing Java in [CMakeLists.txt](https://github.com/alainmarcel/Surelog/blob/f2f6e0784fc12e14ee75b9265ee584140caecdd3/CMakeLists.txt#L179) with `${Java_JAVA_EXECUTABLE}`

```
COMMAND java -jar ${ANTLR_JAR_LOCATION} -Werror -Dlanguage=Cpp
```

becomes
```
COMMAND ${Java_JAVA_EXECUTABLE} -jar ${ANTLR_JAR_LOCATION} -Werror -Dlanguage=Cpp
```

umarcor, 2020-10-21:
@hs-apotell I did that and I rebased on top of master. Now I'm getting an error as soon as `cmake --build build -j 2` is executed: https://github.com/umarcor/Surelog/runs/1284208553?check_suite_focus=true#step:7:108. That is, I think it is not even reaching the point it got to two days ago.

hs-apotell, 2020-10-21:
@umarcor You are missing a whole lot of environment settings. CMake is defaulting to using VS to build. We need to switch that to use Ninja.

From my past comment [here](https://github.com/alainmarcel/Surelog/pull/849), here are the list of environment variables you need to setup.

```
export PREFIX=`pwd`/install
export CMAKE_GENERATOR=Ninja
export JAVA_HOME=/d/Java/jre-1.8.0_241
export PATH=/d/cmake/3.18.4/bin:$PATH
export PATH=/d/GnuMake/3.81/bin:$PATH
export PATH=/d/mingw64/bin:$PATH
export PATH=/d/Python/3.8.0:$PATH
export PATH=/d/swig/3.0.12:$PATH
export PATH=$JAVA_HOME/bin:$PATH
export CC=cl
export CXX=cl
export NO_TCMALLOC=On
source source_bat.bash 'D:\\MSVS\2019\\Community\\VC\\Auxiliary\\Build\\vcvars64.bat'
cmake --version
make --version
swig -version
java -version
python --version
ninja --version
echo 'puts [info patchlevel];exit 0' | tclsh
$CC
$CXX
```

In the mean time I will try and build the project under MSYS locally so I can iterate faster and see what other, if any, hoops we need to jump.

alaindargelas, 2020-10-21:
Why Ninja is a must?Let's try to be as least disruptive as possible during this port to Windows.The Linux build is completely documented on what environment is required in the Makefile and the .travis used to serve as document as well. I hope this is still true after all the recent changes, if not submit changes in the Readme. For windows I expect the same, there should not be any discovery by trial and error for someone trying to build. All dependencies should be already documented in the Readme and whatever github action file.

hs-apotell, 2020-10-21:
The default for CMake on Windows is VS which is a multi-configuration build tool. The default on Linux is Makefile which is a single configuration build tool. This difference makes it non-trivial to write build commands that would work across both platforms. The Makefile will have to provide `--config Debug|Release` specifically for every command (which unfortunately isn't compatible for Linux). That's just one difference that I know of since I ran into it very early in developing a solution for Windows.

I chose Ninja because that is what M$ defaulted to. If you open a CMake project in VS, it defaults to using Ninja as the build generator. Even M$ doesn't use the CMake default (i.e. VS solutions) to build CMake projects. I equate that to a lost battle before it starts.

All that said, I recommend moving to using Ninja as the default build generator across platforms. For whatever little I have seen of CMake projects on Github, most repositories supporting cross-platform development happen to chose Ninja over Makefile (or VS solutions).

umarcor, 2020-10-21:
> You are missing a whole lot of environment settings. CMake is defaulting to using VS to build. We need to switch that to use Ninja.

@hs-apotell, yet, previous runs using VS were going further than yesterday. That was surprising for me.

> From my past comment [here](https://github.com/alainmarcel/Surelog/pull/849), here are the list of environment variables you need to setup.

As said, I would need instructions about where to get those dependencies and where to extract/install them. Setting the envvars alone is likely to fail.

Nevertheless, on MSYS2/MINGW64 most of the dependencies are installed through the package manager. Hence, cmake, make and swig are already available in the PATH. Java was explicitly added too (as we discussed above). Now, I added the following:

```sh
export CMAKE_GENERATOR=Ninja
export CC=gcc
export CXX=g++
export NO_TCMALLOC=On
```

It runs for ~9min before failing: https://github.com/umarcor/Surelog/runs/1287140321?check_suite_focus=true#step:7:1389

> source source_bat.bash 'D:\\MSVS\2019\\Community\\VC\\Auxiliary\\Build\\vcvars64.bat'

What's the purpose of this step?

> In the mean time I will try and build the project under MSYS locally so I can iterate faster and see what other, if any, hoops we need to jump.

See https://github.com/umarcor/Surelog/blob/ci/gha-windows/.github/workflows/test.yml#L79

umarcor, 2020-10-21:
> Let's try to be as least disruptive as possible during this port to Windows. The Linux build is completely documented on what environment is required in the Makefile and the .travis used to serve as [...]

@alaindargelas, the Linux job in GitHub Actions is equivalent to the `.travis.yml`. The steps in https://github.com/alainmarcel/Surelog/blob/master/.github/workflows/test.yml#L24-L51 can be copied and pasted on any Ubuntu host, as long as Python is installed. My porposal is to achieve the same result on Windows.

However, building on Windows was not tested/supported until a few days ago, and there is no documented procedure for installing the dependencies, which I could find. That's why trial and error is required, as it allows us to iterate. Precisely, I am unsure about the following:

- Which are the dependencies of this project? I don't mean the list of packages for some platform, but the list of projects, by name.
- Which compilers are supported? GCC? LLVM/clang? VS?

My proposal is to use MINGW64's GCC or LLVM/clang, and all MSYS2 packages, instead of installing each dependency from a different source. That will allow to rewrite the Windows build procedure as a PKGBUILD build file. That file can be used for testing and also for providing pre-built packages that users can install.

> With #849 change locally Windows builds successfully. I have provided detailed description of my environment as well. 

Would you mind pointing me to that description? I could not find it.

umarcor, 2020-10-21:
> All that said, I recommend moving to using Ninja as the default build generator across platforms. For whatever little I have seen of CMake projects on Github, most repositories supporting cross-platform development happen to chose Ninja over Makefile (or VS solutions).

@hs-apotell I have no preference. If VS is what you want to use, that's ok. However, then the list of sources to get dependencies from needs to be documented. Conversely, using MSYS2/MINGW64 should allow to mimic the workflow on Linux: make, gcc, cmake... All should behave almost as in a POSIX compatible environment/shell.

alaindargelas, 2020-10-21:
In the spirit of FOSS, we should not mandate the VS compiler for Windows, @hzeller @mithro  what do you think?

mithro, 2020-10-21:
I would like to say that supporting the Microsoft Visual C++ compiler is a "nice to have", while the open source compilers on Windows are a "must have".

hs-apotell, 2020-10-22:
In the spirit of general consensus, this is what I tried today locally.

* MSYS + cl
* MSYS + gcc
* MinGW + gcc

None of the above succeeded. Each of them had different issues. Before I invest more time into this, what is the preferred environment we wish to support. I can provide more details on each of them if interested.

@alaindargelas The specific issue I run into with building Visual Studio solution directly is that it needs `--config Debug|Release` option. The effect of this is that the outputs are generated in `bin/[Debug|Release]` and not under `bin` as expected. All other build actions downstream fail because they can't find the binaries. I tried working around this issue but the _expectation_ is hardcoded in too many places.

@umarcor Here's the [last attempt](https://travis-ci.com/github/anadishukla/Surelog/jobs/398755929) on building for Windows on Travis which was building successfully until running out of time during regression tests.

> It runs for ~9min before failing: https://github.com/umarcor/Surelog/runs/1287140321?check_suite_focus=true#step:7:1389

@umarcor Regarding the specific error in your last attempt on GHA, I can't seem to reproduce this issue locally. I always got past that step. However, I still haven't managed to get a successful build under MSYS with any combination of compilers and generators.

> What's the purpose of this step?

@umarcor source_bat.bash was pulled from [here](https://raw.githubusercontent.com/johnwason/source_bat/master/source_bat.bash). This step is required to setup all the necessary VC related environment variables in Gitbash console. You can find more information about this in the Travis build mentioned above.

> Would you mind pointing me to that description? I could not find it.

I already did it [here](https://github.com/alainmarcel/Surelog/pull/849#issue-505340196). Here's the [environment variable](https://github.com/alainmarcel/Surelog/files/5396427/envvar.txt) I had to source and then run the following commands.
```
source envvar.txt
make release
make install
make test_install
make test/unittest
make regression
```

> My proposal is to use MINGW64's GCC or LLVM/clang, and all MSYS2 packages, instead of installing each dependency from a different source.

I tried MinGW64 + GCC locally and didn't have much luck. I will spend more time to see if I can get a successful build here.

> If VS is what you want to use, that's ok.

For Windows, VS will be preferred development environment and cl will be the compiler. We have projects dependent on Surelog (and many other public repos) and they all come together in this environment. Pulling one out, building and maintaining a whole different environment is neither viable nor sustainable solution. That said, I am more than happy to assist getting multiple such development environment working. Clang is probably the most desired compiler because of it being available on all our supported platforms and is also supported on all the _default_ development environments across these platforms.

> However, then the list of sources to get dependencies from needs to be documented

That is exactly what I thought I did with providing the dump of environment variables and the build steps. In fact, we had this nearly working on Travis with few gotchas -
* We were using Gitbash (happens to be default on Travis for Windows builds)
* We were using Choclatey to install dependencies manually (Most of these are already installed on GHA VMs because of Windows platform being a first class citizen)

> we should not mandate the VS compiler for Windows

There is no mandate. This is the environment we had it working locally and so the hope _was_ that it would be easiest to replicate on Travis (and now GHA). The recommended environment i.e. VS + CMake + Ninja + cl is the default on Windows. Most windows only developers aren't necessarily familiar with MSYS or MinGW.

> Microsoft Visual C++ compiler is a "nice to have", while the open source compilers on Windows are a "must have".

+1 to that. We were taking the easiest way out to have a CI system. We chose to replicate what we had working locally.

umarcor, 2020-10-22:
> None of the above succeeded. Each of them had different issues. Before I invest more time into this, what is the preferred environment we wish to support. I can provide more details on each of them if interested.

I think we want to support MINGW64 + GCC, and some other build with VS and without any MSYS2 package. I would consider MSYS out of scope.

> @umarcor Here's the [last attempt](https://travis-ci.com/github/anadishukla/Surelog/jobs/398755929) on building for Windows on Travis which was building successfully until running out of time during regression tests.

I now added two jobs. One based on MSYS2 and another one using chocolatey.

> @umarcor Regarding the specific error in your last attempt on GHA, I can't seem to reproduce this issue locally. I always got past that step. However, I still haven't managed to get a successful build under MSYS with any combination of compilers and generators.

> @umarcor source_bat.bash was pulled from [here](https://raw.githubusercontent.com/johnwason/source_bat/master/source_bat.bash). This step is required to setup all the necessary VC related [...]

[I see](https://github.com/anadishukla/Surelog/blob/1c308b8f27e369903e1131b94f88d33fab5e815a/.travis.yml#L41)... Is using a bash shell a requirement? I would expect to use cmd o powershell instead. As said, the shell in Git for Windows (Gitbash) is really not meant to be used or relied upon. The proper way of using that bash is precisely using MSYS2's MINGW64, which is what the other job is about.

> I already did it [here](https://github.com/alainmarcel/Surelog/pull/849#issue-505340196). Here's the [environment variable](https://github.com/alainmarcel/Surelog/files/5396427/envvar.txt) I had to source and then run the following commands.

My bad. I thought that the quote was written by Alain, and that he was referring to some other description apart from yours.

> I tried MinGW64 + GCC locally and didn't have much luck. I will spend more time to see if I can get a successful build here.

Thanks.

> For Windows, VS will be preferred development environment and cl will be the compiler. We have projects dependent on Surelog (and many other public repos) and they all come together in this [...]

At this point, I think we agree on having two builds. One based on VS, which does not depend on MSYS2 at all. And another one based on MSYS2, which should not depend on any external package apart form Java.

> * We were using Choclatey to install dependencies manually (Most of these are already installed on GHA VMs because of Windows platform being a first class citizen)

I thought you had installed dependencies by using the respective installers/packages. If done through chocolatey, that's straightforward. Sorry about the misunderstanding. See https://github.com/alainmarcel/Surelog/blob/c6c9e48c3998842a731f725d1efa289c88d55750/.github/workflows/test.yml#L125-L164.

hs-apotell, 2020-10-22:
A little progress today. At least I know why the build is failing now.

On MinGW, pacman install version 4.x.x of Swig. Surelog needs version 3.x.x. Online pundits claim there is no way to downgrade or to pick a specific version of a package using pacman. I don't want to believe that but again too many seemed to be claiming it to be true. So, how do we proceed forward? 

@umarcor Here's my environment setup and this seems to work (assuming you can install the specific version of swig on GHA). This is for MinGW + GCC. Only two programs are referenced from outside - make (coming from msys) and java (installed on host windows)

I had to add `-m64 -Wa,-mbig-obj` to UHDM compile options because a few files are too big.

```
export PREFIX=`pwd`/install
export JAVA_HOME=/d/Java/jre-1.8.0_241
export PATH=$JAVA_HOME/bin:$PATH

export CC=gcc
export CXX=g++
export NO_TCMALLOC=On
export CMAKE_GENERATOR=Ninja

echo && where cmake && cmake --version
echo && where make && make --version
echo && where swig && swig -version
echo && where java && java -version
echo && where python && python --version
echo && where ninja && ninja --version
echo && where tclsh && echo 'puts [info patchlevel];exit 0' | tclsh
echo && where $CC && $CC --version
echo && where $CXX && $CXX --version
make release > make_release.log 2>&1
```

Here's the output of running `source <content above>` -

```
D:\msys64\20200903\mingw64\bin\cmake.exe
cmake version 3.17.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

D:\msys64\20200903\usr\bin\make.exe
GNU Make 4.3
Built for x86_64-pc-msys
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

D:\msys64\20200903\mingw64\bin\swig.exe

SWIG Version 4.0.2

Compiled with x86_64-w64-mingw32-g++ [x86_64-w64-mingw32]

Configured options: +pcre

Please see http://www.swig.org for reporting bugs and further information

D:\Java\jre-1.8.0_241\bin\java.exe
java version "1.8.0_241"
Java(TM) SE Runtime Environment (build 1.8.0_241-b07)
Java HotSpot(TM) 64-Bit Server VM (build 25.241-b07, mixed mode)

D:\msys64\20200903\mingw64\bin\python.exe
Python 3.8.5

D:\msys64\20200903\mingw64\bin\ninja.exe
1.10.1

D:\msys64\20200903\mingw64\bin\tclsh.exe
D:\msys64\20200903\usr\bin\tclsh.exe
8.6.10

D:\msys64\20200903\mingw64\bin\gcc.exe
gcc.exe (Rev1, Built by MSYS2 project) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

D:\msys64\20200903\mingw64\bin\g++.exe
g++.exe (Rev1, Built by MSYS2 project) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Attached is the output of the last run reporting the errors.
[make_release.log](https://github.com/alainmarcel/Surelog/files/5422039/make_release.log)

umarcor, 2020-10-22:
> On MinGW, pacman install version 4.x.x of Swig. Surelog needs version 3.x.x. Online pundits claim there is no way to downgrade or to pick a specific version of a package using pacman. I don't want to believe that but again too many seemed to be claiming it to be true. So, how do we proceed forward?

MSYS2 is a rolling distribution, just as Arch Linux. Hence, strictly, a single version of each package is supported. However, some previous versions are available in the data store: https://www.msys2.org/docs/package-management/#installing-a-specific-version-of-a-package-or-a-stand-alone-packages. Furthermore, PKGBUILD files are quite simple bash scripts: https://github.com/hdl/MINGW-packages/blob/upstream/mingw-w64-swig/PKGBUILD. Hence, it is possible to checkout an older version and build it.

Anyway, Swigh 4 has been around for 2-3 years, and branch 3 seems not to be updated since then. Is it not possible to fix Surelog?
 
> @umarcor Here's my environment setup and this seems to work (assuming you can install the specific version of swig on GHA). This is for MinGW + GCC. Only two programs are referenced from outside - make (coming from msys) and java (installed on host windows)

I will tray with an specific version of Swig.

> I had to add `-m64 -Wa,-mbig-obj` to UHDM compile options because a few files are too big.

I guess you did this in the submodule?

---

Could you have a look at the other job?

- https://github.com/alainmarcel/Surelog/blob/c6c9e48c3998842a731f725d1efa289c88d55750/.github/workflows/test.yml#L125-L164
- https://github.com/alainmarcel/Surelog/runs/1291278423?check_suite_focus=true

That's also using swig 4 (coming from chocolatey), and my guess is that is is also failing because of the missing UHDM compile options you mentioned.

umarcor, 2020-10-23:
> I guess you did this in the submodule?

I found: https://github.com/anadishukla/Surelog/tree/winbuilds

umarcor, 2020-10-23:
I tried building Swig 3.0.12 from sources and using @hs-apotell's fork of UHDM. Still getting the same `unknown signal` error:

- https://github.com/umarcor/Surelog/runs/1295635699?check_suite_focus=true
- https://github.com/umarcor/Surelog/commits/ci/gha-windows

hs-apotell, 2020-10-23:
@umarcor @alaindargelas So, I finally narrowed down the root cause of the error during model generation. In fact, it has nothing to do with model_gen.tcl script at all, at least not directly. The error was actually coming from its call to capnp.exe. So, it boils down to missing dll dependencies. capnp is dependent on the following dlls and are required to be in the same folder as the executable (or System32 folder on Windows).

* libstdc++.dll
* libstdc++-6.dll
* libwinpthread-1.dll
* libgcc_s_seh-1.dll

Once I made the files available capnp started working as expected. Now the important question - How to know where the dependencies come from i.e. what files are supposed to be copied and from where? Should this part of capnp project or UHDM or Surelog?

Do we know whether or not capnp even supports building/running under MSYS?

Another issue is that within tcl scripts all exec commands are branched for windows vs. unix. Under MSYS (or MinGW), the platform reported is windows and we use cmd to execute external processes. We are _always_ going to run into this dependency issue because the executables that are built under this environment have dependencies on files that are known only under this environment.

I don't feel like I have all the information needed to be able to successfully deliver a full build under MSYS/MinGW.

umarcor, 2020-10-23:
@hs-apotell `capnp` is available as a package in MSYS2: https://packages.msys2.org/package/mingw-w64-x86_64-capnproto?repo=mingw64. Shall we try installing that instead of building it?

> Under MSYS (or MinGW), the platform reported is windows and we use cmd to execute external processes.

A possible solution is to use `bash` for executing processes. That is, detect that MINGW environments behave as Linux in this regard.

Alternatively, `C:\msys64\mingw64\lib\gcc\x86-64-w64-mingw32` can be aded to the context of the call.

>I don't feel like I have all the information needed to be able to successfully deliver a full build under MSYS/MinGW.

What about the non MSYS2 approach?

hs-apotell, 2020-10-23:
> Shall we try installing that instead of building it?

I don't think this would be a good idea because we might run into conflicting version problem. Surelog/UHDM links to capnp libraries explicitly. It's always a good idea to keep the generated code and the library in sync. @alaindargelas might have a different opinion here.

> That is, detect that MINGW environments behave as Linux in this regard.

That was my initial thought too but I couldn't find any relevant help on tcl in this context. I am not sure if identifying MinGW within tcl scripts would be possible. @alaindargelas Do you know of a way to detect MinGW/MSYS environment in tcl?

I am also divided on this. I am not sure if this would be a good/stable build environment.

> Alternatively, `C:\msys64\mingw64\lib\gcc\x86-64-w64-mingw32` can be aded to the context of the call.

One of the files wasn't even named with the extension '.dll'. It had extenstion '.dll.a'. I am not sure if the process will find the file with that extension. Basically, I just don't know enough about these environment to be sure of how dlls are loaded and what location (and order) are scanned to find them.

> What about the non MSYS2 approach?

That is my next stop. I am moving away from MinGW+GCC and on to MinGW+VC. I am tiny bit more confident about this because I have a full working environment locally.

alaindargelas, 2020-10-23:
You can maybe check an " if [info exist env(...)] " 
about an env variable that is specific to MinGW/MSYS.
Or a particular executable path exists if [file exists "/...path...."]

[13 more replies]

closed 2020-10-26 by hs-apotell