chipsalliance/Surelog #705: Missing typespec in cast to unsigned

rkapuscik, 2020-07-30:
For this line: https://github.com/alainmarcel/Surelog/blob/master/third_party/tests/Earlgrey_0_1/src/lowrisc_ip_usbdev_0.1/rtl/usbdev.sv#L617
UHDM:
```
\_if_else: , line:617
  |vpiCondition:
  \_operation: , line:617
    |vpiOpType:26
    |vpiOperand:
    \_ref_obj: (setup_received), line:617
      |vpiName:setup_received
      |vpiFullName:work@usbdev.proc_stall_tieoff.setup_received
    |vpiOperand:
    \_operation: , line:617
      |vpiOpType:14
      |vpiOperand:
      \_ref_obj: (usb_out_endpoint), line:617
        |vpiName:usb_out_endpoint
        |vpiFullName:work@usbdev.proc_stall_tieoff.usb_out_endpoint
      |vpiOperand:
      \_operation: , line:617
        |vpiOpType:67
        |vpiOperand:
        \_operation: , line:617
          |vpiOpType:67
          |vpiOperand:
          \_ref_obj: (i), line:617
            |vpiName:i
        |vpiTypespec:
        \_integer_typespec: , line:617
          |INT:4
```
Typespec for `unsigned` is missing.

alaindargelas, 2020-07-31:
There is no such typespec in the standard. Not sure what to do, create one for signed and unsigned?

rkapuscik, 2020-07-31:
What Verilator does in this case is create the same nodes as for `$signed` and `$unsigned` calls, skipping the cast altogether. Indeed, the standard in section 6.24.1 says:

> NOTE—The $signed() and $unsigned() system functions (see 11.7) return the same results as signed'() and unsigned'(), respectively

Maybe it could be represented in UHDM with such call?

alaindargelas, 2020-07-31:
Right on!

PR #706 (2020-08-01): better casting support
Fixes #705.
closed 2020-08-01 by alaindargelas