chipsalliance/Surelog #732: Wrong value of a parameter in Ibex

kbieganski, 2020-08-13:
In Ibex, there's this parameter in `ibex_pkg.sv`:
```systemverilog
parameter int unsigned PMP_MAX_REGIONS      = 16;
```
This is the only place it's defined, so it should always equal 16.

And we can see this correct param assignment in UHDM:
```
|vpiParamAssign:
   \_param_assign: , line:281
     |vpiRhs:
     \_constant: , line:281
       |vpiConstType:7
       |vpiDecompile:16
       |vpiSize:32
       |INT:16
     |vpiLhs:
     \_parameter: (PMP_MAX_REGIONS), line:281
       |vpiName:PMP_MAX_REGIONS
       |vpiFullName:ibex_pkg::PMP_MAX_REGIONS
       |vpiTypespec:
       \_int_typespec: , line:281
```
But throughout the Ibex UHDM tree, there's a lot of these:
```
|vpiParameter:
\_parameter: (PMP_MAX_REGIONS), line:281
  |vpiName:PMP_MAX_REGIONS
  |INT:3
```
Somehow, `PMP_MAX_REGIONS` was assigned the value of 3.

If we look at `ibex_cs_registers.sv`, there are two arrays whose sizes depend on this parameter:
```systemverilog
logic [31:0]                 pmp_addr_rdata  [PMP_MAX_REGIONS];
logic [PMP_CFG_W-1:0]        pmp_cfg_rdata   [PMP_MAX_REGIONS];
```
And they have the wrong size as well:
```
|vpiArrayNet:
\_array_net: (pmp_addr_rdata), line:199, parent:cs_registers_i
  |vpiName:pmp_addr_rdata
  |vpiFullName:work@ibex_core.cs_registers_i.pmp_addr_rdata
  |vpiSize:3
  |vpiNet:
  \_logic_net: , parent:pmp_addr_rdata
    |vpiFullName:work@ibex_core.cs_registers_i.pmp_addr_rdata
    |vpiNetType:36
    |vpiRange:
    \_range: , line:199
      |vpiLeftRange:
      \_constant: , line:199
        |vpiConstType:7
        |vpiDecompile:31
        |vpiSize:32
        |INT:31
      |vpiRightRange:
      \_constant: , line:199
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
  |vpiRange:
  \_range: , line:199
    |vpiLeftRange:
    \_constant: , line:199
      |vpiConstType:7
      |vpiDecompile:0
      |vpiSize:32
      |INT:0
    |vpiRightRange:
    \_constant: , line:199
      |vpiConstType:7
      |vpiDecompile:2
      |vpiSize:32
      |INT:2
|vpiArrayNet:
\_array_net: (pmp_cfg_rdata), line:200, parent:cs_registers_i
  |vpiName:pmp_cfg_rdata
  |vpiFullName:work@ibex_core.cs_registers_i.pmp_cfg_rdata
  |vpiSize:3
  |vpiNet:
  \_logic_net: , parent:pmp_cfg_rdata
    |vpiFullName:work@ibex_core.cs_registers_i.pmp_cfg_rdata
    |vpiNetType:36
    |vpiRange:
    \_range: , line:200
      |vpiLeftRange:
      \_constant: , line:200
        |vpiConstType:7
        |vpiDecompile:14
        |vpiSize:32
        |INT:14
      |vpiRightRange:
      \_constant: , line:200
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
  |vpiRange:
  \_range: , line:200
    |vpiLeftRange:
    \_constant: , line:200
      |vpiConstType:7
      |vpiDecompile:0
      |vpiSize:32
      |INT:0
    |vpiRightRange:
    \_constant: , line:200
      |vpiConstType:7
      |vpiDecompile:2
      |vpiSize:32
      |INT:2
```
Even if I'm missing something about the param, we can see from this part of `ibex_cs_registers.sv` that these arrays should be of length 16:
```systemverilog
CSR_PMPCFG0:   csr_rdata_int = {pmp_cfg_rdata[3],  pmp_cfg_rdata[2],
                                pmp_cfg_rdata[1],  pmp_cfg_rdata[0]};
CSR_PMPCFG1:   csr_rdata_int = {pmp_cfg_rdata[7],  pmp_cfg_rdata[6],
                                pmp_cfg_rdata[5],  pmp_cfg_rdata[4]};
CSR_PMPCFG2:   csr_rdata_int = {pmp_cfg_rdata[11], pmp_cfg_rdata[10],
                                pmp_cfg_rdata[9],  pmp_cfg_rdata[8]};
CSR_PMPCFG3:   csr_rdata_int = {pmp_cfg_rdata[15], pmp_cfg_rdata[14],
                                pmp_cfg_rdata[13], pmp_cfg_rdata[12]};
CSR_PMPADDR0:  csr_rdata_int = pmp_addr_rdata[0];
CSR_PMPADDR1:  csr_rdata_int = pmp_addr_rdata[1];
CSR_PMPADDR2:  csr_rdata_int = pmp_addr_rdata[2];
CSR_PMPADDR3:  csr_rdata_int = pmp_addr_rdata[3];
CSR_PMPADDR4:  csr_rdata_int = pmp_addr_rdata[4];
CSR_PMPADDR5:  csr_rdata_int = pmp_addr_rdata[5];
CSR_PMPADDR6:  csr_rdata_int = pmp_addr_rdata[6];
CSR_PMPADDR7:  csr_rdata_int = pmp_addr_rdata[7];
CSR_PMPADDR8:  csr_rdata_int = pmp_addr_rdata[8];
CSR_PMPADDR9:  csr_rdata_int = pmp_addr_rdata[9];
CSR_PMPADDR10: csr_rdata_int = pmp_addr_rdata[10];
CSR_PMPADDR11: csr_rdata_int = pmp_addr_rdata[11];
CSR_PMPADDR12: csr_rdata_int = pmp_addr_rdata[12];
CSR_PMPADDR13: csr_rdata_int = pmp_addr_rdata[13];
CSR_PMPADDR14: csr_rdata_int = pmp_addr_rdata[14];
CSR_PMPADDR15: csr_rdata_int = pmp_addr_rdata[15];
```
I tried to reproduce it with a minimal example, but I was unsuccessful. Seems to be a very specific edge case.

kbieganski, 2020-08-13:
Found two more cases of this. Again, the parameters are defined in `ibex_pkg.sv`. The first one:
```systemverilog
parameter int unsigned PMP_I = 0;
```
Relevant correct UHDM:
```
|vpiParamAssign:
\_param_assign: , line:285
  |vpiRhs:
  \_constant: , line:285
    |vpiConstType:7
    |vpiDecompile:0
    |vpiSize:32
    |INT:0
  |vpiLhs:
  \_parameter: (PMP_I), line:285
    |vpiName:PMP_I
    |vpiFullName:ibex_pkg::PMP_I
    |vpiTypespec:
    \_int_typespec: , line:285
```
Relevant incorrect UHDM:
```
|vpiParameter:
\_parameter: (PMP_I), line:285
  |vpiName:PMP_I
  |INT:19
```
The second one:
```systemverilog
parameter int unsigned PMP_D = 1;
```
Relevant correct UHDM:
```
|vpiParamAssign:
\_param_assign: , line:286
  |vpiRhs:
  \_constant: , line:286
    |vpiConstType:7
    |vpiDecompile:1
    |vpiSize:32
    |INT:1
  |vpiLhs:
  \_parameter: (PMP_D), line:286
    |vpiName:PMP_D
    |vpiFullName:ibex_pkg::PMP_D
    |vpiTypespec:
    \_int_typespec: , line:286
```
Relevant incorrect UHDM:
```
|vpiParameter:
\_parameter: (PMP_D), line:286
  |vpiName:PMP_D
  |INT:23
```

kbieganski, 2020-08-14:
[This](https://github.com/alainmarcel/Surelog/blob/4e7573fb266a108d3c39a3f1c589c3ad3c1f7ba1/src/DesignCompile/DesignElaboration.cpp#L1308) is the line in `DesignElaboration.cpp` in Surelog that causes this issue:
```cpp
Value* value = m_exprBuilder.clone(def->getValues()[i]);
```
The indexes seem to be incorrect. After replacing it with:
```cpp
Value* value = m_exprBuilder.clone(def->getValue(name));
```
the result is correct. I see this exact line just above the erroneous one, but it's commented out, so I suppose there was some problem with it?

alaindargelas, 2020-08-18:
Yes, I remember now, there is a problem with the name lookup as well, as it does not find the right value either if the value gets overridden.
The addValue append to the list of values, it needs to change to a replacement when values gets overriden. I'm on it.

closed 2020-08-18 by alaindargelas

kbieganski, 2020-08-18:
Fixed by #738.

PR #738 (2020-08-18): fix parameter map issue
Hopefully fixes #732
@kbieganski, please confirm.