chipsalliance/Surelog #733: Formal arguments(io_decls) and Actual_Group of Local variables(ref_obj) have different point address in function

bsp13, 2020-08-14:
curr_state and next_state have not same point address as in formal argument (io_decls) have.

Verilog Code:
```
module fsm_2_always_block (input clk, rst_n);

parameter IDLE = 2'b00, READ = 2'b01, DLY = 2'b11, DONE = 2'b10;
reg [1:0] state, next;

always @(posedge clk or negedge rst_n) begin
if (!rst_n) 
  state <= IDLE;
else  
  stateAsmt(state,next);
end

function stateAsmt(reg [1:0] curr_state,next_state) ;
     curr_state = next_state;
endfunction

endmodule
```

alaindargelas, 2020-08-18:
@bsp13, can you inline a piece of uhdm dump that shows the problem?
To add the IDs, recompile and rerun after modifying locally third_party/UHDM/templates/vpi_visitor.cpp:
static bool showIDs = **true;**

alaindargelas, 2020-08-19:
Based on the dump below, I see this case is fixed, probably the previous issue which fixed the elaboration of function fixed this problem as well.
On the other hand, I see that the range of the io decl is not carried other the 2nd io, I'll fix that part of this case.

`====== UHDM =======
design: (work@fsm_2_always_block), id:10
|vpiName:work@fsm_2_always_block
|uhdmallPackages:
\_package: builtin, id:11, parent:work@fsm_2_always_block, parID:10
  |vpiDefName:builtin
  |vpiFullName:builtin
|uhdmallClasses:
\_class_defn: (builtin::array), id:12, parID:11
  |vpiName:builtin::array
  |vpiFullName:builtin::builtin::array
|uhdmallClasses:
\_class_defn: (builtin::queue), id:13, parID:11
  |vpiName:builtin::queue
  |vpiFullName:builtin::builtin::queue
|uhdmallClasses:
\_class_defn: (builtin::string), id:14, parID:11
  |vpiName:builtin::string
  |vpiFullName:builtin::builtin::string
|uhdmallClasses:
\_class_defn: (builtin::system), id:15, parID:11
  |vpiName:builtin::system
  |vpiFullName:builtin::builtin::system
|uhdmallModules:
\_module: work@fsm_2_always_block, id:16, file:tests/ScratchPad.sv, line:1, parent:work@fsm_2_always_block, parID:10
  |vpiDefName:work@fsm_2_always_block
  |vpiFullName:work@fsm_2_always_block
  |vpiTaskFunc:
  \_function: (stateAsmt), id:0, line:4, parID:16
    |vpiName:stateAsmt
    |vpiFullName:work@fsm_2_always_block.stateAsmt
    |vpiIODecl:
    \_io_decl: (curr_state), id:1
      |vpiName:curr_state
      |vpiDirection:5
      |vpiExpr:
      \_logic_var: , id:5, line:4, parent:curr_state, parID:1
        |vpiFullName:curr_state
        |vpiRange:
        \_range: , id:2, line:4
          |vpiLeftRange:
          \_constant: , id:3, line:4
            |vpiConstType:7
            |vpiDecompile:1
            |vpiSize:32
            |INT:1
          |vpiRightRange:
          \_constant: , id:4, line:4
            |vpiConstType:7
            |vpiDecompile:0
            |vpiSize:32
            |INT:0
    |vpiIODecl:
    \_io_decl: (next_state), id:6
      |vpiName:next_state
      |vpiDirection:5
    |vpiStmt:
    \_assignment: , id:9, line:5, parent:stateAsmt, parID:0
      |vpiOpType:82
      |vpiBlocking:1
      |vpiLhs:
      \_ref_obj: (curr_state), id:7, line:5, parent:stateAsmt, parID:0
        |vpiName:curr_state
        |vpiFullName:work@fsm_2_always_block.stateAsmt.curr_state
        |vpiActual:
        \_io_decl: (curr_state), id:1
      |vpiRhs:
      \_ref_obj: (next_state), id:8, line:5, parent:stateAsmt, parID:0
        |vpiName:next_state
        |vpiFullName:work@fsm_2_always_block.stateAsmt.next_state
        |vpiActual:
        \_io_decl: (next_state), id:6
|uhdmtopModules:
\_module: work@fsm_2_always_block (work@fsm_2_always_block), id:17, file:tests/ScratchPad.sv, line:1
  |vpiDefName:work@fsm_2_always_block
  |vpiName:work@fsm_2_always_block
  |vpiTaskFunc:
  \_function: (stateAsmt), id:0, line:4, parID:16
    |vpiName:stateAsmt
    |vpiFullName:work@fsm_2_always_block.stateAsmt
    |vpiIODecl:
    \_io_decl: (curr_state), id:1
      |vpiName:curr_state
      |vpiDirection:5
      |vpiExpr:
      \_logic_var: , id:5, line:4, parent:curr_state, parID:1
        |vpiFullName:curr_state
        |vpiRange:
        \_range: , id:2, line:4
          |vpiLeftRange:
          \_constant: , id:3, line:4
            |vpiConstType:7
            |vpiDecompile:1
            |vpiSize:32
            |INT:1
          |vpiRightRange:
          \_constant: , id:4, line:4
            |vpiConstType:7
            |vpiDecompile:0
            |vpiSize:32
            |INT:0
    |vpiIODecl:
    \_io_decl: (next_state), id:6
      |vpiName:next_state
      |vpiDirection:5
    |vpiStmt:
    \_assignment: , id:9, line:5, parent:stateAsmt, parID:0
      |vpiOpType:82
      |vpiBlocking:1
      |vpiLhs:
      \_ref_obj: (curr_state), id:7, line:5, parent:stateAsmt, parID:0
        |vpiName:curr_state
        |vpiFullName:work@fsm_2_always_block.stateAsmt.curr_state
        |vpiActual:
        \_io_decl: (curr_state), id:1
      |vpiRhs:
      \_ref_obj: (next_state), id:8, line:5, parent:stateAsmt, parID:0
        |vpiName:next_state
        |vpiFullName:work@fsm_2_always_block.stateAsmt.next_state
        |vpiActual:
        \_io_decl: (next_state), id:6
===================`

bsp13, 2021-04-14:
I got same problem again with function like formal arguments(io_decls) and Actual_Group of Local variables(ref_obj) have different point address in function.

```
module fsm_2_always_block (input clk, rst_n);

parameter IDLE = 2'b00, READ = 2'b01, DLY = 2'b11, DONE = 2'b10;
reg [1:0] state, next;

always @(posedge clk or negedge rst_n) begin
if (!rst_n) 
  state <= IDLE;
else  
  stateAsmt(state,next);
end

function stateAsmt(reg [1:0] curr_state,next_state) ;
     curr_state = next_state;
endfunction

endmodule
```

alaindargelas, 2021-04-15:
Functions had to be deep cloned so parameter could be uniquified.
That re-introduced the issue.

PR #739 (2020-08-19): Fix implicit function arg type spec
Fixes #733
PR #1201 (2021-04-15): function io_decl binding after cloning
Fixes #733
closed 2021-04-15 by alaindargelas