chipsalliance/Surelog #762: Constant expression support

kbieganski, 2020-09-02:
Surelog does not seem to support constant expressions. Consider this simple example:
```systemverilog
module ibex_register_file #(
    parameter bit          RV32E             = 0,
    parameter int unsigned DataWidth         = 32
);
  localparam int unsigned ADDR_WIDTH = RV32E ? 4 : 5;
  localparam int unsigned NUM_WORDS  = 2**ADDR_WIDTH;

  logic [NUM_WORDS-1:0][DataWidth-1:0] rf_reg;
  logic [NUM_WORDS-1:1][DataWidth-1:0] rf_reg_q;
  logic [NUM_WORDS-1:1]                we_a_dec;
endmodule
```
which corresponds to the following UHDM:
```
\_module: work@ibex_register_file (work@ibex_register_file), file:ibex_register_file_ff.sv, line:1
  |vpiDefName:work@ibex_register_file
  |vpiName:work@ibex_register_file
  |vpiNet:
  \_logic_net: (rf_reg), line:8, parent:work@ibex_register_file
    |vpiName:rf_reg
    |vpiFullName:work@ibex_register_file.rf_reg
    |vpiNetType:36
    |vpiRange:
    \_range: , line:8
      |vpiLeftRange:
      \_constant: , line:8
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:32
        |INT:1
      |vpiRightRange:
      \_constant: , line:8
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
    |vpiRange:
    \_range: , line:8
      |vpiLeftRange:
      \_constant: , line:8
        |vpiConstType:7
        |vpiDecompile:31
        |vpiSize:32
        |INT:31
      |vpiRightRange:
      \_constant: , line:8
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
  |vpiNet:
  \_logic_net: (rf_reg_q), line:9, parent:work@ibex_register_file
    |vpiName:rf_reg_q
    |vpiFullName:work@ibex_register_file.rf_reg_q
    |vpiNetType:36
    |vpiRange:
    \_range: , line:9
      |vpiLeftRange:
      \_constant: , line:9
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:32
        |INT:1
      |vpiRightRange:
      \_constant: , line:9
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:32
        |INT:1
    |vpiRange:
    \_range: , line:9
      |vpiLeftRange:
      \_constant: , line:9
        |vpiConstType:7
        |vpiDecompile:31
        |vpiSize:32
        |INT:31
      |vpiRightRange:
      \_constant: , line:9
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
  |vpiNet:
  \_logic_net: (we_a_dec), line:10, parent:work@ibex_register_file
    |vpiName:we_a_dec
    |vpiFullName:work@ibex_register_file.we_a_dec
    |vpiNetType:36
    |vpiRange:
    \_range: , line:10
      |vpiLeftRange:
      \_constant: , line:10
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:32
        |INT:1
      |vpiRightRange:
      \_constant: , line:10
        |vpiConstType:7
        |vpiDecompile:1
        |vpiSize:32
        |INT:1
  |vpiParameter:
  \_parameter: (ADDR_WIDTH), line:5
    |vpiName:ADDR_WIDTH
    |INT:0
  |vpiParameter:
  \_parameter: (DataWidth), line:3
    |vpiName:DataWidth
    |INT:32
  |vpiParameter:
  \_parameter: (NUM_WORDS), line:6
    |vpiName:NUM_WORDS
    |INT:2
  |vpiParameter:
  \_parameter: (RV32E), line:2
    |vpiName:RV32E
    |INT:0
```
Notice that parameters `ADDR_WIDTH` and `NUM_WORDS` have incorrect values, and all the nets have incorrect ranges because of that.

As you probably guessed, this is taken from Ibex, from `ibex_register_file_ff.sv`. A similar thing happens in `ibex_register_file_fpga.sv`.

alaindargelas, 2020-09-03:
The example you give does not yield the UHDM dump in the main branch of Surelog.

```
|uhdmallModules:
\_module: work@ibex_register_file, id:48, file:tests/ScratchPad.sv, line:1, parent:work@ibex_register_file, parID:42
  |vpiDefName:work@ibex_register_file
  |vpiFullName:work@ibex_register_file
  |vpiNet:
  \_logic_net: (rf_reg), id:49, line:8, parID:48
    |vpiName:rf_reg
    |vpiFullName:work@ibex_register_file.rf_reg
    |vpiNetType:36
  |vpiNet:
  \_logic_net: (rf_reg_q), id:50, line:9, parID:48
    |vpiName:rf_reg_q
    |vpiFullName:work@ibex_register_file.rf_reg_q
    |vpiNetType:36
  |vpiNet:
  \_logic_net: (we_a_dec), id:51, line:10, parID:48
    |vpiName:we_a_dec
    |vpiFullName:work@ibex_register_file.we_a_dec
    |vpiNetType:36
  |vpiParamAssign:
  \_param_assign: , id:2, line:2
    |vpiRhs:
    \_constant: , id:3, line:2
      |vpiConstType:7
      |vpiDecompile:0
      |vpiSize:32
      |INT:0
    |vpiLhs:
    \_parameter: (RV32E), id:1, line:2
      |vpiName:RV32E
      |vpiTypespec:
      \_bit_typespec: , id:0, line:2
  |vpiParamAssign:
  \_param_assign: , id:6, line:3
    |vpiRhs:
    \_constant: , id:7, line:3
      |vpiConstType:7
      |vpiDecompile:32
      |vpiSize:32
      |INT:32
    |vpiLhs:
    \_parameter: (DataWidth), id:5, line:3
      |vpiName:DataWidth
      |vpiTypespec:
      \_int_typespec: , id:4, line:3
  |vpiParamAssign:
  \_param_assign: , id:10, line:5
    |vpiRhs:
    \_operation: , id:12, line:5
      |vpiOpType:32
      |vpiOperand:
      \_constant: , id:11, line:5, parID:12
        |vpiConstType:7
        |vpiDecompile:0
        |vpiSize:32
        |INT:0
      |vpiOperand:
      \_constant: , id:13, line:5, parID:12
        |vpiConstType:7
        |vpiDecompile:4
        |vpiSize:32
        |INT:4
      |vpiOperand:
      \_constant: , id:14, line:5, parID:12
        |vpiConstType:7
        |vpiDecompile:5
        |vpiSize:32
[... 225 lines cut ...]
          |INT:4
        |vpiOperand:
        \_constant: , id:75, line:5, parID:72
          |vpiConstType:7
          |vpiDecompile:5
          |vpiSize:32
          |INT:5
    |vpiLhs:
    \_parameter: (NUM_WORDS), id:55, line:6, parID:48
      |vpiName:NUM_WORDS
      |vpiFullName:work@ibex_register_file.NUM_WORDS
      |vpiLocalParam:1
      |vpiTypespec:
      \_int_typespec: , id:80, line:6, parent:NUM_WORDS, parID:55
  |vpiParameter:
  \_parameter: (RV32E), id:56, line:2, parID:48
  |vpiParameter:
  \_parameter: (DataWidth), id:54, line:3, parID:48
  |vpiParameter:
  \_parameter: (ADDR_WIDTH), id:53, line:5, parID:48
  |vpiParameter:
  \_parameter: (NUM_WORDS), id:55, line:6, parID:48
```
Are you operating on an older version?
Surelog does not compute the end value, so it is easy to spot issues, but the expression tree is correct on the main branch.

alaindargelas, 2020-09-03:
Got it, I didn't check the signals.
I see several problems in the code, #1 the power operator ** is not handled in evaluation.

PR #764 (2020-09-04): Support cond op and power op in elaboration expression eval
Fixes #762
closed 2020-09-04 by alaindargelas