chipsalliance/Surelog #763: Functions definition is not called when we call them repeat

bsp13, 2020-09-03:
Verilog code::
```
module fsm_2_always_block (output reg rd, ds,input go, ws, clk, rst_n);

parameter IDLE = 2'b00, READ = 2'b01, DLY = 2'b11, DONE = 2'b10;
reg [1:0] state, next;

always @(posedge clk or negedge rst_n) begin
  nextStateAsmt_IDLE();
  nextStateAsmt_IDLE();  <---------------------definition is not called.... 
end

function nextStateAsmt_IDLE();
    next = IDLE; <<--------------stmt is only called one times.
endfunction

endmodule

alaindargelas, 2020-09-04:
I'm not sure I understand the problem.
In the dump you can see 2 function call statements pointing to the same function.
In the dump you can see the first call is being expanded and you can see the function guts.
The 2nd call is not expanded as it would be a repeat of the same data, the "visited" map in vpi_visitor.cpp prevents that data to be dumped twice, is this what you are referring as "is only called on time"?
The visitor is not a "Simulator" it does not execute code, it only traverses the data model and does not visit twice the same nodes.

```
|uhdmtopModules:
\_module: work@fsm_2_always_block (work@fsm_2_always_block), id:79, file:tests/ScratchPad.sv, line:1
  |vpiDefName:work@fsm_2_always_block
  |vpiName:work@fsm_2_always_block
  |vpiProcess:
  \_always: , id:84, line:6, parID:58
    |vpiAlwaysType:1
    |vpiStmt:
    \_event_control: , id:85, line:6, parID:84
      |vpiCondition:
      \_operation: , id:86, line:6, parID:85
        |vpiOpType:35
        |vpiOperand:
        \_operation: , id:87, line:6, parID:86
          |vpiOpType:39
          |vpiOperand:
          \_ref_obj: (clk), id:88, line:6, parID:87
            |vpiName:clk
            |vpiFullName:work@fsm_2_always_block.clk
            |vpiActual:
            \_logic_net: (clk), id:36, line:1, parent:work@fsm_2_always_block, parID:79
              |vpiName:clk
              |vpiFullName:work@fsm_2_always_block.clk
        |vpiOperand:
        \_operation: , id:89, line:6, parID:86
          |vpiOpType:40
          |vpiOperand:
          \_ref_obj: (rst_n), id:90, line:6, parID:89
            |vpiName:rst_n
            |vpiFullName:work@fsm_2_always_block.rst_n
            |vpiActual:
            \_logic_net: (rst_n), id:37, line:1, parent:work@fsm_2_always_block, parID:79
              |vpiName:rst_n
              |vpiFullName:work@fsm_2_always_block.rst_n
      |vpiStmt:
      \_begin: , id:91, line:6, parID:85
        |vpiFullName:work@fsm_2_always_block
        |vpiStmt:
        \_func_call: (nextStateAsmt_IDLE), id:92, line:7, parID:91
          |vpiName:nextStateAsmt_IDLE
          |vpiFunction:
          \_function: (nextStateAsmt_IDLE), id:12, line:11, parID:58
            |vpiName:nextStateAsmt_IDLE
            |vpiFullName:work@fsm_2_always_block.nextStateAsmt_IDLE
            |vpiStmt:
            \_assignment: , id:15, line:12, parent:nextStateAsmt_IDLE, parID:12
              |vpiOpType:82
              |vpiBlocking:1
              |vpiLhs:
              \_ref_obj: (next), id:13, line:12, parent:nextStateAsmt_IDLE, parID:12
                |vpiName:next
                |vpiFullName:work@fsm_2_always_block.nextStateAsmt_IDLE.next
                |vpiActual:
                \_logic_net: (next), id:45, line:4, parent:work@fsm_2_always_block, parID:79
                  |vpiName:next
                  |vpiFullName:work@fsm_2_always_block.next
                  |vpiNetType:48
                  |vpiRange:
                  \_range: , id:42, line:4
                    |vpiLeftRange:
                    \_constant: , id:43, line:4, parID:42
                      |vpiConstType:7
                      |vpiDecompile:1
                      |vpiSize:32
                      |INT:1
                    |vpiRightRange:
                    \_constant: , id:44, line:4, parID:42
                      |vpiConstType:7
                      |vpiDecompile:0
                      |vpiSize:32
                      |INT:0
              |vpiRhs:
              \_ref_obj: (IDLE), id:14, line:12, parent:nextStateAsmt_IDLE, parID:12
                |vpiName:IDLE
                |vpiFullName:work@fsm_2_always_block.nextStateAsmt_IDLE.IDLE
                |vpiActual:
                \_parameter: (IDLE), id:82, line:3, parID:58
                  |vpiName:IDLE
                  |vpiFullName:work@fsm_2_always_block.IDLE
        |vpiStmt:
        \_func_call: (nextStateAsmt_IDLE), id:93, line:8, parID:91
          |vpiName:nextStateAsmt_IDLE
          |vpiFunction:
          \_function: (nextStateAsmt_IDLE), id:12, line:11, parID:58
```

bsp13, 2020-09-04:
Okey!, I understood problem. It is misunderstading from my side.

closed 2020-09-04 by bsp13