chipsalliance/Surelog #776: System Verilog Classes support

bsp13, 2020-09-09:
UHDM is not support system verilog classes.

```
virtual class door;
    int state;
    int timer;
    logic open,lock,clock;
    parameter OPENED    = 2'b00,
	      CLOSED    = 2'b01,
	      LOCKED    = 2'b10;
    pure virtual task door_fsm();
endclass

class doorOpen extends door;
    virtual task door_fsm();
        if (open)  state = OPENED;
    endtask
endclass

module door_mod;
    doorOpen open;
    initial begin
        open = new();       
        open.door_fsm();     
    end
endmodule
```

alaindargelas, 2020-09-10:
It's going to be a while before I get to that.
In order of priority from high to low:
Yosys and Verilator support for Earlgrey design
Assertions support
Column information
Testbench Classes
Constraints

alaindargelas, 2020-09-16:
Actually, our timeline has changed too, so I'm going to reverse some order here:
Classes
Column information
Assertions support
Constraints

alaindargelas, 2020-10-01:
Basic support for class definition is done:
- inheritance, 
- class def in package, module, program, at top level
- class parameter
- subclasses
- functions/tasks
- member properties

Still to do: constraints, covergroups

alaindargelas, 2020-10-01:
Class instantiation (class vars) is still to do.

alaindargelas, 2020-10-11:
I'm resuming working on this again.

alaindargelas, 2020-10-15:
Still working on static (early) method binding

alaindargelas, 2020-10-22:
func method binding is now working, but not the task method binding yet.
Reason is when creating the cal object we don't know the nature of the task/func

alaindargelas, 2020-10-23:
UHDM does support this testcase now.

closed 2020-10-23 by alaindargelas