chipsalliance/Surelog #778: Function definition iterate without func_call in VpiListener

bsp13, 2020-09-11:
Verilog Code:
```
module test (output reg rd, ds,
input go, ws, clk, rst_n);
parameter IDLE = 2'b00, READ = 2'b01, DLY = 2'b11, DONE = 2'b10;

reg [1:0] state, next, curr_state, next_state;

always @(posedge clk or negedge rst_n) begin
  state <= IDLE;  
end

function nextStateAsmt_DONE();
  state = DONE;  <-------I got this assignment in vpilistener without called function
endfunction

endmodule
```

It enter in "enterTask_func" function in vpilistener. Is this valid? because I do not call function here.

alaindargelas, 2020-09-11:
Yes, it is valid.
The listener traverses the whole static tree, unlike an interpreter, the listener does not care if something is called 0, 1, or more times, instead the listener will traverse all the nodes in the tree, hopefully only once.

It seems you need to build something that looks like an interpreter (or a subset of an interpreter), from what I can infer from your questions.

bsp13, 2020-09-11:
When I called function that time it iterate from "enterFunction" but it also iterate from "enterTask_func". Why does it iterate from "enterTask_func" in vpiListener. how to differentiate this "enterFunction", "enterTask_func" , "enterTf_call" or "enterTask".

alaindargelas, 2020-09-12:
I'll need a piece of C++ or at least some text trace that shows the problem to understand what you are trying to do.
Have you based your code from the UHDM/tests/test_listener.cpp example?

To your question: how to differentiate this "enterFunction", "enterTask_func" , "enterTf_call" or "enterTask"
Can't you use the context push/pop like in the example to know what is the context when the listener is calling you?

For your question "Why does it iterate from "enterTask_func" in vpiListener":
Take a look at UHDM/src/vpi_listener.cpp generated code to understand the listener call stack.

bsp13, 2020-09-16:
Okey, I understood vpiListener. Thanks for reply!!

closed 2020-09-16 by bsp13