chipsalliance/Surelog #784: parsing problem : ambiguity between module_instantiation and udp_instantiation

DamienLeBars, 2020-09-16:
It seems that there is a parsing problem due to an ambiguity between module_instantiation (list_of_port_connections ) and udp_instantiation (udp_instance).

[svfile_and_result.zip](https://github.com/alainmarcel/Surelog/files/5231832/svfile_and_result.zip)

For my attached example , i don't expect any Udp_instantiation, but only Module_instantiation.
```verilog

`timescale 1ns / 1ns

module moduleA (inout port0, port1);
endmodule

module top;
    wire [1:0] topA, topB;
    moduleA instanceA(.port0(topA[0]), .port1(topB[0]));
    moduleA instanceB(topA[1], topB[0]);
    moduleA instanceC(topA[1], topB[0]||topB[1]);
    moduleA instanceD(topA[1]||topA[0], topB[0]);
endmodule

```
instanceA and instanceD are ok. 
instanceB and instanceC are failed.

- instanceA:
```xml
n<moduleA> u<47> t<StringConst> p<76> s<75> l<9>
n<instanceA> u<48> t<StringConst> p<49> l<9>
n<> u<49> t<Name_of_instance> p<75> c<48> s<74> l<9>
n<port0> u<50> t<StringConst> p<61> s<60> l<9>
n<topA> u<51> t<StringConst> p<58> s<57> l<9>
n<0> u<52> t<IntConst> p<53> l<9>
n<> u<53> t<Primary_literal> p<54> c<52> l<9>
n<> u<54> t<Primary> p<55> c<53> l<9>
n<> u<55> t<Expression> p<56> c<54> l<9>
n<> u<56> t<Bit_select> p<57> c<55> l<9>
n<> u<57> t<Select> p<58> c<56> l<9>
n<> u<58> t<Complex_func_call> p<59> c<51> l<9>
n<> u<59> t<Primary> p<60> c<58> l<9>
n<> u<60> t<Expression> p<61> c<59> l<9>
n<> u<61> t<Named_port_connection> p<74> c<50> s<73> l<9>
n<port1> u<62> t<StringConst> p<73> s<72> l<9>
n<topB> u<63> t<StringConst> p<70> s<69> l<9>
n<0> u<64> t<IntConst> p<65> l<9>
n<> u<65> t<Primary_literal> p<66> c<64> l<9>
n<> u<66> t<Primary> p<67> c<65> l<9>
n<> u<67> t<Expression> p<68> c<66> l<9>
n<> u<68> t<Bit_select> p<69> c<67> l<9>
n<> u<69> t<Select> p<70> c<68> l<9>
n<> u<70> t<Complex_func_call> p<71> c<63> l<9>
n<> u<71> t<Primary> p<72> c<70> l<9>
n<> u<72> t<Expression> p<73> c<71> l<9>
n<> u<73> t<Named_port_connection> p<74> c<62> l<9>
n<> u<74> t<List_of_port_connections> p<75> c<61> l<9>
n<> u<75> t<Hierarchical_instance> p<76> c<49> l<9>
n<> u<76> t<Module_instantiation> p<77> c<47> l<9>
n<> u<77> t<Module_or_generate_item> p<78> c<76> l<9>
n<> u<78> t<Non_port_module_item> p<185> c<77> s<104> l<9>
```
- instanceB:
```xml
n<moduleA> u<79> t<StringConst> p<102> s<101> l<10>
n<instanceB> u<80> t<StringConst> p<81> l<10>
n<> u<81> t<Name_of_instance> p<101> c<80> s<90> l<10>
n<topA> u<82> t<StringConst> p<83> l<10>
n<> u<83> t<Ps_or_hierarchical_identifier> p<90> c<82> s<89> l<10>
n<1> u<84> t<IntConst> p<85> l<10>
n<> u<85> t<Primary_literal> p<86> c<84> l<10>
n<> u<86> t<Constant_primary> p<87> c<85> l<10>
n<> u<87> t<Constant_expression> p<88> c<86> l<10>
n<> u<88> t<Constant_bit_select> p<89> c<87> l<10>
n<> u<89> t<Constant_select> p<90> c<88> l<10>
n<> u<90> t<Net_lvalue> p<101> c<83> s<100> l<10>
n<topB> u<91> t<StringConst> p<98> s<97> l<10>
n<0> u<92> t<IntConst> p<93> l<10>
n<> u<93> t<Primary_literal> p<94> c<92> l<10>
n<> u<94> t<Primary> p<95> c<93> l<10>
n<> u<95> t<Expression> p<96> c<94> l<10>
n<> u<96> t<Bit_select> p<97> c<95> l<10>
n<> u<97> t<Select> p<98> c<96> l<10>
n<> u<98> t<Complex_func_call> p<99> c<91> l<10>
n<> u<99> t<Primary> p<100> c<98> l<10>
n<> u<100> t<Expression> p<101> c<99> l<10>
n<> u<101> t<Udp_instance> p<102> c<81> l<10>
n<> u<102> t<Udp_instantiation> p<103> c<79> l<10>
n<> u<103> t<Module_or_generate_item> p<104> c<102> l<10>
n<> u<104> t<Non_port_module_item> p<185> c<103> s<142> l<10>
```
- instanceC:
```xml
n<moduleA> u<105> t<StringConst> p<140> s<139> l<11>
n<instanceC> u<106> t<StringConst> p<107> l<11>
n<> u<107> t<Name_of_instance> p<139> c<106> s<116> l<11>
n<topA> u<108> t<StringConst> p<109> l<11>
n<> u<109> t<Ps_or_hierarchical_identifier> p<116> c<108> s<115> l<11>
n<1> u<110> t<IntConst> p<111> l<11>
n<> u<111> t<Primary_literal> p<112> c<110> l<11>
n<> u<112> t<Constant_primary> p<113> c<111> l<11>
n<> u<113> t<Constant_expression> p<114> c<112> l<11>
n<> u<114> t<Constant_bit_select> p<115> c<113> l<11>
n<> u<115> t<Constant_select> p<116> c<114> l<11>
n<> u<116> t<Net_lvalue> p<139> c<109> s<138> l<11>
n<topB> u<117> t<StringConst> p<124> s<123> l<11>
n<0> u<118> t<IntConst> p<119> l<11>
n<> u<119> t<Primary_literal> p<120> c<118> l<11>
n<> u<120> t<Primary> p<121> c<119> l<11>
n<> u<121> t<Expression> p<122> c<120> l<11>
n<> u<122> t<Bit_select> p<123> c<121> l<11>
n<> u<123> t<Select> p<124> c<122> l<11>
n<> u<124> t<Complex_func_call> p<125> c<117> l<11>
n<> u<125> t<Primary> p<126> c<124> l<11>
n<> u<126> t<Expression> p<138> c<125> s<137> l<11>
n<topB> u<127> t<StringConst> p<134> s<133> l<11>
n<1> u<128> t<IntConst> p<129> l<11>
n<> u<129> t<Primary_literal> p<130> c<128> l<11>
n<> u<130> t<Primary> p<131> c<129> l<11>
n<> u<131> t<Expression> p<132> c<130> l<11>
n<> u<132> t<Bit_select> p<133> c<131> l<11>
n<> u<133> t<Select> p<134> c<132> l<11>
n<> u<134> t<Complex_func_call> p<135> c<127> l<11>
n<> u<135> t<Primary> p<136> c<134> l<11>
n<> u<136> t<Expression> p<138> c<135> l<11>
n<> u<137> t<BinOp_LogicOr> p<138> s<136> l<11>
n<> u<138> t<Expression> p<139> c<126> l<11>
n<> u<139> t<Udp_instance> p<140> c<107> l<11>
n<> u<140> t<Udp_instantiation> p<141> c<105> l<11>
n<> u<141> t<Module_or_generate_item> p<142> c<140> l<11>
n<> u<142> t<Non_port_module_item> p<185> c<141> s<184> l<11>
```
- instanceD:
```xml
n<moduleA> u<143> t<StringConst> p<182> s<181> l<12>
n<instanceD> u<144> t<StringConst> p<145> l<12>
n<> u<145> t<Name_of_instance> p<181> c<144> s<180> l<12>
n<topA> u<146> t<StringConst> p<153> s<152> l<12>
n<1> u<147> t<IntConst> p<148> l<12>
n<> u<148> t<Primary_literal> p<149> c<147> l<12>
n<> u<149> t<Primary> p<150> c<148> l<12>
n<> u<150> t<Expression> p<151> c<149> l<12>
n<> u<151> t<Bit_select> p<152> c<150> l<12>
n<> u<152> t<Select> p<153> c<151> l<12>
n<> u<153> t<Complex_func_call> p<154> c<146> l<12>
n<> u<154> t<Primary> p<155> c<153> l<12>
n<> u<155> t<Expression> p<167> c<154> s<166> l<12>
n<topA> u<156> t<StringConst> p<163> s<162> l<12>
n<0> u<157> t<IntConst> p<158> l<12>
n<> u<158> t<Primary_literal> p<159> c<157> l<12>
n<> u<159> t<Primary> p<160> c<158> l<12>
n<> u<160> t<Expression> p<161> c<159> l<12>
n<> u<161> t<Bit_select> p<162> c<160> l<12>
n<> u<162> t<Select> p<163> c<161> l<12>
n<> u<163> t<Complex_func_call> p<164> c<156> l<12>
n<> u<164> t<Primary> p<165> c<163> l<12>
n<> u<165> t<Expression> p<167> c<164> l<12>
n<> u<166> t<BinOp_LogicOr> p<167> s<165> l<12>
n<> u<167> t<Expression> p<168> c<155> l<12>
n<> u<168> t<Ordered_port_connection> p<180> c<167> s<179> l<12>
n<topB> u<169> t<StringConst> p<176> s<175> l<12>
n<0> u<170> t<IntConst> p<171> l<12>
n<> u<171> t<Primary_literal> p<172> c<170> l<12>
n<> u<172> t<Primary> p<173> c<171> l<12>
n<> u<173> t<Expression> p<174> c<172> l<12>
n<> u<174> t<Bit_select> p<175> c<173> l<12>
n<> u<175> t<Select> p<176> c<174> l<12>
n<> u<176> t<Complex_func_call> p<177> c<169> l<12>
n<> u<177> t<Primary> p<178> c<176> l<12>
n<> u<178> t<Expression> p<179> c<177> l<12>
n<> u<179> t<Ordered_port_connection> p<180> c<178> l<12>
n<> u<180> t<List_of_port_connections> p<181> c<168> l<12>
n<> u<181> t<Hierarchical_instance> p<182> c<145> l<12>
n<> u<182> t<Module_instantiation> p<183> c<143> l<12>
n<> u<183> t<Module_or_generate_item> p<184> c<182> l<12>
n<> u<184> t<Non_port_module_item> p<185> c<183> l<12>
```

alaindargelas, 2020-09-17:
Yes, this is indeed a known problem with the SystemVerilog grammar, there are several instances of ambiguities like this one.
The way out of this problem is during binding/elaboration, essentially during the semantic passes, not during parsing, Surelog looks at the object type that this symbols matches with and correct the parser assumption when building the elaborated model.
If you notice the Surelog instance tree and UHDM dumps:  
surelog -d inst -d uhdm file.sv

Instance tree:
[TOP] work@top work@top
[MOD] work@moduleA work@top.instanceA 
[MOD] work@moduleA work@top.instanceB  << [MOD] stands for module, there is another type for [UDP]
[MOD] work@moduleA work@top.instanceC
[MOD] work@moduleA work@top.instanceD

UHDM dump:

...
|vpiModule:
  \_module: work@moduleA (instanceB), id:70, file:tests/ScratchPad.sv, line:9, parent:work@top, parID:68

The instantiation type fir instanceB is corrected to be a module.

Fixing the AST once the Parser generated it is impossible, that is why only subsequent data structures get to see the fix, but not the AST as you noticed.
If the grammar produced the wrong AST, the only way to fix it is by fixing the grammar. If you know of a grammar rule change that would allow a disambiguation, please propose it. I think I tried a few things before and didn't succeed.

I suggest you use the UHDM model for the instantiation tree and maybe for your entire application. If you still want to create your application at the AST level, you'll have to do the same fix as I did by looking at the type of the object the instance binds to.

DamienLeBars, 2020-09-17:
Thanks for your reply.
Effectively, this ambiguity does not seem removable at the level of the AST.
My goal is to use the elaborated UHDM instead.

> Surelog looks at the object type that this symbols matches with and correct the parser assumption when building the elaborated model.

The instance type is effectively corrected, but the port connection in vpiHighConn seems not ok (_ref_obj instead of a _bit_select). 
```
  |vpiModule:
  \_module: work@moduleA (instanceB), id:78, file:/home/dlb/verilog-parser/verilog/27_0_flat.sv, line:10, parent:work@top, parID:76
    |vpiDefName:work@moduleA
    |vpiName:instanceB
    |vpiFullName:work@top.instanceB
    |vpiPort:
    \_port: (port0), id:24, line:4, parent:instanceB, parID:78
      |vpiName:port0
      |vpiDirection:3
      |vpiHighConn:
      \_ref_obj: (topA), id:30, line:10          /*expected a "bit_select" here instead of "ref_obj"*/
        |vpiName:topA
        |vpiActual:
        \_logic_net: (topA), id:9, line:8, parent:work@top, parID:76
          |vpiName:topA
          |vpiFullName:work@top.topA
          |vpiNetType:1
          |vpiRange:
          \_range: , id:6, line:8
            |vpiLeftRange:
            \_constant: , id:7, line:8, parID:6
              |vpiConstType:7
              |vpiDecompile:1
              |vpiSize:32
              |INT:1
            |vpiRightRange:
            \_constant: , id:8, line:8, parID:6
              |vpiConstType:7
              |vpiDecompile:0
              |vpiSize:32
              |INT:0
```

alaindargelas, 2020-09-17:
That's a bug

PR #788 (2020-09-18): Fix high conn bug
Fixes #784
closed 2020-09-18 by alaindargelas