chipsalliance/Surelog #812: Various segmentation faults when parsing UVM

rkapuscik, 2020-09-29:
When parsing https://github.com/antmicro/uhdm-integration/tree/ibex-module-tests-uvm/tests/hello-uvm using this command line: https://github.com/antmicro/uhdm-integration/blob/ibex-module-tests-uvm/tests/hello-uvm/top.sl
I'm running into segmentation faults from various places during compilation step, traces attached.
This started happening somewhere around https://github.com/alainmarcel/Surelog/commit/037876fec81dc3b72005ef075313d34bd78f3c65, before that I saw some errors and incomplete UHDM, but no crashes.
[surelog_bt1.txt](https://github.com/alainmarcel/Surelog/files/5299114/surelog_bt1.txt)
[surelog_bt3.txt](https://github.com/alainmarcel/Surelog/files/5299115/surelog_bt3.txt)
[surelog_bt2.txt](https://github.com/alainmarcel/Surelog/files/5299116/surelog_bt2.txt)

Thomasb81, 2020-09-29:
I confirm the crash on the testcase.

But the issue is not always reproductive.
For me the last wokring commit on the provided testcase is : 5bbb18130dd77b1fc1759e87d56906af8e92d559
Next commit bring an update of flatbuffers ...

command line used:
`../bin/surelog +incdir+.+../../third_party/UVM/uvm-1.2/src/ ../../third_party/UVM/uvm-1.2/src/uvm_pkg.sv  -sverilog  -writepp -parse -d inst -mt 1 -mp 1 top.sv`

alaindargelas, 2020-09-30:
Remove the -mt and -mp args.
UHDM Compilation is not multithread safe at this point, the rest of Surelog is.
No time to work on it for a while.

Also, please point to the the latest UVM, not the 1.2 version (parallel directory: third_party/UVM/1800.2-2017-1.0)

closed 2020-09-30 by alaindargelas

Thomasb81, 2020-09-30:
Letting this issue open and eventually renaming the issue title to something _UHDM Compilation not multithread safe_, would allow to document the current limitation.

alaindargelas, 2020-09-30:
Create a separate issue.