chipsalliance/Surelog #815: Design Elaboration: type parameter identifier not supported in net declaration

DamienLeBars, 2020-10-01:
Hello,
If i have enough time for that i will contribute.
I report here a maybe known elaboration problem.

This example return an error  **Undefined type "TP"**
```verilog

module top #(parameter type TP = logic);
    TP DATA;
endmodule

```

```
********************************************
*  SURELOG SystemVerilog  Compiler/Linter  *
********************************************

Copyright (c) 2017-2019 Alain Dargelas,
http://www.apache.org/licenses/LICENSE-2.0

VERSION: 1.00
BUILT  : Sep 30 2020
DATE   : 2020-09-30.17:46:47
COMMAND: -d ast -parse -o /work /work/test.sv

[INF:CM0023] Creating log file /work/slpp_all/surelog.log.

[WRN:PA0205] /work/test.sv:2: No timescale set for "top".

[INF:CP0300] Compilation...

[INF:CP0303] /work/test.sv:2: Compile module "work@top".

[INF:CP0302] builtin.sv:4: Compile class "work@mailbox".

[INF:CP0302] builtin.sv:33: Compile class "work@process".

[INF:CP0302] builtin.sv:58: Compile class "work@semaphore".

[INF:EL0526] Design Elaboration...

[NTE:EL0503] /work/test.sv:2: Top level module "work@top".

[ERR:CP0317] /work/test.sv:3: Undefined type "TP".

[NTE:EL0508] Nb Top level modules: 1.

[NTE:EL0509] Max instance depth: 1.

[NTE:EL0510] Nb instances: 1.

[NTE:EL0511] Nb leaf instances: 1.

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 1
[WARNING] : 1
[   NOTE] : 5

********************************************
*   End SURELOG SVerilog Compiler/Linter   *
********************************************
```

alaindargelas, 2020-10-01:
I just added similar support for Type parameter in classes (CompileClass.cpp line 746).
- To add the type param to UHDM for modules, a similar line:
 if (fC->Type(list_of_type_assignments) == slList_of_type_assignments) 
has to happen in CompileModule.cpp.

- The error message you are getting is not from the UHDM level. To get rid of the message the type param has to be also registered in Surelog parameter list for the module. The 
ElaborationStep::bindPortType_ method is not looking for parameters as a possible bind yet.

alaindargelas, 2020-10-06:
I'm starting working on this.

alaindargelas, 2020-10-07:
The default type param value is now supported by #820

PR #827 (2020-10-10): Support for type param elab
Fixes #815
closed 2020-10-10 by alaindargelas