chipsalliance/Surelog #816: x86 + x64 or x64 only?

hs-apotell, 2020-10-02:
Are x86 builds officially supported? Surelog doesn't currently build under x86 environment, however, UHDM does. Would the decision be applicable to UHDM also?

mithro, 2020-10-02:
I don't think we really want to /officially/ support 32bit x86, but I see no reason to deliberately break it either. There could be good reason to support 32bit ARM or other 32bit architectures.

alaindargelas, 2020-10-03:
I think there is a lot of 64 bit hardcoded data types, especially in the Surelog serialization code (flatbuffer).
UHDM should be 32 or 64 bit as you noted.
Practically, a lot of the testcases I have for Surelog require more than 32bit address space to parse with Antlr (4, 5, 6 Gb of RAM for some)!
I don't see it practical to parse in 32 bit for real tests, only small to medium designs.

hs-apotell, 2020-10-03:
Fair enough. The smaller the build matrix, the better.

closed 2020-10-03 by hs-apotell