chipsalliance/Surelog #817: tcl vs. python as build scripting tool

hs-apotell, 2020-10-03:
Any rationale to using tcl as a build scripting tool? All of the current build scripts could have very well be written in python. The pool of developers that can contribute to python is multiple folds larger than one for tcl.

I raise the issue since I keep stumbling upon issues after issues in my quest to get a stable build on windows i.e. with tests. Each one either requires a platform based branching or finding some workaround. I have no more love for one language vs. the other but I contest that tcl fails to qualify for a "cross platform build scripting tool".

Is there any scope for reconsidering this decision before this project moves too far in this direction making it impossible to switch?

Please don't let this turn into a quest to find the best programming language. I bring this up with one and one intention only - I would like to get a stable cross platform build which builds somewhat identical across all platforms. I have had this thought a number of times before while working on this project and every time the counter thought of turning this into a war kept me away from bringing it up. But working to get CI for Surelog has once again brought some of these issues to the forefront that I had sidelined in the past.

alaindargelas, 2020-10-04:
Primary reason, I was brought up to scripting with Tcl, Python didn't exist in my days. And I have not jumped on the bandwagon when Python came to life.
I have absolutely 0 bandwidth to rewrite all the Tcl scripts in Python, and don't have the necessary experience with Python either to do so. 
So it would be a winner-takes-all situation, if you want to rewrite, you'll have to rewrite them all and get them to match the current behavior, also you'll have to maintain them going forward as I'll probably won't be able to contribute to them if they are in Python. 

Please note the most complex of the Tcl scripts is in the UHDM repo, not the ones in Surelog.
The regression.tcl and code generation scripts written in Tcl in Surelog are fairly simple and their dependency on system level calls and file manipulations should be fairly easily made portable in Windows.

I think that would be a much much easier route than the "rewrite all" one.

alaindargelas, 2020-10-04:
Can you bring a concrete example of the failure you see?

hs-apotell, 2020-10-05:
It is not the complexity of the script that hinders progress. It is the nature of what's being done in the scripts itself. For instance, cmake_gen.tcl is a very short script (under 100 lines) and simple logic but it has a few system calls like `free` making it hard to port. System calls on both unix and windows differ largely and the language isn't hiding the complexity of having to deal with them. In every script I have had to make it work on windows, I had to branch it off for the platform and do something different. Also having to deal with individual exec calls.

Let me see what options and bandwidth I have. Or may be I can find a volunteer to do it all.

alaindargelas, 2020-10-05:
I still maintain system call porting should be the easiest route.

alaindargelas, 2020-10-05:
BTW, the example you bring can be skipped on Windows, the free and lscpu calls were more debug information than anything else. If you add a catch command (like in regression.tcl) around these 2 commands, the failure won't be noticed on Windows.

If you want to maintain the behavior on Windows:
https://www.magicsplat.com/tclwinfaq.html
5.2 How can I get system information?

hs-apotell, 2021-09-12:
Reopening this issue as per the discussion [here](https://github.com/chipsalliance/UHDM/issues/484)

alaindargelas, 2021-10-03:
Both tool coexist

closed 2021-10-03 by alaindargelas